MRPT/mrpt #520: Integration of ClangFormat in git-hooks, Travis build

bergercookie, 2017-05-24:


jolting, 2017-05-25:
I'm not opposed to this idea, but proceed with caution.

I've used an atom plugin that automatically formats with clang format whenever you save. I found it useful and burdensome at the same time. It sometimes makes bad decisions.

bergercookie, 2017-06-14:
Hey @jlblancoc, @jolting

I spent some time on how we can reformat our codebase to comply to a particular
Coding style.

## Motivation

My main source of inspiration is the following articles on how MongoDB
refactored and maintained its own codebase using
[clang-format](https://clang.llvm.org/docs/ClangFormat.html).

- [Part 1](https://engineering.mongodb.com/post/succeeding-with-clangformat-part-1-pitfalls-and-planning/)
- [Part 2](https://engineering.mongodb.com/post/succeeding-with-clangformat-part-2-the-big-reformat)
- [Conclusions](https://engineering.mongodb.com/post/succeeding-with-clangformat-part-3-persisting-the-change)

After reading this and having experimented for some time with clang-format I
think we should go for the following plan of action:

## Goals

- [ ] Script for the initial (one-time) formatting of the MRPT codebase

Tailor [clang_format.py](https://github.com/mongodb/mongo/blob/master/buildscripts/clang_format.py) to the needs of our codebase and use it to refactor it. 
This script would also take care of pre-format topic branches 
either of MRPT or of its contributors forks. This is described in the [Conclusions](https://engineering.mongodb.com/post/succeeding-with-clangformat-part-3-persisting-the-change) of the article.

I 've already modified the clang_format.py script and have successfully run on
all the `.h, .cpp` files found in the following locations under MRPT root.

- apps
- libs
- samples
- python

I skipped the doc directory on purpose since it contains doxygen documentation
files not meant to be formatted as code.

The end result is just exciting!! Same consistent indentation everywhere (we can
pick either tabs or spaces) arguments of function at the same with each other,
maximum column length!

The following remarks should also be made:

- I am currently compiling the whole project to see if something needs to be
    fixed.
- One of my primary concerns was about the MRPT macros. Having inspected the
    utils/mrpt_macros.h file with multiple default clang-format formats (e.g.,
    Google, Mozilla) its macros seem to be correct.
- I would suggest that the initial formatting happens on the whole codebase in
    one single commit so that we could continue working with the things that
    matter from the get-go.

- [ ] Maintaining consistent code style

After the initial formatting, all code that is to be merged should comply to
the standards that we have set. To do that, I suggest that we:

  + Encourage users, depending on their editor of choice, to use one of the
      clang **auto-formatting plugins**. The latter would probably be able to
      read the configuration from the MRPT `.clang-format` file and use that as
      a guide (I know for a fact that the
      [vim-plugin](https://github.com/rhysd/vim-clang-format) does that).

  + In case there exists no such plugin for a particular editor, that
      contributor can make use of a **custom pre-commit git hook**. Its main job
      would be simply to call clang-format on the files to be commited.

  + **Write instructions** on the changes; modify `CONTRIBUTING`,
    `MRPT_Coding_Style.md`.

  + Add a new **travis test** that uses the clang_format.py on the changed files
      of every commit that is bundled in a PR and reports an error in case it
      doesn't comply with the coding standards. This way the contributor making
      the PR would be responsible to fix the coding style and not the project's
      main developers.

## Misc

- [ ] Preferred coding style

clang-format offers offers built-in support for the following code styles:

`LLVM` A style complying with the LLVM coding standards
http://llvm.org/docs/CodingStandards.html
`Google` A style complying with Google’s C++ style guide
https://google.github.io/styleguide/cppguide.html#Template_metaprogramming
`Chromium` A style complying with Chromium’s style guide
https://chromium.googlesource.com/chromium/src/+/master/styleguide/c++/c++.md
`Mozilla` A style complying with Mozilla’s style guide
https://developer.mozilla.org/en-US/docs/Mozilla/Developer_guide/Coding_Style
`WebKit` A style complying with WebKit’s style guide
https://webkit.org/code-style-guidelines/

Another alternative would be to have one of the predefined styles as a backbone
and then tweak with the numerous options that clang-format offers. This way we
could keep using Tabs for indentation change the position of opening/closing
brackets etc. A detailed list of these configuration variables are provided
[here](https://clang.llvm.org/docs/ClangFormatStyleOptions.html)

Ultimately I don't believe this is an important choice to make, as long as it is
consistent across the entire DB. :-)

What are your thoughts  on these?
Nikos

jolting, 2017-06-14:
Show me the code.

Can you create a branch based on MRPT-2.0, so I can see what the changes look like based on your clang format?

bergercookie, 2017-06-14:
Sure thing @jolting...

Here's the branch: https://github.com/bergercookie/mrpt/tree/mrpt-2.0-devel

You can find the script in scripts/clang-format/clang_format.py. It's a
pretty rough version that I stitched together.
To run it against your MRPT local repo run it from the root of MRPT like
this:

./scripts/clang-format/clang_format.py format # format the existing code
files.

A few more remarks:

- Code compiles successfully in MRPT-1.5. An interesting option to consider
is the "SortIncludes" which sorts the includes alphabetically. Turning this
on makes compilation fail but it's not that hard to include the missing
headers anyway..
- I think clang-format doesn't know how to deal with (very) long inline
doxygen comments (//!< ...). So what it does is indent them on the same
level with each other but this way, all but one are not on the same level
as the variable that they are documenting. This could easily be fixed
however either by letting clang-format not to reflow comments or by
converting them to standard /** ... */ doxygen comments
- MRPT license text is a touch off. This is because Google style sets a
maximum linewidth of 80 characters and this goes further than that.

bergercookie, 2017-06-17:
Hey @jolting,

Have you checked the output of clang-format yet? 
Do you have any considerations/suggestions you'd like to discuss?

@jlblancoc I have reformatted the clang_format  module [into a separate repo](https://github.com/bergercookie/clang_git_format) that we could utililze it in the child projects as well (mrpt_slam, mrpt_navigation). I think it's quite easy to use to follow. Also we can use the `lint` and `reformat_branch` methods take care of checking the syntax of upcoming PRs (e.g., if placed in a travis test) and migrate work that was started prior to the major formatting change(s).

I think the formatting module is pretty much in a working state. Assuming that we can agree in a common set of style rules, we could give it a go in either one of the child projects or in (part of) the MRPT codebase?

Cheers,
Nikos

jolting, 2017-06-18:
Hmmm. I think you have spaces instead of tabs. I have a slight preference for spaces, but MRPT uses tabs. I actually like 8-character tabs because I have done a little kernel programming. MRPT doesn't use 8-character tabs. I find 8-character tabs prevents long functions and deep nesting. I think MRPT uses 4-character tabs. Since tabs are simply a matter of preference, I think this is Jose's call.

Before any change we need to merge as much into the mrpt-2.0 branch as possible. My concern is merges might become a little trickier with the format modification. As long as there isn't a lot of churn on the 1.5 branch I don't think it will be a problem.

It looks like it modifies include order in alphabetical order. Good to know that doesn't break anything.

I'm OK with the slightly modified header as long as it's consistent now.

I couldn't review everything, but I think for the most part it looks good. It actually modified less than I was expecting.

bergercookie, 2017-06-18:
>Hmmm. I think you have spaces instead of tabs. I have a slight preference
for spaces, but MRPT uses tabs. I actually like 8-character tabs because I
have done a little kernel programming. MRPT doesn't use 8-character tabs. I
find 8-character tabs prevents long functions and deep nesting. I think
MRPT uses 4-character tabs. Since tabs are simply a matter of preference, I
think this is Jose's call.

Indeed, I think I used the default Google configuration which resorts to
spaces. But it's just enabling the UseTab flag and it will use tabs for
indentation instead. Both seem fine with me.
@jlblancoc, your call :-)

>Before any change we need to merge as much into the mrpt-2.0 branch as
possible. My concern is merges might become a little trickier with the
format modification. As long as there isn't a lot of churn on the 1.5
branch I don't think it will be a problem.

That's a concern indeed. I think the transition of old code would be easier
with the reformat_branch option of the format_code.py but it would'nt solve
everything.

>It looks like it modifies include order in alphabetical order. Good to
know that doesn't break anything.

This is controlled by the "SortIncludes" flag of .clang-format. You **have
to turn it off**  for compilation to be successful. This would most likely
need just some more "#include" lines to work.

jlblancoc, 2017-06-20:
Thanks a lot for the effort! I can't take a look at it right now, just want to let you know I didn't overlooked it and will give it a try asap

jlblancoc, 2017-06-30:
Some answers: 

@bergercookie :
> * I think clang-format doesn't know how to deal with (very) long inline
> doxygen comments (//!< ...). So what it does is indent them on the same
> level with each other but this way, all but one are not on the same level
> as the variable that they are documenting. This could easily be fixed
> however, either by making clang-format not reflow comments or by
> converting them to standard /** ... */ doxygen comments.

The point is that we must be careful: notation ` //!< XXX` refers to the entity (variable, argument,...) at its **left**, the more common `/** XXX */` refers to the **next** item (at its right)!
That's why I used the first form so often ;-)

I like the idea of adding this as another Travis test.
How should we call it? Does the python script return 0 on success as expected? 

On the Eternal debate on spaces vs tabs... You "spaces guys" are totally wrong! XD
A bit more seriously: changing to spaces, that alone, would increase the "weight" of the source code by several MBs... Just thinking of space, that's a non trivial amount. Perhaps not too relevant, but inefficient nevertheless.

I have just [proposed](https://github.com/bergercookie/mrpt/pull/3/files) to both of you a few changes to the format file. 
Let me know what do you think of that format. 

When we agree on the format, we could accept a PR from your repo, @bergercookie , to integrate the clang-format script , then make the nuke commit with the million format changes...

PS: For ROS packages, take a look at [catkin_lint](http://fkie.github.io/catkin_lint/).... it detects interesting stuff!

bergercookie, 2017-07-01:
>The point is that we must be careful: notation //!< XXX refers to the entity (variable, argument,...) at its left, the more common /** XXX */ refers to the next item (at its right)!
That's why I used the first form so often ;-)

Correct, I think I can format them during the next couple of days. 

>I like the idea of adding this as another Travis test.
How should we call it? 

style_checker.sh would be nice..

>Does the python script return 0 on success as expected?

Yup.

>On the Eternal debate on spaces vs tabs... You "spaces guys" are totally wrong! XD
A bit more seriously: changing to spaces, that alone, would increase the "weight" of the source code by several MBs... Just thinking of space, that's a non trivial amount. Perhaps not too relevant, but inefficient nevertheless.

Indeed, this is a valid point. We could stick with Tabs. I made some quick tests and even with the "Always" on, it still formats correctly the spacing around operators (e.g. single space instead of the preexisting tabs around an `=` op). One small glitch that I notice with setting this to Always is the wrong continuation indentation in some if conditionals like in the following case:

[image]
[image]

However, this may be related to some other parameter and not necessarily to tabs so I'll have to check it out again.

>I have just proposed to both of you a few changes to the format file.
Let me know what do you think of that format.

Changes seem logical. I think I commented out parts of them because they were taken care of by the `BasedOnStyle` directive.

>PS: For ROS packages, take a look at catkin_lint.... it detects interesting stuff!

This is a sidenote right? I mean, `catkin_lint` deals with static analysis of `package.xml` and `CMakeLists.txt` files, not formatting changes. Should we consider adding additional travis tests based on the results by catkin_lint?

jlblancoc, 2017-07-01:
> This is a sidenote right? I mean, catkin_lint deals with static analysis of package.xml and CMakeLists.txt files, not formatting changes. Should we consider adding additional travis tests based on the results by catkin_lint?

Sure! Sorry for the offtopic, but I just wanted to let you know about the tool, since you previously mentioned the possibility of extending the format checker to the ROS projects. Indeed, it's a different topic, but I remembered it because of the similarity in one of its recommendation about alphabetically sorting lists in CMake and alike.

bergercookie, 2017-07-04:
Hi @jlblancoc, @jolting,

I have successfully applied clang_format to my MRPT 2.0 fork. I still need to fix the corresponding travis tests, but there's a working version of it. I am currently not at home to make the PR, but please don't push directly to mrpt-2.0-devel as I' ll have to rebase and apply clang_format again. I'll probably fix the travis test tonight and make the PR tomorrow.

I'll soon provide more details on this as well as docs on how to port existing code to the new style..

Cheers,
Nikos

jlblancoc, 2017-07-04:
Copy that!

Looking forward to your PR...

bergercookie, 2017-07-05:
The travis test is in a working state. I have also moved all the travis-related files in a separate directory 
`travis` under the MRPT root. In case of a direct commit to the repo or in case of a PR containing multiple commits, these are parsed and the clang-format lint runs only on the changed files. Following are sample results of the travis output for files that successfully and unsuccessfully comply with the `.clang-format` configuration.

[image]
[image]

The same works for pull-requests as you can see [here](https://github.com/bergercookie/mrpt/pulls). One thing to note is that I detect changed files in a pull-requsts using the `TRAVIS_COMMIT_RANGE` which *doesn't run for builds triggered by initial commits* as documented [here](https://docs.travis-ci.com/user/environment-variables/). So the script isn't executed for one-commit PRs. As an alternative, we could run a full lint of the codebase every time (takes less than ~30') but this could cause a travis error on files unmodified by the PR.
 
With regards to the codebase reformat, the procedure was successful, but was not as automatic as I hoped for. The following should be noted:

- I used a SED expression for converting doxygen comments. Even though, this is 99% correct, there is also the case of a `//!< ...` comment inside a macro as [ in CObject.h](https://github.com/MRPT/mrpt/blob/mrpt-2.0-devel/libs/base/include/mrpt/utils/CObject.h#L196). Another case where `//!<...` comments aren't handled correctly is doxygen code snippets like in the case of [CImage_FlyCapture2.h](https://github.com/MRPT/mrpt/blob/mrpt-2.0-devel/libs/hwdrivers/include/mrpt/hwdrivers/CImageGrabber_FlyCapture2.h#L71). I made [manual corrections](https://github.com/bergercookie/mrpt/blob/mrpt-2.0-devel/libs/hwdrivers/include/mrpt/hwdrivers/CImageGrabber_FlyCapture2.h#L123) for this to not affect compilation but If you think this or any other code block doesn't need any reformatting, let's add the [clang-format off/on](https://philwrightdev.wordpress.com/2014/10/01/making-clang-format-ignore-sections/) directives around it.

- There exist cases that  `//!< ...` comments are suffixed with `*/`. So when they are converted to `/**...*/ they are actually turned to `/**...*/ */` which causes a compilation error. I corrected these cases manually but I could modify the convert [convert_inline_doxygen_comments.sh](https://github.com/bergercookie/mrpt/blob/mrpt-2.0-devel/scripts/clang_git_format/scripts/convert_inline_doxygen_comments.sh) to do that automatically.

- I 've wrapped up the code at 80 lines. For this to work  I've stripped some spaces off the MRPT logo. Also with this change, the inline `//` comments are wrapped up in the next line.  However, they don't affect either the doxygen documentation or the code compilation, so I think we should leave them wrapped for now and just move them over the directive they referring to, whenever we find such occurrences..

---

- [.clang-format config](https://github.com/bergercookie/mrpt/tree/mrpt-2.0-devel/scripts/clang_git_format/.clang-format)
- [check_style.sh](https://github.com/bergercookie/mrpt/tree/mrpt-2.0-devel/travis/check_style.sh)
- [MRPT codebase formatting script](https://github.com/bergercookie/mrpt/blob/mrpt-2.0-devel/scripts/clang_format_codebase.sh)
- [Instructions on working after the reformat](https://github.com/bergercookie/mrpt/blob/mrpt-2.0-devel/doc/ClangFormat.md) (not complete yet)

What do you think of these?
I could clean up the git commits and make the PR in a couple of hours.

Cheers,
Nikos

jlblancoc, 2017-07-06:
**Huge** work, Nikos!

Thanks for the great effort you devoted to this...

Yes, sure, go on and prepare the PR.
I only miss the possibility of saving one line per declaration via the
"//!< " notation. That alone would increase MRPT source code length in a
1000-2000 lines or more...

jlblancoc, 2017-07-18:
I think this one is also ready to be closed... ping if it's not.

closed 2017-07-18 by jlblancoc