MRPT/mrpt #526: Investigate build failure on Ubuntu 17.04

jolting, 2017-05-28:
The workaround is to install libmpich-dev, but I don't know why libopenmpi-dev isn't working. I think libmpi is a dependency of vtk. Perhaps libvtk6-dev has a problem.

Here's a full log:
https://launchpadlibrarian.net/321484880/buildlog_ubuntu-zesty-i386.mrpt_1%3A1.5.0~snapshot201705262258-git-0c6a3ea-201705280416~ubuntu17.04.1_BUILDING.txt.gz

The build fails with:
```
make[4]: *** No rule to make target '/usr/lib/libmpi.so', needed by 'lib/libmrpt-maps.so.1.5.0'.  Stop.
```

There are also many cmake generation warnings
```
CMake Warning at cmakemodules/DeclareMRPTLib.cmake:149 (ADD_LIBRARY):
  Cannot generate a safe runtime search path for target mrpt-detectors
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libmpi.so] in /usr/lib may be hidden by files in:
      /usr/lib/i386-linux-gnu/openmpi/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  cmakemodules/DeclareMRPTLib.cmake:4 (internal_define_mrpt_lib)
  libs/detectors/CMakeLists.txt:8 (define_mrpt_lib)

CMake Warning at cmakemodules/DeclareMRPTLib.cmake:149 (ADD_LIBRARY):
  Cannot generate a safe runtime search path for target mrpt-graphslam
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libmpi.so] in /usr/lib may be hidden by files in:
      /usr/lib/i386-linux-gnu/openmpi/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  cmakemodules/DeclareMRPTLib.cmake:4 (internal_define_mrpt_lib)
  libs/graphslam/CMakeLists.txt:4 (define_mrpt_lib)

CMake Warning at cmakemodules/DeclareMRPTLib.cmake:149 (ADD_LIBRARY):
  Cannot generate a safe runtime search path for target mrpt-hmtslam because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libmpi.so] in /usr/lib may be hidden by files in:
      /usr/lib/i386-linux-gnu/openmpi/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  cmakemodules/DeclareMRPTLib.cmake:4 (internal_define_mrpt_lib)
  libs/hmtslam/CMakeLists.txt:2 (define_mrpt_lib)

CMake Warning at cmakemodules/DeclareMRPTLib.cmake:149 (ADD_LIBRARY):
  Cannot generate a safe runtime search path for target mrpt-hwdrivers
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libmpi.so] in /usr/lib may be hidden by files in:
      /usr/lib/i386-linux-gnu/openmpi/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  cmakemodules/DeclareMRPTLib.cmake:4 (internal_define_mrpt_lib)
  libs/hwdrivers/CMakeLists.txt:162 (define_mrpt_lib)

CMake Warning at cmakemodules/DeclareMRPTLib.cmake:149 (ADD_LIBRARY):
  Cannot generate a safe runtime search path for target mrpt-maps because
  files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libmpi.so] in /usr/lib may be hidden by files in:
[... 459 lines cut ...]

    runtime library [libmpi.so] in /usr/lib may be hidden by files in:
      /usr/lib/i386-linux-gnu/openmpi/lib

  Some of these libraries may not be found correctly.

CMake Warning at tests/CMakeLists.txt:59 (ADD_EXECUTABLE):
  Cannot generate a safe runtime search path for target test_mrpt_slam
  because files in some directories may conflict with libraries in implicit
  directories:

    runtime library [libmpi.so] in /usr/lib may be hidden by files in:
      /usr/lib/i386-linux-gnu/openmpi/lib

  Some of these libraries may not be found correctly.

CMake Warning at python/CMakeLists.txt:16 (add_library):
  Cannot generate a safe runtime search path for target pymrpt because files
  in some directories may conflict with libraries in implicit directories:

    runtime library [libmpi.so] in /usr/lib may be hidden by files in:
      /usr/lib/i386-linux-gnu/openmpi/lib

  Some of these libraries may not be found correctly.
```

jlblancoc, 2017-05-28:
In the PPA, Zesty 17.04 [builds ok](https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt/+build/12639705)...

Neither PCL nor VTK are installed in the PPA packages, so your guess about the suspect seems right. 
Both have very little use (marginal, actually) inside MRPT, so the easiest thing to do would be... don't build against them!

If for some reason the libs are installed, setting DISABLE_PCL and DISABLE_VTK would make the work...

jolting, 2017-05-30:
This bug was reported and apparently fixed in the latest VTK6. Hopefully VTK6 gets patched/upgraded in 17.10.
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=848785

closed 2017-05-30 by jolting

jolting, 2017-05-30:
@jlblancoc 
By the way the difference between the standard PPA dependencies and the ones I build with is:
```
	libpcl-dev | libpcl-1.7-all-dev,
	liblas-dev,
	liblas-c-dev,
```

For completeness I turned everything on.
liblas seems fairly benign, but PCL is always broken due to some VTK cmake issues.

jlblancoc, 2017-05-30:
Good to know, thanks! 👍 

BTW: I didn't enabled by default `liblas-dev` in PPA or Debian/Ubuntu because it has what I believe is a marginal use, but instead it has **huge** dependencies, in the order of hundreds of MBs, if I recall it right (!!)