MRPT/mrpt #543: (For MRPT 2.0) WriteVariant compilation error

bergercookie, 2017-06-13:
Hi @jolting,

I am trying to port my [recent PR](https://github.com/MRPT/mrpt/pull/542) to the MRPT-2.0-devel branch, so first thing I did was to try and compile without merging pulling my changes. I get the following error in the CStream class. Any ideas why or how can I get past this?

[image]

[image]

I am using `gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4`

Cheers,
Nikos

jlblancoc, 2017-06-13:
I think that `o` should be a `t`, right? Please, try it... 

                        template <typename T>
			void WriteVariant(T t)
			{
				t.match([&](auto &o){this->WriteObject(o);});
			}

bergercookie, 2017-06-13:
>I think that o should be a t, right? Please, try it...

I think `t` is the object whose method gets called (so it shouldn't be placed inside the lambda as well).

Anyway, after having upgraded gcc, g++ from 4.9 to 6 (!) this builds successfully.

bergercookie, 2017-06-13:
So, just to wrap this up, having upgraded to gcc solves this.
To overcome similar situations in the future, we should have a minimum compiler version with something like:

```cmake
if("${CMAKE_CXX_COMPILER_ID}" STREQUAL "GNU")
    # require at least gcc 4.8
    if (CMAKE_CXX_COMPILER_VERSION VERSION_LESS 5.1)
        message(FATAL_ERROR "GCC version must be at least 5.1!")
    endif()
elseif ("${CMAKE_CXX_COMPILER_ID}" STREQUAL "Clang")
    # require at least clang 3.2
    if (CMAKE_CXX_COMPILER_VERSION VERSION_LESS 3.4)
        message(FATAL_ERROR "Clang version must be at least 3.4!")
    endif()
else()
    message(WARNING "You are using an unsupported compiler! Compilation has only been tested with Clang and GCC.")
endif()
```

closed 2017-06-13 by bergercookie

jlblancoc, 2017-06-13:
On the lambda... forget me, ooops! Too little sleep haha

Minimum compiler versions for c++11/14 stuff are updated for now in the first entry of #66 . We must be sure of minimum versions, and whether they may be a limitation when distributing future debian/ubuntu packages, that is, if the default compiler version is too old we won't be able to make releases in that distro!

jolting, 2017-06-13:
The 5 branch officially supports c++11. The 6 branch supports c++14. Presumably the 7 branch will support c++17.

I think mrpt-2.0 Just needs to compile on Ubuntu 16.04 and vs2015. This may change in the future.

bergercookie, 2017-06-13:
Hmm, I think we could get away with 4.X if only we were to use C++11 See C++11 Features in [this](http://en.cppreference.com/w/cpp/compiler_support)

However, I do agree with @jolting on this, since this is a new major release we could go ahead and target for 16.04 and higher versions of gcc/clang from the start.

jlblancoc, 2017-06-13:
Sure, support for 16.04 or newer only is what makes sense for mrpt 2.0.

Just you be sure: xenial comes by default with gcc 5.3:
https://packages.ubuntu.com/xenial/gcc

The cmake code that checks for minimum gcc version shouldn't error with
5.3! ;-)