MRPT/mrpt #545: Bulid grid map by using gramslamengine

lawrence2q, 2017-06-14:
The "gramslam-engine" is very impressive.

But, I didnot get a satisfactory map when using simulated rawlog file as input of the "gramslam-engine".
The rawlog file was produced by your " Gridmap Navigation Dataset Gennerator ".
Somestimes  the results is not better than your "ICP-SLAM" method.

Could you give some suggestion how to improve the performance.

Thank you!

jlblancoc, 2017-06-14:
Hi @lawrence2q ,

Thanks for the feedback. Please, upload some example `.rawlog` file for testing. 

(cc: @bergercookie )

lawrence2q, 2017-06-14:
Hi, @jlblancoc 

      The github website doesn't support the uploading of rawlog file and zip file. 

      So, I send you an email. (jlblanco@ual.es )

      I am sorry for the inconveninent I have caused.

jlblancoc, 2017-06-14:
Hi, 
It seems it can be uploaded via browser... here it's:
[test files.ZIP](https://github.com/MRPT/mrpt/files/1073853/test.files.ZIP)

bergercookie, 2017-06-15:
Hi @lawrence2q,

Thanks for the feedback, the zip you provided is quite detailed so I 'll make
some comments on each one of the datasets.

1. Robot travels quite a short distance and the defect is merely a glitch. I bet
if you go slower in the beginning or return to that position graphSLAM is going
to correct it.

2. Indeed the output gridmap contains a duplicate wall. This can easily be
solved by modifying the optimization distance and/or the registration max
distance/angle. For me the following parameters worked.

```ini
registration_max_distance = 0.2 // meters
registration_max_angle = 5 // degrees
ICP_goodness_thresh = 0.60 // threshold for accepting the ICP constraint in the graph
optimization_distance = 3.0
```

The resulting gridmap is given below.

[image]
[image]

3. Hmm, can't see any defect in this map.

4. That's indeed bad mapping, I managed to solve this by using the
   CLoopCloserERD (see below..) along with [this configuration file](https://github.com/MRPT/mrpt/files/1077611/graphslam.txt). Notice that you are using a laser_odometry.ini file specialized to be used in laser-only situations. However this dataset also provides odometry info so I'd use an odometry_* dataset.

The end product is given below:
[image]
[image]

There are minor defects (always will be..) but the end-result is surely manageable and is surely usuable by, let's say, a path-planning algorithm

### General remarks

* graphslam-engine supports multiple sources of input for executing graphSLAM.
Most notably:

- odometry (optional)
- 2D RangeScans

Laser-odometry works but if you do provide odometry information the resulting graph is almost certainly going to be more accurate

* There's a second Edge Registration Decider [CLoopCloserERD](http://reference.mrpt.org/devel/classmrpt_1_1graphslam_1_1deciders_1_1_c_loop_closer_e_r_d.html)
module whose task is to find potential loop closures and register the correct
ones. The algorithm isn't as straightforward as the default [CICPCriteriaERD](http://reference.mrpt.org/devel/classmrpt_1_1graphslam_1_1deciders_1_1_c_i_c_p_criteria_e_r_d.html)
but the end-results should be superior. For its configuration parameters see
.ini files suffixed with `_LC.ini`.

You can use CLoopCloserERD by issuing the corresponding flag `--edge-reg
CLoopCLoserERD`.

* While it depends on the actual deciders/optimizers that are used during
    graphSLAM, the followng parameters are most-likely to be utilized to affect
    the algorithm significantly.

    ```ini
    [NodeRegistrationDecider]
    registration_max_distance = 0.5 // what distance after prior node am I going to register a new one - the smaller the bettter - exec. will be slower
    registration_max_angle = 15  // what angle after prior node am I going to register a new one - the smaller the bettter - exec. will be slower

    [EdgeRegistrationDecider]
    ICP_max_distance =  2 // Applicable only if you use CICPCriteriaERD - maximum distance for checking other nodes for ICP constraints
    ICP_goodness_thresh = 0.80 // threshold for accepting the ICP constraint in the graph - Tinker with this value depending on the dataset - Should be set if running with CICPCriteriaERD

    use_scan_matching = true // strongly suggested that this is set to TRUE -
    prev_nodes_for_ICP = 10
    LC_eigenvalues_ratio_thresh = 2 // only utilized by CLoopCloserERD
    LC_min_remote_nodes = 2 // how many out "remote" nodes should exist in a partition for the partition to be examined for potential loop closures - only utilized by CLoopCloserERD
    LC_check_curr_partition_only = true // No reason to turn this false.

    [GraphSlamOptimizer]
    optimization_distance = 1.5; // Radius from current node in which all edges are going to be optimized - the higher the better - exec. will be slower

    ```

Cheers,
Nikos

lawrence2q, 2017-06-20:
Hi @bergercookie,
    Thank you very much for your prompt point-to-point response.

    Better results can be obtained by taking your suggestions, such as, slower speed at the beginning to get a better initial status, going back to the start point to create loop closures, adjusting the optimization distance and the registration distance.

closed 2017-06-30 by jlblancoc