MRPT/mrpt #550: Avoid pedantic warnings on extra ";"

jlblancoc, 2017-06-15:
Example warning: 

```
.../mrpt/libs/base/include/mrpt/poses/CPose2D.h:19:34: warning: extra ‘;’ [-Wpedantic]
  DEFINE_SERIALIZABLE_PRE(CPose2D);
```

This can be avoided by using the `do {... } while (0)` trick in all macros. As already done [here](https://github.com/MRPT/mrpt/blob/master/libs/base/include/mrpt/utils/COutputLogger.h#L312).

jolting, 2017-06-22:
`DEFINE_SERIALIZABLE_PRE` can actually be removed in mrpt-2.0. I think the pre and post are no longer necessary.

This is probably more about the general use of the macros though.

The ones that are still important in mrpt-2.0 are:
`DEFINE_MRPT_OBJECT`, `IMPLEMENTS_MRPT_OBJECT`, `DEFINE_VIRTUAL_MRPT_OBJECT`, `IMPLEMENTS_VIRTUAL_MRPT_OBJECT`, `DEFINE_SERIALIZABLE` and `IMPLEMENTS_SERIALIZABLE`

Those could probably be converted to CRTP without a lot of effort.

My thought on writeToStream and readFromStream is that those could be handled by template specialization. Those would become part of a CSerializer class instead, so your methods would be.

```
CSerializer<mrpt::poses::CPose2D>::writeToStream(...)
CSerializer<mrpt::poses::CPose2D>::readFromStream(...)
```

then the generic CSerializable CRTP would be
```
template<typename T>
CSerializable
{
public:
void writeToStream(...)
{
CSerializer<T>::writeToStream(...);
}
void readFromStream(...)
{
CSerializer<T>::readFromStream(...);
}
}
```

jlblancoc, 2017-06-24:
Thanks for the idea!

Obviously, I would love getting rid of all macros if possible. 

We might give an attempt at your `CSerializer` idea; but I anticipate we'll find problems with private members... will need to add a lot of ugly `friend` declarations, right?

jolting, 2017-06-27:
Right.
Almost done, much cleanup and proper commit messages needed.
https://github.com/jolting/mrpt/tree/wip1

jlblancoc, 2017-06-29:
Still have to look at all your changes, but remember something: at least
for some time I would like to support both Foo::Ptr and FooPtr. Right now,
it's supported in the 2.0 branch via a cmake switch that controls one of
those POST macros, if I recall it right...

jolting, 2017-06-29:
After I get everything compiling I'm adding some of those things back.

jlblancoc, 2017-06-29:
I feel like your 

`template <> const char * mrpt::utils::CSerializer<CObservationVisualLandmarks>::getClassName() `

could / should be merged with the existing `mrpt::utils::TTypeName<T>::get()`... they basically do the same (!!).

Perhaps both should be merged in one, with a different name than those two. Something like `...Traits...`? 
It should provide the `getClassName()` for all classes, and binary serialization only for those that make sense (most of them?). 

Thoughts?

Also: all implementation (and declaration) lines like the one quoted above should have the annoying DLL export macro for Windows 😬. But I'll add them, I know you never touch Windows if not forced to ;-)

jolting, 2017-06-29:
I will add the DLL macros.

I was thinking the same thing about TTypeName, but realized the TTypeName stuff too late. It should be a relatively simple to fix with a sed string replacement to get back to something sensible. 

I was also wondering if the abi::__cxa_demangle would give sensible types names that match the mrpt type names. Then if it doesn't just declare a specialized template to override it. This isn't technically standard, but that would avoid the whole need to manually instantiate each one. If Visual studio has a similar thing then I think we can get away with it.

My next thought was to provide a default writeToStream function based off some declarative class annotations, which I think I discussed a while ago. One benefit of putting writeToStream in a separate class it that it will be easy to add this stuff later using templates. I'll save this idea for another time. though.

jlblancoc, 2017-06-29:
In MSVC 2015, names seem to be already demangled. Still, we face the challenge of making names implementation independent, so MRPT binary files can be interchanged between OSs and compilers. 

Example: 

```
struct TPoint2D
{
};

int main()
{
	std::cout << typeid(TPoint2D).name() << std::endl;
	std::cout << typeid(std::vector<double>).name() << std::endl;
	std::cout << typeid(std::vector<TPoint2D>).name() << std::endl;
}
```

Output:
```
struct TPoint2D
class std::vector<double,class std::allocator<double> >
class std::vector<struct TPoint2D,class std::allocator<struct TPoint2D> >
```

jolting, 2017-06-29:
Yeah. I guess there are probably the odd ball people using something other than gcc and clang. I think that there is some demangle stuff for OSx as well.

I'm surprised that Windows encodes the struct/class into the symbol. gcc treats these the same.

```
#include <iostream>
#include <vector>

#include <cstdlib>
#include <memory>
#include <cxxabi.h>

std::string demangle(const char* name) {

    int status = -4; // some arbitrary value to eliminate the compiler warning

    // enable c++11 by passing the flag -std=c++11 to g++
    std::unique_ptr<char, void(*)(void*)> res {
        abi::__cxa_demangle(name, NULL, NULL, &status),
        std::free
    };

    return (status==0) ? res.get() : name ;
}
struct TPoint2D
{
};

int main()
{
        std::cout << demangle(typeid(TPoint2D).name()) << std::endl;
        std::cout << demangle(typeid(std::vector<double>).name()) << std::endl;
        std::cout << demangle(typeid(std::vector<TPoint2D>).name()) << std::endl;
}
```
Output:
```
TPoint2D
std::vector<double, std::allocator<double> >
std::vector<TPoint2D, std::allocator<TPoint2D> >
```

Almost the same.

jlblancoc, 2017-06-30:
It seems we could arrive at a custom `demangle()` code that could
unify the differences between compilers. May be a good think to do. I
would also add a cache to avoid the cost of post-processing the result
of typeid(), e.g. `std::map<std::type_info (& or *??), std::string>`
or alike.

Funny fact: MSVC returns "class" **or** "struct" depending on the
keywork used to declare it... the compiler makes a difference,
somehow, despite they being totally equivalent at the language level
(wtf!)

jolting, 2017-06-30:
This is going to bite them if/ when c++ adds static reflection.

I can't imagine anyone ever actually nameing a struct and a class the same name. Handling the ambiguity if both are declared in the same namespace would not be fun. hopefully that is still disallowed. I have no idea how one would distinguish between the types.

jlblancoc, 2018-06-29:
The original issue should be finally fixed now (in mrpt-1.5 branch; it was time ago fixed in master) after bf3f880276623ca2ff6e6f047601f6f965587f94.

Also, I think that the rest of the discussions here were somehow already integrated in the  [mrpt-typemeta](http://mrpt.ual.es/reference/master/group__mrpt__typemeta__grp.html) lib.

commit bf3f8802 (2018-06-29): Fix pedantic extra ; warnings (Closes #550)
closed 2018-06-29 by jlblancoc