MRPT/mrpt #558: Error compilación

andres-ss25, 2017-07-09:
Hello everyone

I would like to please help me, I have Ubuntu 16.04 LTS, downloaded and installed MRPT 1.5.0 from the source code, compiled without any problem. But when compiling a modification of the pbmap-tutorial example I get an error when linking a library.
[image]

jlblancoc, 2017-07-09:
¡Hola Andrés! 

Does the error also happen with the **unmodified** version of the example? 
Could you please post in a ZIP file your exact CMakeLists and .cpp files so we can replicate the issue?

andres-ss25, 2017-07-10:
Hello jose
Thanks for the prompt response. Being more detailed, I downloaded compiled and installed with sudo make install.
The example is the same tutorial-pbmap-construction.cpp.  I took it from the samples folder to desktop and  i wish compile it and test it there. Attachment .cpp and CMakeLists.txtx
[pbmap-tutorials.zip](https://github.com/MRPT/mrpt/files/1134389/pbmap-tutorials.zip)

jolting, 2017-07-10:
You're missing a space in your CMakeLists.txt file.
```
-TARGET_LINK_LIBRARIES(mrpt_pbmap-construction ${MRPT_LIBS}${PCL_LIBRARIES})
+TARGET_LINK_LIBRARIES(mrpt_pbmap-construction ${MRPT_LIBS} ${PCL_LIBRARIES})
```

closed 2017-07-10 by jolting

andres-ss25, 2017-07-18:
Hello

Jose and Joltin

Thanks for your prompt response, I really feel sorry for the mistake in the CMakeLists edition. I will pay more attention.