MRPT/mrpt #561: ROS kinetic catkin_make fails with latest mrpt

dkati, 2017-07-11:
Good evening people and thanks for everything!

trying to build mrpt_slam package i get a warning mentioning 

_CMake Warning at mrpt_slam/mrpt_ekf_slam_3d/CMakeLists.txt:52 (MESSAGE):
  Target mrpt_ekf_slam_3d will not be built.  It requires MRPT version 1.5.0
  or higher, but you have 1.3.2._

i updated the mrpt using

**sudo apt-get install libmrpt-dev=1:1.9.9~snapshot20170711-0309-git-be75da17-xenial-1~ppa1~xenial**

from **ppa:joseluisblancoc/mrpt** ppa .

I dont see the warning anymore but build fails with MANY errors on libraries,so i reverted back to my old version of mrpt with
sudo apt-get install libmrpt-dev=1:1.3.2-1

any chance to fix/remove that warning?

Thanks again

jlblancoc, 2017-07-11:
Well, we have so many open fronts to work in... :-)

You could use the 1.5 branch in this repo to keep using existing ros packages; soon we'll have to update them to work with mrpt 2.0, but any pull-request in this line would be appreciated! ;-)

dkati, 2017-07-11:
thanks for that quick reply :) me and my team will work on it and if we manage to compile,we will make a pull-request

jlblancoc, 2017-07-11:
Brilliant! Thanks.

closed 2017-07-11 by jlblancoc