MRPT/mrpt #568: 16-byte memory alignment has been lost while porting code to 2.0

jlblancoc, 2017-07-23:
I'm not sure where did it happen, but I suspect it was either in the change from smart pointer notation:

`CFooPtr a = CFooPtr(new XX);` 

to the new syntax,  and/or with the (unintended?) removal of macros for overloading the `new` operator in `MRPT_MAKE_ALIGNED_OPERATOR_NEW` (which seems to still be correctly placed [here](https://github.com/MRPT/mrpt/blob/master/libs/base/include/mrpt/utils/CObject.h#L201)).

Build fails in non 64bit archs (arm, i386) due to Eigen mem alignment requirements for all statically and dynamically-allocated objects.

Error example (see more in [launchpad](https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt/+packages)):
´´´
[ RUN      ] ICPTests.AlignScans_icpClassic
test_mrpt_slam: /usr/include/eigen3/Eigen/src/Core/DenseStorage.h:109: Eigen::internal::plain_array<T, Size, MatrixOrArrayOptions, 16>::plain_array() [with T = double; int Size = 2; int MatrixOrArrayOptions = 0]: Assertion `(internal::UIntPtr(eigen_unaligned_array_assert_workaround_gcc47(array)) & (15)) == 0 && "this assertion is explained here: " "http://eigen.tuxfamily.org/dox-devel/group__TopicUnalignedArrayAssert.html" " **** READ THIS WEB PAGE !!! ****"' failed.
Aborted (core dumped)
´´´

Ideally `alignas(16)` would be a solution, but it seems `new` does not honor it... until C++17, if I understood it right (sigh).

(cc: @jolting ; I'll continue investigating it, but just wanted to ping you in case this rings a bell to you...)

jolting, 2017-07-25:
I thought alignas were a c++11 feature. I admittedly have little experience with them. On the other hand I can't thing of any changes which would explicitly break them.

Perhaps this might be a clue:
```
std::shared_ptr<T>(new T(args...)) performs at least two allocations (one for the object T and one for the control block of the shared pointer), while std::make_shared<T> typically performs only one allocation (the standard recommends, but does not require this, all known implementations do this)
```

jolting, 2017-07-25:
Give this a shot...
```c++
#include <eigen3/Eigen/Core>
#include <memory>

template <typename T, class ...Args>
std::shared_ptr<T> make_aligned_shared(Args &&... args) {
  return std::allocate_shared<T>(Eigen::aligned_allocator<T>(), args...);
}
```

jolting, 2017-07-25:
woops Forgot to forward
```c++
#include <eigen3/Eigen/Core>
#include <memory>

template <typename T, class ...Args>
std::shared_ptr<T> make_aligned_shared(Args &&... args) {
  return std::allocate_shared<T>(Eigen::aligned_allocator<T>(), std::forward<Args>(args)...);
}
```

jlblancoc, 2017-07-26:
I think you found the key: before the switch to `shared_ptr` we explicitly called `new`. The `new` overloads are still there but apparently `std::allocate_shared` is not using it.

closed 2017-07-28 by jlblancoc