MRPT/mrpt #571: CDisplayWindow3D crashes after Qt-related refactoring

jlblancoc, 2017-07-27:
How to reproduce: 

Build MRPT samples, execute: 

```
gdb samples/opengl_objects_demo/opengl_objects_demo
run
(upon crash)
bt
```

Output:

```
Thread 2 "opengl_objects_" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffe47c0700 (LWP 2106)]
0x00007ffff7bb0abd in mrpt::gui::CMyGLCanvasBase::OnMouseMove(wxMouseEvent&) () from /home/jlblanco/code/mrpt/build/lib/libmrpt-gui.so.1.9
(gdb) bt
#0  0x00007ffff7bb0abd in mrpt::gui::CMyGLCanvasBase::OnMouseMove(wxMouseEvent&) () from /home/jlblanco/code/mrpt/build/lib/libmrpt-gui.so.1.9
```

Any quick idea on the cause of the problem, @LisGein ?

LisGein, 2017-07-27:
No, but it crashed on the line 
136 this->SetFocus();

LisGein, 2017-07-27:
I am checking it.

LisGein, 2017-07-27:
I found several problems:
* I don't know how and why, but 
in the constructor sets m_gl_context as nullptr
after calls OnWindowCreation and "if (!m_gl_context)" == false. 

* This is a second problem:
[image]

jlblancoc, 2017-07-27:
Thanks!

I recall adding a SetFocus in each event of the mouse as a workaround to
some other issues... I can't test it now, but probably an "if" that checks
if the window is not nullptr (or something alike...) before doing the
SetFocus would help. Or even the SetFocus could be commented out to confirm
it's the cause...

commit b8583fb9 (2017-07-30): workaround to wx crash

@lisgein this solves the wx crashes.... let's move on for now

Closes #571
closed 2017-07-30 by jlblancoc