MRPT/mrpt #572: Any plans to release 1.5 debian packages?

corot, 2017-08-04:
@jlblancoc ,  debian web is calling for your attention:   :grin:
https://packages.qa.debian.org/m/mrpt.html

jlblancoc, 2017-08-05:
Thanks! We had some problems, but I'm precisely today talking to mentors
and the ball is rolling again, very soon we'll have a new pkg! ... and,
probably, the usual cascade of unexpected build errors in rare
architectures 😂😂

Thanks for the kind reminder ;-)

jlblancoc, 2017-08-17:
The latest attempt (which might be uploaded to Debian soon) is [here](https://mentors.debian.net/package/mrpt) just in case you are curious...

closed 2017-08-17 by jlblancoc

jlblancoc, 2017-08-23:
[Here](https://ftp-master.debian.org/new/mrpt_1:1.5.3-1.html) is it in the [NEW](https://wiki.debian.org/NewQueue) queue, due to the minor version change... waiting for approval.