MRPT/mrpt #573: Visual system freeze and crash in CMyGLCanvas_DisplayWindow3D::OnMouseDown

spsancti, 2017-08-09:
Application: [mvsim](https://github.com/ual-arm-ros-pkg/mvsim)
Problem: visual subsystem freezes on startup, clicking mouse inside of the window causes crash.

Stack trace of crash: https://pastebin.com/eRQEevtE
This problem exists after [b8583fb](https://github.com/MRPT/mrpt/commit/b8583fb98a4241d87ae08ac78b0420c154f5e1a5) and doesn't occur before [2c97406](https://github.com/MRPT/mrpt/commit/2c97406a50153ca5a890c5e8612e6df2f31b3988). Also, build type (debug/release) doesn't affect the issue.

Frozen window upon terminal window: [image]

@LisGein , @jlblancoc any ideas about this?

jolting, 2017-08-11:
CObserver/CObservable isn't thread safe. mvsim runs the UI in separate thread and the simulation in the main thread. Is it possible that the main thread is accessing CObserver/ CObservable objects? Those should only be accessed by the UI thread right?

spsancti, 2017-08-11:
@jolting 
Thanks for your remark, I will look into it now. However, it worked before...

jolting, 2017-08-11:
That happens a lot with race conditions. The code changes in question may have exposed the race condition and made it more likely by changing the timing of the code. I'm not sure it's a race condition, but that's my suspicion.

spsancti, 2017-08-14:
Well, I investigated for some more time.
The problem persists even when I just create the window in main.
This is the code that also causes the problem: https://pastebin.com/BDEbJsRx
The minimal code is just to create window.

I will investigate it deeper after GSoC ends because I'm out of my main schedule now.

jlblancoc, 2017-08-15:
@spsancti thanks, Borys for the minimal code to reproduce the problem!! Yes, please try to get over (if possible) this and go on with the remaining issues for GSoC and we'll try to debug it... 

Before, I must integrate into `master` a lot of pending changes from the 1.5 branch (!!). I'm sure they will not interfere with the GUI subsystem, but I prefer to debug after integrating the pending changes.

jlblancoc, 2017-08-16:
I can reproduce the error in both, Linux and Windows. 

For now, I traced the problem to `this` being a few bytes off when invoking `CMyGLCanvas_DisplayWindow3D::OnMouseDown()`. It's true that the observer/observable pattern is not thread-safe, but this bug seems unrelated to that... probably a missing static_cast<> / dynamic_cast<> or alike seems to me to be the problem... Will continue digging in this.

jlblancoc, 2017-08-16:
@spsancti Please, update after the commit that just closed this issue and give it a try... 

It seems to be working for me now ;-)

jlblancoc, 2017-08-16:
FYI: there were two independent problems here: 
* wxWidgets `Connect()` messing around with object pointers (fixed using the more modern, and safer, `Bind()` API), and
* The opengl scene seems not to be copied, since a smart pointer was used instead of a **reference** to a smart pointer. See [here](https://github.com/MRPT/mrpt/commit/3fb3bdd62e8b521baaff17c8cb1d1aef67773ea2#diff-ca719dfcecaed7216b7de8f640560d68R189), @LisGein , perhaps the equivalent code for Qt also needs to be fixed? (didn't looked at it...)

closed 2017-08-16 by jlblancoc

spsancti, 2017-08-16:
@jlblancoc thanks! Graphics work fine now!

commit 3fb3bdd6 (2017-08-16): Fix wxWidgets OpenGL rendering freezing

Closes #573

cc: @LisGein, @spsancti