MRPT/mrpt #576: lightweight geometry data initialization

tyuownu, 2017-08-14:
Hi jlblancoc and guys, when I ran the `icp-slam` under windows(rawlog with only lidar scan data) with some errors. Than  I found that in lightweight_geom_data.h, all geometry-relative class like `TPose2d`, with default constructor
``` C++
inline TPose2D() {}
```
Under `linux`, it initialize to a small value close to 0, `[x = 0, y = 0, phi = 0]`, and thing goes all right; but on  `windows` it initialize to `[x = -9.25596e+61, y = -9.25596e+61, phi = -9.25596e+61]`, that make a big difference when do `sin`, `cos` or some other relative operation.
```
Environment:
linux: ubuntu 16.04 amd64
windows: win7 64bit
```
And my question:
1. why not initialized it to zeros directly?
2. How does the value `-9.25596e+61` come from?

Thanks

jlblancoc, 2017-08-14:
Hi! 

That's done on purpose: imagine initializing a vector of 1e6 TPose2D elements. Initializing to zeros will be a waste of time if the user knows that he/she must do it later. That's why this set of classes are called "light-weight", vs the other set (e.g. CPose2D, CPoint3D,...) which indeed are initialized to zeros but also have more memory and run-time cost.

On that weird number: it's a good old known one ;-) I can guess you are running your app in MSVC in Debug mode. In Debug, Visual C++ includes a "safety" mechanism: all uninitialized memory is set to a predefined bit pattern (0xCD). See this [great summary of predefined bit values](https://stackoverflow.com/a/370362/1631514). A double with 0xcdcdcdcd is read exactly as that number.

closed 2017-08-14 by jlblancoc

tyuownu, 2017-08-15:
@jlblancoc .Got it, I didn't notice the difference between `debug` and `release` mode on MSVC. Thanks a lot.