MRPT/mrpt #579: Octomap from ROS is not include problem

spsancti, 2017-08-16:
When building current [master](https://github.com/MRPT/mrpt/commit/1f47b0361a0755ed1a86d77b123587ed31993e00), if there is ros-octomap installed, cmake script finds octomap directory correctly. 
However, when compiling, there is an error ``` octomap/octomap.h: No such file or directory ```: [build log](https://pastebin.com/mVjSaz1j)
When uninstalling ros-octomap and using internal one, there is no problem. 
Built with internal octomap, MRPT installs and then there is an error when building catkin workspace with MRPT included: [build log](https://pastebin.com/b3Ujni6y)
Installing ros-octomap between building MRPT and catkin workspace solves compilation problem but looks like a piece of duct tape.

Sincerely, Borys.

jolting, 2017-08-16:
I think it's missing a `INCLUDE_DIRECTORIES("${OCTOMAP_INCLUDE_DIRS}")` line. It works with the debian packaged version because /usr/include is already in the system path.

On ROS the include path should is  `/opt/ros/$ROS_RELEASE/include`

jlblancoc, 2017-08-17:
@spsancti :
> "MRPT installs"

this means you do a `sudo make install`? (OK, I saw your build log and it's clear that's the case).

The `INCLUDE_DIRECTORIES ` directive mentioned by Hunter is already there, both for building mrpt-maps itself, and for any user program via [these lines](https://github.com/MRPT/mrpt/commit/6e2003d7f3035f61cbbaa0ed1068a18aa1de5ede)... which specifically are disabled for the version of MRPTConfig.cmake used to "make install". 

Why is it disabled? Because, when doing "make install", MRPT CMake scripts **should** be copying octomap headers to a subdirectory under `mrpt-maps/include`. I'll check if that is working. I never do "make install" (and don't recommend using it; it's better IMO to simple set MRPT_DIR=XXX before running CMake in ROS or the user apps), so that case wasn't tested....

Thanks for the feedback and the detailed descriptions of the errors!

spsancti, 2017-08-17:
Hi @jlblancoc !
Thanks for your review. 
However, the issue still here :(
I made every build clean (that's long).

Here is [cmake](https://pastebin.com/d752VmkX) and [make](https://pastebin.com/0fffA2N1) outputs with octomap installed.
And also [cmake](https://pastebin.com/MBXUfWgC) and [make](https://pastebin.com/kQJd9fcJ) with octomap uninstalled (everything OK).

There is also [catkin_make](https://pastebin.com/UHqGSAar) output when octomap installed (build failed). 

Sincerely, Borys.

jolting, 2017-08-17:
`INCLUDE_DIRECTORIES` is for the `IF (NOT OCTOMAP_FOUND)` case
https://github.com/MRPT/mrpt/blob/master/cmakemodules/script_octomap.cmake#L66

Don't we want one for the `IF (OCTOMAP_FOUND)` case or is that handled somewhere else?
Maybe somewhere about here?
https://github.com/MRPT/mrpt/blob/master/cmakemodules/script_octomap.cmake#L15

jlblancoc, 2017-08-17:
@spsancti : 

> I made every build clean (that's long).

For these tests, and since you only need MRPT libs, you should disable building all MRPT applications and tests from CMake. If you like GUIs, you can do it via cmake-gui or ccmake. It will save *a lot* of build time ;-)

For your case "octomap uninstalled": when you do "(sudo) make install", can you see it actually copies the octomap headers under "/usr/local/include/mrpt/maps/include/mrpt/..." ??
I think I tested it and it worked (?).

Also, please check the contents of the file "/usr/local/share/cmake/MRPTConfig.cmake" (or something alike) and search for "octomap" to see if what you see makes sense. 

Will continue debugging it...

@jolting :
> INCLUDE_DIRECTORIES is for the IF (NOT OCTOMAP_FOUND) case
> https://github.com/MRPT/mrpt/blob/master/cmakemodules/script_octomap.cmake#L66
>
> Don't we want one for the IF (OCTOMAP_FOUND) case or is [...]

Thanks! My idea (despite the huge mess of old CMake code accumulated over the years, in the wait to get it better organized someday....) was trying to write new CMake code as "respectful" as possible: don't add the INCLUDE_DIRECTORIES and "pollute" **all** MRPT examples, libraries and apps, only those that use `mrpt-maps`. 

For now I will probably just pollute everything with a global INCLUDE_DIRECTORIES . I would rather prefer to rewrite all dependencies to external/system libraries as CMake >=3.0's constructs of the type `add_library(<name> INTERFACE )` , such that adding a dependency to those external libs will automatically . Probably this implies using EXPORTED targets, etc. A lot of work to do here!

Will try to give a quick, easy fix to this to unblock the situation...

jlblancoc, 2017-08-18:
@spsancti can you please confirm whether the latest commit has fixed this one?

jolting, 2017-08-18:
specifically in the ros-octomap build

spsancti, 2017-08-19:
Hi, @jlblancoc 
Sorry for the long time to reply.

For now, MRPT compiles in both situations. 
When ros-octomap not installed ```make install``` doesn't install octomap and catkin build fails with ```no such file or directory``` (do I need to add some ```include_directories``` to my cmake?)
With ros-octomap installed everything goes well :+1: 

Sincerely, Borys.

jlblancoc, 2017-08-19:
Thanks for the tests!

In this last situation that fails: could you please rerun 

VERBOSE=1 cmake .

In the mrpt build directory and grab the output?

Please, also attach the file CMakeCache.txt

spsancti, 2017-08-20:
Hi, @jlblancoc !

MRPT builds both times successfully.

[Here](https://pastebin.com/zQCfb2fX) is verbose output of MRPT cmake.
[Here](https://pastebin.com/JqhyWz5h) is MRPT's CMakeCache.txt.
[Here](https://pastebin.com/xvU7y4QA) is verbose build of my catkin workspace.
[Here](https://pastebin.com/D7GjPRx9) is catkin's CMakeCache.txt.

Sincerely, Borys.

jlblancoc, 2017-08-21:
mmm... thanks again for the tests!

I just re-tested it (with the current "master"), and doing a "make install" when octomap is NOT detected in the system (and using the built-in version inside MRPT, as in your configuration, correctly installs a subdirectory like: 

[prefix]/libs/maps/include/octomap/

with prefix for your case `/usr/local`. **Can you please check** if there is or not an "octomap" directory under "libs/maps/include"? 

You can also re-run "make install" and see the log of installed files. 

If these files are installed, the MRPTConfig.cmake script does not need to do anything at all, since any ROS package marking mrpt-maps as a dependence will automatically find octomap headers as well...

spsancti, 2017-08-21:
Hi, @jlblancoc !

Now I'm absolutely clueless.
Catkin build fails with octomap not found.
[Here](https://pastebin.com/DUYcA1Qv) is my make install output. There is clearly seen ```octomap```.
[Here](https://pastebin.com/WBsMCxWA) is the ```locate octomap``` result. It is definitely in ```/usr/local/libs/maps/include/octomap``` and its includes are in ```/usr/local/include/mrpt/maps/include/mrpt/otherlibs/octomap```
[Here](https://pastebin.com/3vPSBNJk) is the gcc (and g++) default include path. Here is definitely ```/usr/local/include/``` in it.

Am I experiencing any glitch or is the matrix broken again?
Sincerely, Borys.

jlblancoc, 2017-08-21:
The only explanation I can find is that the ROS node that fails to build has a missing dependency on `mrpt-maps`. You can verify it via: 
`cd ~/catkin_ws/build && VERBOSE=1 make NODE_THAT_FAILS -j1` 

and inspecting if the compiler cmd line has or not `-I/usr/local/include/mrpt/maps/include/`. Having `/usr/local/include/` is not enough...

> Am I experiencing any glitch or is the matrix broken again?

XD

jlblancoc, 2017-08-21:
BTW: that would mean that the bug is in the ROS package instead, so I'm closing this ticket, but feel free to reopening it if you find something else suspicious, ok? 
Cheers.

closed 2017-08-21 by jlblancoc

jolting, 2017-08-22:
ROS really should be built against ros-octomap instead of mrpt built-in octomap. This works now right?

jlblancoc, 2017-08-24:
I tried to build both MRPT master + mrpt_navigation master with the pkg `ros-${ROS_DISTRO}-octomap` installed, and both built cleanly.

Only remaining test is to build MRPT + any other user app (making use mrpt-maps) and if it works, we should consider this issue as closed... unless you find something else that fails (?).

spsancti, 2017-08-24:
Hi, all!
I made the same setup on my computer at work and tried to build it. 
Everything works fine in both cases.

Probably, there is really a glitch at my computer.
I think, we can consider it as closed. 

Sincerely, Borys,

jolting, 2017-08-24:
Make install puts things in `/usr/local`. Debian packages aren't suppose to and typically do not install in `/usr/local`.

If you find files in there it's probably because you did a `make install`. Unfortunately cmake does not provide a `make uninstall` rule. The manual install often interferes with a package install. I avoid messing up my dev system this way by using checkinstall or debuild.

CMake should output a install_manifest.txt file after `make install`, so you can manually delete the installed files.

jlblancoc, 2017-08-24:
I totally agree in the idea of "make install" being a last resort
mechanism, and mrpt documentation never mention it for those reasons...

But just one note: cmake does not provide make uninstall by default, but
mrpt scripts DO generate it, so as long as one does not delete the build
directory, you could go there again and run the uninstall ;-)

Cheers

Wehrdo, 2017-10-15:
If anybody else is here because, like me, you're getting an `<octomap/octomap.h>: No such file or directory` error when compiling your application, I installed `liboctomap-dev`, and it successfully built after that.

As a note, I did run `make install`, but after reading the comment above, did `make uninstall` to undo it, but was still getting the same errors.

commit ac2ff3fe (2017-08-17): Fix octomap -I directories (Fixes #579)