MRPT/mrpt #582: CMultiMetricMap::writeToStream and CMultiMetricMap::readFromStream functions

yanjundream, 2017-08-24:
Hi dear developers, I am currently using mrpt in my project. I want to use `CMemoryStream` to transfer a `CSimplePointsMap` got from `CMetricMapBuilderICP` in different program. The problem is about the version parameter in two functions. 
1. What is the version here? How can I get my version to fed in function?
2. I tried to set version = 11. The function `writeToStream` will do nothing. However if I give a nullptr as version parameter, the `writeToStream` seems work.
3. Howver when run `readFromStream`,  I got ` =============== MRPT EXCEPTION =============
void mrpt::utils::CStream::internal_ReadObjectHeader(std::__cxx11::string&, bool&, int8_t&), line 450:
Expecting 0x00 00 00 while parsing old streaming header (Perhaps it's a gz-compressed stream? Use a GZ-stream for reading)
Exception while parsing typed object  from stream!
`
when I set version in `readFromStream` is 11. Aparently if I set version to other number, the function `readFromStream` will not work.

So I still do not know how to use the CStream idea in mrpt. My purpose here is to use UDP socket transfer the `CSimplePointsMap`. I have see the `CClientTCPSocket` example, but it is TCP and does not provide UDP method. Anyway, is there anything I understand wrongly? Thanks!

jlblancoc, 2017-08-24:
Hi!

Those write/read methods are not intended for direct use by the user. Instead, use the operators  `>>` and `<<` which, when applied to an MRPT's `CSerializable` and `CStream` objects, perform the binary (de)serialization.

```cpp
// Sender:
CMemoryStream ms;
ms << myObject;

// Send contents of the "ms" buffer.
// ms.getRawBufferData()  and ms.getTotalBytesCount()
```

and: 

```cpp
// Receiver:
CMemoryStream ms;
// Receive the data and write it to "ms" with:
ms.WriteBuffer(&buffer[0], buffer.size());

ms.Seek(0); // Rewind to initial position of the "file" / "memory buffer"

// De-serialize object:
CSerializable::Ptr  obj = ms.ReadObject(); // Or "CSerializablePtr" for mrpt <1.9.9

```

To convert the `CSerializable` smart pointer to the desired object type, read section "Smart pointer typecasting" [here](http://mrpt.ual.es/reference/master/porting_mrpt2.html).

Cheers.

closed 2017-08-24 by jlblancoc

yanjundream, 2017-08-25:
Hi, thank you for your reply and examples! It helps a lot. But I still have problems.
```
    CMemoryStream mr;
    mr.WriteBuffer(&ch, n);

    mr.Seek(0);
    CSerializablePtr obj = mr.ReadObject();
```
I use this code in receiver part. But last line gives me `Segmentation fault ` error. I have check `mr` and I am preety sure it receive the data. I also try `CSerializable::Ptr  obj = ms.ReadObject(); ` and same  Segmentation fault ` error.
 
I also tried 
```
   CMultiMetricMap obj;
    mr>>obj;
```
. But this one is weird more, it use up my memory and computer crash totally. Do you know what should I do to change the `CMemoryStream` received to `CMultiMetricMap`? Thanks！

jolting, 2017-08-25:
If you don't mind can you post the backtrace of the segfault?

yanjundream, 2017-08-25:
Here is the debug info  when using 

```
    CMemoryStream mr;
    mr.WriteBuffer(&ch, n);
    mr.Seek(0);
    CSerializable::Ptr  obj = mr.ReadObject(); 
```
```
Binding...
Binding succed!...
20856 byte receive Connection created!

Program received signal SIGSEGV, Segmentation fault.
mrpt::utils::TRuntimeClassId::createObject (this=0x0) at /home/jun/projects/mrpt/libs/base/src/utils/CObject.cpp:90
90		if (!ptrCreateObject)
(gdb) backtrace
#0  mrpt::utils::TRuntimeClassId::createObject (this=0x0) at /home/jun/projects/mrpt/libs/base/src/utils/CObject.cpp:90
#1  0x00000000004087c5 in mrpt::utils::CStream::ReadObject<mrpt::utils::CSerializable> (this=0x7ffffff09660)
    at /usr/local/include/mrpt/base/include/mrpt/utils/CStream.h:226
#2  0x0000000000407cfc in mrpt::utils::CStream::ReadObject (this=0x7ffffff09660)
    at /usr/local/include/mrpt/base/include/mrpt/utils/CStream.h:206
#3  0x0000000000407b79 in main (argc=1, argv=0x7fffffffdc58) at /home/jun/projects/slam/footbot_sslam/vismap.cpp:135
(gdb) l
85	/*---------------------------------------------------------------
86						TRuntimeClassId::createObject
87	 ---------------------------------------------------------------*/
88	CObject* TRuntimeClassId::createObject() const
89	{
90		if (!ptrCreateObject)
91		{
92			perror(
93				"[TRuntimeClassId::createObject] Trying to create an object with "
94				"not dynamic constructor\n");
(gdb) 
```

This is when using    ` CSerializablePtr obj = mr.ReadObject();
`

```
Binding...
Binding succed!...
20856 byte receive Connection created!

Program received signal SIGSEGV, Segmentation fault.
mrpt::utils::TRuntimeClassId::createObject (this=0x0) at /home/jun/projects/mrpt/libs/base/src/utils/CObject.cpp:90
90		if (!ptrCreateObject)
(gdb) bt
#0  mrpt::utils::TRuntimeClassId::createObject (this=0x0) at /home/jun/projects/mrpt/libs/base/src/utils/CObject.cpp:90
#1  0x00000000004087c5 in mrpt::utils::CStream::ReadObject<mrpt::utils::CSerializable> (this=0x7ffffff09660)
    at /usr/local/include/mrpt/base/include/mrpt/utils/CStream.h:226
#2  0x0000000000407cfc in mrpt::utils::CStream::ReadObject (this=0x7ffffff09660)
    at /usr/local/include/mrpt/base/include/mrpt/utils/CStream.h:206
#3  0x0000000000407b79 in main (argc=1, argv=0x7fffffffdc58) at /home/jun/projects/slam/footbot_sslam/vismap.cpp:134
(gdb) l
85	/*---------------------------------------------------------------
86						TRuntimeClassId::createObject
87	 ---------------------------------------------------------------*/
88	CObject* TRuntimeClassId::createObject() const
89	{
90		if (!ptrCreateObject)
91		{
92			perror(
93				"[TRuntimeClassId::createObject] Trying to create an object with "
94				"not dynamic constructor\n");
(gdb) 

```

Thanks!

jolting, 2017-08-25:
`this` is null.
Well this line of code won't throw an exception if the class is not found.
```
		const TRuntimeClassId* ret = registeredClasses[className];
```

https://github.com/MRPT/mrpt/blob/0b42871d769f527a0cf1168c890b2ed4e9bed9c6/libs/base/src/utils/internal_class_registry.cpp#L101

But it suggest that the class you are trying to deserialize has not been registered with the class registry.
```
https://github.com/MRPT/mrpt/blob/master/libs/base/include/mrpt/utils/CStream.h#L225
```

yanjundream, 2017-08-25:
Hi Jolting, thank you for your clue. That is the reason of segment fault.
Now I change to 
```
    CMultiMetricMap* temp = new CMultiMetricMap();
    CMultiMetricMap ::Ptr  obj(temp);
    dynamic_pointer_cast<CSerializable>(obj) =  mr.ReadObject(); 
```
and huge problem comes again. My computer stuck as same when I use 
```
   CMultiMetricMap obj;
    mr>>obj;
```.
My computer died and I have to hard restarted my computer. Do you no what happened about this and how to make it work? I already hard restarted my laptop more than ten times and I am worried about it ^^. Thanks!

jolting, 2017-08-25:
I think you actually want

```
CMultiMetricMap ::Ptr  obj(new CMultiMetricMap());
mr.ReadObject(obj.get());
```

jlblancoc, 2017-08-26:
Try these two things:
- as a general rule, ALWAYS wrap your code inside main() with a try ...
catch and show catch'ed exceptions to cerr, so you don't miss them.

- before starting comms, register the class you expect to receive: (search
for many instances of this across mrpt sources)

registerClass(CLASS_ID(CFoo));