MRPT/mrpt #585: Add "nodelets" for in-process efficient communications

jlblancoc, 2017-08-27:
* [ ] Add @jolting 's [idea](https://github.com/MRPT/mrpt/issues/66#issuecomment-288636679) to the new module mrpt-comms.
* [ ] Write examples.
* [ ] Write unit tests.

jlblancoc, 2018-07-31:
This is finally in progress!  :+1:  
Changed the name "node" to "topic" since, I think, it's more accurate...
And created a simple example (subscriber callback not being called yet for some reason, but not looked much to this yet).

jolting, 2018-08-01:
I forgot about this one. We have c++17 on now.

jolting, 2018-08-01:
I think a filter class might be cool. On construction it subscribes and on destruction it unsubscribes.
Also a method of instantiating filters based on a yaml file.

jlblancoc, 2018-08-01:
And we'll end up with a mini ROS2 XDD

> I think a filter class might be cool. 

👍 yeap!

jolting, 2018-08-01:
WIthout being encumbered by interprocess communication. At least initially.

jlblancoc, 2018-09-04:
This one was already done time ago: 240f9d4317123498dc223b11aa2a831d0cf083a6
Closing.

closed 2018-09-04 by jlblancoc