MRPT/mrpt #589: Website

balloch, 2017-08-31:
Ya'll may already be aware, but your website is down...just thought that you would want to know.

jlblancoc, 2017-08-31:
Thanks! 

Nope actually... visitors keep coming because the error apparently "only" affects the home page, so I didn't realize. I can't wait to port the entire web to some static generator and forget about PHP & DBs (sigh)

jolting, 2017-08-31:
Looks like a redirect loop.
I'm learning vuejs.  You would probably like it a lot. Maybe the next site can be built on that.

jlblancoc, 2017-08-31:
Thanks! Didn't know that one...

There are **so** many possible frameworks! After testing a few ones, I liked and got minimally productive with [Pelican](http://docs.getpelican.com/en/stable/) (my personal web is generated with it).

Probably many frameworks would do the job for MRPT, as long as they are minimally configurable via plugins (in Python or whatever). 
In general, I like the idea of generating the web contents from plain text (Markdown, etc.) files stored in the main MRPT repo, so each version has its associated docs. sphinx is a good alternative, as used in ReadTheDocs.io, or in (I think) OpenCV. Perhaps one half of the web should go inside the main MRPT repo (the part most related to classes, real code, etc.) while the other half might go in a separate repo (for generic pages like "downloads", "datasets", etc. things that won't change often). 

Just some wild ideas...
Porting will take a considerable effort, so it's worth thinking carefully about all the options.
Though, nobody has spare time for it now, I guess! ;-)

jlblancoc, 2017-09-07:
It's back online!  https://www.mrpt.org/

closed 2017-09-07 by jlblancoc