MRPT/mrpt #594: Problem with simplemap after SLAM

RastaMarcus, 2017-09-25:
Hi,

I am using the SLAM demo rbpf with the gridmapping_optimal_sampling.ini config file. Everything seems to be correct while running.
However, when using the produced simplemap (for example with pf-localization), I get a Segmentation Fault...

jlblancoc, 2017-09-26:
Hi, 

Do you also have a segfault when using any other .simplemaps of those provided as examples along MRPT?

Do you use the offline app `pf-localization` for localization or a custom app?

RastaMarcus, 2017-09-28:
I only get segfaults when using .simplemaps created by the rbpf-slam app.
I am using a slightly customed app based on pf-localization.
However, Markus Bader got the same result on his computer with the pf-localization app.

jlblancoc, 2017-09-28:
Have you tried running it with `gdb`, then running `bt` at the segfault position? That would be helpful to debug this...

jlblancoc, 2017-09-28:
Or just attach an example .simplemap file that makes pf-localization crash...

RastaMarcus, 2017-09-28:
Yes !
Sorry for the lack of details.
The problem appears when calling : `metricMap.loadFromProbabilisticPosesAndObservations(simpleMap);`
in pf_localization.
I upload a screenshot of the frame stack (the lines might not be exact due to my modifications)
[image]
Here is the map (I added .txt so that github doesn't complain about the file type)
[_finalmap_.simplemap.txt](https://github.com/MRPT/mrpt/files/1340707/_finalmap_.simplemap.txt)

jlblancoc, 2017-09-29:
The .simplemap file was the key!

This was "solved" with the commit referenced above... the current problem seems to be that there are no observations each of the robot pose-observations pairs of the simplemap. They seem to be empty. My fix has been an assert() so an exception is thrown to detect this condition instead of lead to a segfault. 

Now: please, if you have time to debug this, dig into this to see if the CSensoryFrame's are actually empty when building the map. Otherwise, this might be a bug in the de-serialization from file, but many other .simplemap files seem to load without problems...

jlblancoc, 2017-10-09:
Any news on this topic? I'll close the issue for now, since it **seems** that the problem is the way in which the map is generated in the mapper program. If you used an out-of-the-box node from the MRPT ROS packages, please move the discussion to the corresponding repository. Thanks.

closed 2017-10-09 by jlblancoc