MRPT/mrpt #598: ppa versions

paul-mesnilgrente, 2017-10-06:
Hi,

just a question on the development branch. There is written in this page: https://www.mrpt.org/MRPT_in_GNU/Linux_repositories

that it corresponds to the "branch 2.0" but there is no such branch on Github and in the ppa, the version 1.9.9. So the PPA corresponds to what? I am a bit confused. You maybe need to update your documentation.

jolting, 2017-10-06:
2.0 isn't officially released. The reference to it in the PPA is really the master branch. 1.9.9 will be the version until an official 2.0 release has been made.

jlblancoc, 2017-10-07:
Hi @paul-mesnilgrente , 
jolting explained it perfectly. 

Anyway, I have added a few more comments on version numbers to [that page](https://www.mrpt.org/MRPT_in_GNU/Linux_repositories).

closed 2017-10-07 by jlblancoc