MRPT/mrpt #599: Build issues when Qt or OpenGL disabled

jolting, 2017-10-09:
- [ ] Qt5 components shouldn't be required. They should be optional and build to succeed if the optional components are not found.
- [ ] Disabling OpenGL should not break the build.

From: https://gitter.im/MRPT/mrpt

Alex Giokas @alexge233 Oct 05 04:47
Hi Hunter, I'm trying to build a minimal version of MRPT-1.9.9 (from github) without any GUI stuff.
I keep running into the same issue:
```
-- Architecture (uname -m): x86_64
-- Kernel name (uname -s): Linux
-- Could NOT find OpenGL (missing:  OPENGL_INCLUDE_DIR) 
-- Could NOT find GLUT (missing:  GLUT_glut_LIBRARY GLUT_INCLUDE_DIR) 
-- **Warning**: OpenGL and/or GLUT not found! OpenGL capabilities will be disabled.
--                      OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so
--                      OPENGL_glu_LIBRARY: OPENGL_glu_LIBRARY-NOTFOUND
--                      GLUT_glut_LIBRARY: GLUT_glut_LIBRARY-NOTFOUND
--                      OPENGL_INCLUDE_DIR: OPENGL_INCLUDE_DIR-NOTFOUND
-- Could NOT find wxWidgets (missing:  wxWidgets_LIBRARIES wxWidgets_INCLUDE_DIRS) 
wxWidgets was not found automatically. Please, set wxWidgets_ROOT_DIR to the lib directory to enable it in MRPT.
CMake Warning at cmakemodules/script_qt.cmake:10 (find_package):
  By not providing "FindQt5.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5" with any of
  the following names:

    Qt5Config.cmake
    qt5-config.cmake

  Add the installation prefix of "Qt5" to CMAKE_PREFIX_PATH or set "Qt5_DIR"
  to a directory containing one of the above files.  If "Qt5" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:234 (include)

-- cotire 1.7.9 loaded.
And when I build:
[ 45%] Building CXX object libs/gui/CMakeFiles/mrpt-gui.dir/src/CGlCanvasBase.cpp.o
/root/mrpt/libs/gui/src/CGlCanvasBase.cpp: In member function ‘void mrpt::gui::CGlCanvasBase::resizeViewport(int, int)’:
/root/mrpt/libs/gui/src/CGlCanvasBase.cpp:119:20: error: ‘GLint’ was not declared in this scope
  glViewport(0, 0, (GLint)w, (GLint)h);
                    ^
/root/mrpt/libs/gui/src/CGlCanvasBase.cpp:119:37: error: ‘glViewport’ was not declared in this scope
  glViewport(0, 0, (GLint)w, (GLint)h);
                                     ^
/root/mrpt/libs/gui/src/CGlCanvasBase.cpp: In member function ‘void mrpt::gui::CGlCanvasBase::clearColors()’:
/root/mrpt/libs/gui/src/CGlCanvasBase.cpp:124:65: error: ‘glClearColor’ was not declared in this scope
  glClearColor(clearColorR, clearColorG, clearColorB, clearColorA);
                                                                 ^
/root/mrpt/libs/gui/src/CGlCanvasBase.cpp: In member function ‘virtual double mrpt::gui::CGlCanvasBase::renderCanvas(int, int)’:
/root/mrpt/libs/gui/src/CGlCanvasBase.cpp:260:16: error: ‘GL_ALL_ATTRIB_BITS’ was not declared in this scope
   glPushAttrib(GL_ALL_ATTRIB_BITS);
                ^
/root/mrpt/libs/gui/src/CGlCanvasBase.cpp:260:34: error: ‘glPushAttrib’ was not declared in this scope
   glPushAttrib(GL_ALL_ATTRIB_BITS);
                                  ^
/root/mrpt/libs/gui/src/CGlCanvasBase.cpp:263:12: error: ‘GL_DEPTH_TEST’ was not declared in this scope
   glEnable(GL_DEPTH_TEST);
            ^
/root/mrpt/libs/gui/src/CGlCanvasBase.cpp:263:25: error: ‘glEnable’ was not declared in this scope
   glEnable(GL_DEPTH_TEST);
                         ^
/root/mrpt/libs/gui/src/CGlCanvasBase.cpp:264:12: error: ‘GL_ALPHA_TEST’ was not declared in this scope
   glEnable(GL_ALPHA_TEST);
            ^
/root/mrpt/libs/gui/src/CGlCanvasBase.cpp:265:12: error: ‘GL_TEXTURE_2D’ was not declared in this scope
   glEnable(GL_TEXTURE_2D);
            ^
[... 111 lines cut ...]
/home/zuperath/code/mrpt/libs/gui/src/CGlCanvasBase.cpp:263:25: error: ‘glEnable’ was not declared in this scope
   glEnable(GL_DEPTH_TEST);
                         ^
/home/zuperath/code/mrpt/libs/gui/src/CGlCanvasBase.cpp:264:12: error: ‘GL_ALPHA_TEST’ was not declared in this scope
   glEnable(GL_ALPHA_TEST);
            ^
/home/zuperath/code/mrpt/libs/gui/src/CGlCanvasBase.cpp:265:12: error: ‘GL_TEXTURE_2D’ was not declared in this scope
   glEnable(GL_TEXTURE_2D);
            ^
/home/zuperath/code/mrpt/libs/gui/src/CGlCanvasBase.cpp:269:19: error: ‘GLsizei’ was not declared in this scope
   resizeViewport((GLsizei)width, (GLsizei)height);
                   ^
/home/zuperath/code/mrpt/libs/gui/src/CGlCanvasBase.cpp:293:17: error: ‘GL_MODELVIEW’ was not declared in this scope
    glMatrixMode(GL_MODELVIEW);
                 ^
/home/zuperath/code/mrpt/libs/gui/src/CGlCanvasBase.cpp:293:29: error: ‘glMatrixMode’ was not declared in this scope
    glMatrixMode(GL_MODELVIEW);
                             ^
/home/zuperath/code/mrpt/libs/gui/src/CGlCanvasBase.cpp:294:19: error: ‘glLoadIdentity’ was not declared in this scope
    glLoadIdentity();
                   ^
/home/zuperath/code/mrpt/libs/gui/src/CGlCanvasBase.cpp:307:11: error: ‘glFlush’ was not declared in this scope
   glFlush();
           ^
```

alexge233, 2017-10-09:
Hi @jolting thanks for moving this here. I'll try with `BUILD_mrpt-gui=OFF` and get back to you.

jlblancoc, 2017-10-09:
> Qt5 components shouldn't be required. They should be optional and build to succeed if the optional components are not found.
> Disabling OpenGL should not break the build.

Absolutely! Thanks for taking care of this... I missed the Gitter discussion.

paul-mesnilgrente, 2017-10-12:
The flag `BUILD_mrpt-gui=OFF` is disabling most of the other modules... So this is not a solution.

alexge233, 2017-10-13:
@paul-mesnilgrente which modules are being disabled? Can you copy-paste a list please?

paul-mesnilgrente, 2017-10-13:
This is the module list before to put gui off:
```
base       ON
bayes      ON
comms      ON
detectors  ON
graphs     ON
graphslam  ON
gui        ON
hmtslam    ON
hwdrivers  ON
kinematics ON
maps       ON
nav        ON
obs        ON
opengl     ON
pbmap      OFF
slam       ON
tfest      ON
topography ON
vision     ON
```
And this is the list after:
```text
base       ON
bayes      ON
comms      ON
detectors  OFF
graphs     ON
graphslam  OFF
gui        OFF
hmtslam    OFF
hwdrivers  OFF
kinematics ON
maps       ON
nav        ON
obs        ON
opengl     ON
pbmap      OFF
slam       ON
tfest      ON
topography ON
vision     ON
```

alexge233, 2017-10-13:
Paul this is fine, test it by trying to link?

alexge233, 2017-10-17:
@jolting Why is `hmtslam` tied to the GUI libraries? It seems that if we disable `BUILD_mrpt-gui=OFF` we can't use `CMetricMapBuilderICP` because it links to it.

jolting, 2017-10-17:
Good question. It might take some work to decouple those. 

```
define_mrpt_lib(
	# Lib name
	slam
	# Dependencies
	mrpt-bayes
	mrpt-graphs
	mrpt-vision
	mrpt-tfest
	mrpt-maps
	# And deps of the deps:
	mrpt-obs
	mrpt-opengl
	mrpt-base
	)
```

The opengl dependency should be removed. I think that is the source of the problem. I'm just not certain what ramifications that has until I try it.

jlblancoc, 2017-10-17:
Hi guys, 

There's work to do here, just first, just some clarifications: 

@alexge233: 

> Why is hmtslam tied to the GUI libraries? It seems that if we disable BUILD_mrpt-gui=OFF we can't use CMetricMapBuilderICP because it links to it.

CMetricMapBuilderICP belongs to mrpt-slam, you should be able to use it.

@jolting :
`The opengl dependency should be removed. I think that is the source of the problem. I'm just not certain what ramifications that has until I try it.`

The idea is that the module `mrpt-opengl` can be cleanly built WITHOUT a real GL/GLU/... library in the system. In that case, the module still provides the valuable support to build 3D representations of all kind of sensor observations, robot maps, etc. Even if they can't be rendered, they can be streamed over a socket, to a file, etc. and rendered in another machine. 

bottom line: disabling `mrpt-gui` is not actually a good solution. MRPT design criteria are that, even without neither Qt nor wxWidgets, all modules should build cleanly. Only when, at runtime, one attempts to open a GUI window, we have an exception like "this feature is not supported, rebuild with blah blah".

So, I'll take a look at your build errors, and try to add `#if ` guards or whatever it takes to fix the origin of all problems.

jlblancoc, 2017-10-17:
FYI: there is a graphical representation of modules dependencies on the [main Doxygen page](http://mrpt.ual.es/reference/master/).

jolting, 2017-10-18:
That makes sense. I was thinking it *might* be possible to remove the opengl dependency.

One drawback that I see is that the packaged version of mrpt-opengl and mrpt-gui depends many system dependencies. Just pulling in opengl on a Ubuntu system will install tons of packages. If you're trying to ship a rootfs to an embedded device, that is quite a lot of overhead if you just want to run some slam algorithms. Decoupling slam from opengl would avoid that problem, but that might be tricky.

The best alternative is to build it yourself with the right options turned off. I think that should work after my last pull request, but you still need to manually disable Qt. Am I wrong?

jlblancoc, 2017-10-18:
Good point... it might make sense to publish an mrpt-headless version of the packages, at least via ppa.

On Qt: if it's not detected in the system, just like so many other dependencies, all should build cleanly. I put much effort in this aim over the years, and if it's not the case with Qt it's only because of the recent gsoc changes and the lack of testing.

Might even add one more travis configuration to test building with minimal dependencies...

alexge233, 2017-10-18:
Just FYI (and you probably already know this) I went looking at the `readelf -d` output of the libraries in question, and they all seem to depend on opengl:

```
readelf -d /usr/lib/x86_64-linux-gnu/libmrpt-vision.so.1.9

Dynamic section at offset 0x349c70 contains 36 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libmrpt-obs.so.1.9]
 0x0000000000000001 (NEEDED)             Shared library: [libmrpt-opengl.so.1.9]
 0x0000000000000001 (NEEDED)             Shared library: [libmrpt-base.so.1.9]
 0x0000000000000001 (NEEDED)             Shared library: [libopencv_video.so.2.4]
 0x0000000000000001 (NEEDED)             Shared library: [libopencv_calib3d.so.2.4]
 0x0000000000000001 (NEEDED)             Shared library: [libopencv_features2d.so.2.4]
 0x0000000000000001 (NEEDED)             Shared library: [libopencv_highgui.so.2.4]
 0x0000000000000001 (NEEDED)             Shared library: [libopencv_imgproc.so.2.4]
 0x0000000000000001 (NEEDED)             Shared library: [libopencv_core.so.2.4]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000e (SONAME)             Library soname: [libmrpt-vision.so.1.9]
```

```
readelf -d /usr/lib/x86_64-linux-gnu/libmrpt-slam.so.1.9

Dynamic section at offset 0x224aa8 contains 33 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libmrpt-vision.so.1.9]
 0x0000000000000001 (NEEDED)             Shared library: [libmrpt-tfest.so.1.9]
 0x0000000000000001 (NEEDED)             Shared library: [libmrpt-maps.so.1.9]
 0x0000000000000001 (NEEDED)             Shared library: [libmrpt-obs.so.1.9]
 0x0000000000000001 (NEEDED)             Shared library: [libmrpt-opengl.so.1.9]
 0x0000000000000001 (NEEDED)             Shared library: [libmrpt-base.so.1.9]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000e (SONAME)             Library soname: [libmrpt-slam.so.1.9]
```

```
readelf  -d /usr/lib/x86_64-linux-gnu/libmrpt-obs.so.1.9

Dynamic section at offset 0x177dc0 contains 29 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libmrpt-opengl.so.1.9]
 0x0000000000000001 (NEEDED)             Shared library: [libmrpt-base.so.1.9]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
 0x000000000000000e (SONAME)             Library soname: [libmrpt-obs.so.1.9]
```

```
readelf -d /usr/lib/x86_64-linux-gnu/libmrpt-maps.so.1.9

Dynamic section at offset 0x1c3808 contains 34 entries:
  Tag        Type                         Name/Value
 0x0000000000000001 (NEEDED)             Shared library: [libmrpt-obs.so.1.9]
 0x0000000000000001 (NEEDED)             Shared library: [libmrpt-graphs.so.1.9]
 0x0000000000000001 (NEEDED)             Shared library: [libmrpt-opengl.so.1.9]
 0x0000000000000001 (NEEDED)             Shared library: [libmrpt-base.so.1.9]
 0x0000000000000001 (NEEDED)             Shared library: [liboctomap.so.1.6]
 0x0000000000000001 (NEEDED)             Shared library: [liboctomath.so.1.6]
 0x0000000000000001 (NEEDED)             Shared library: [libGL.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
[... 58 lines cut ...]
 0x0000000000000001 (NEEDED)             Shared library: [libopencv_objdetect.so.3.3]
 0x0000000000000001 (NEEDED)             Shared library: [libopencv_cudaimgproc.so.3.3]
 0x0000000000000001 (NEEDED)             Shared library: [libopencv_cudawarping.so.3.3]
 0x0000000000000001 (NEEDED)             Shared library: [libopencv_cudafilters.so.3.3]
 0x0000000000000001 (NEEDED)             Shared library: [libopencv_cudaarithm.so.3.3]
 0x0000000000000001 (NEEDED)             Shared library: [libopencv_imgcodecs.so.3.3]
 0x0000000000000001 (NEEDED)             Shared library: [libopencv_imgproc.so.3.3]
 0x0000000000000001 (NEEDED)             Shared library: [libopencv_core.so.3.3]
 0x0000000000000001 (NEEDED)             Shared library: [libstdc++.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libm.so.6]
 0x0000000000000001 (NEEDED)             Shared library: [libgcc_s.so.1]
 0x0000000000000001 (NEEDED)             Shared library: [libc.so.6]
```

Even more bizarelly I've seen it failing to execute (after building & link) because `libassimp.so.4` was not found, which if I understand correctly is brought in by the mrpt-gui?

I'd like to help, but I may need some guidance here. 

I see that `define_mrpt_lib` is used by each `CMakeLists.txt` under `libs/XXX/` where XX is the name of the library being built, but I'm trying to figure if there is a way to edit the cmake in order to set *which* libraries I want built. 

My hope is that I won't have to remove opengl, but rather build only a "core" version which doesn't require it.

I've tried searching for:

`grep -rnw . -e "find_package(OpenGL"` but can't find where it is being searched for, and I've also tried finding where the link flags are being used `grep -rnw . -e "OPENGL_LIBRARIES"` but again to no avail.

jolting, 2017-10-18:
https://github.com/MRPT/mrpt/blob/master/cmakemodules/script_gl_glut.cmake

Basically, DISABLE_OPENGL should disable OPENGL in mrpt as a whole, but shouldn't prevent mrpt-opengl from building. That may be counter intuitive, but necessary because mrpt-slam depends on mrpt-opengl.

The headless comment was suggesting a second set of mrpt package which would rebuild all packages without the opengl and other dependencies for a minimal build. If you're building it yourself that will be irrelevant. Simply use the DISABLE_* mechanism to disable all of those external libraries you don't want.

jlblancoc, 2017-10-18:
Slowly, slowly... 

@alexge233 : I think there's some misunderstanding or missing information here: I have reproduced your conditions by UNINSTALLING all qt, wxWidgets, GL, GLU, etc. libraries from my Ubuntu 16.04 system. 
Then, compiling mrpt from scratch (**git master branch**) gives no error at all. Everything builds, including mrpt-gui, etc. 

What version of MRPT are you using? Please, attach your CMakeCache.txt file.

alexge233, 2017-10-19:
@jlblancoc I'm using the master branch from github.
Someone from the team went ahead and compiled it on the cloud instance, so I'll have to recreate the process in a VM. Please bear with me.

alexge233, 2017-10-19:
Here's the cmake output:

```
cmake ..
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Architecture (uname -m): x86_64
-- Kernel name (uname -s): Linux
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Looking for timegm
-- Looking for timegm - found
-- Looking for _mkgmtime
-- Looking for _mkgmtime - not found
-- Looking for alloca
-- Looking for alloca - not found
-- Looking for gettid
-- Looking for gettid - not found
-- Looking for sincos
-- Looking for sincos - found
-- Looking for posix_memalign
-- Looking for posix_memalign - found
-- Looking for _aligned_malloc
-- Looking for _aligned_malloc - not found
-- Looking for strtok_r
-- Looking for strtok_r - found
-- Looking for alloca.h
-- Looking for alloca.h - found
-- Looking for linux/serial.h
-- Looking for linux/serial.h - found
-- Looking for linux/input.h
-- Looking for linux/input.h - found
-- Looking for malloc.h
-- Looking for malloc.h - found
-- Looking for malloc/malloc.h
-- Looking for malloc/malloc.h - not found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for inttypes.h
-- Looking for inttypes.h - found
-- Looking for winsock2.h
-- Looking for winsock2.h - not found
-- Looking for windows.h>
#include <winusb.h
-- Looking for windows.h>
#include <winusb.h - not found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of long double
-- Check size of long double - done
-- Check if the system is big endian
-- Searching 16 bit integer
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- Could NOT find OpenGL (missing:  OPENGL_gl_LIBRARY OPENGL_INCLUDE_DIR) 
[... 146 lines cut ...]
--  NITE2 library                        :  No 
--  OpenKinect libfreenect               : Yes (Built-in) 
--  OpenNI2                              :  No 
--  PGR FlyCapture2                      :  No 
--  PGR Triclops                         :  No 
--  Phidgets                             :  No 
--  RoboPeak LIDAR                       : Yes 
--  SwissRanger 3/4000 3D camera         :  No 
--  Videre SVS stereo camera             :  No 
--  libudev (requisite for XSensMT4)     :  No 
--  xSENS MT 3rd generation              : Yes (Built-in) 
--  xSENS MT 4th generation              :  No 
--  Intersense sensors                   :  No 
-- 
-- 
-- ***********************************************************************
-- * WARNING: It's STRONGLY recommended to build MRPT with OpenCV support.
-- *  To do so, set OpenCV_DIR to its CMake build dir. If you want to go 
-- *  on without OpenCV, proceed to build instead. 
-- ***********************************************************************
-- 
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ubuntu/mrpt/build
```

alexge233, 2017-10-19:
This is a build in a blank VM, build failed with:

```
/home/ubuntu/mrpt/libs/gui/src/CAboutBoxBase.cpp: In member function ‘std::__cxx11::string CAboutBoxBase::information(const string&, const string&) const’:
/home/ubuntu/mrpt/libs/gui/src/CAboutBoxBase.cpp:89:19: error: invalid operands of types ‘const char [5]’ and ‘const char [2]’ to binary ‘operator+’
   str += "None" + "\n";
                   ^
libs/gui/CMakeFiles/mrpt-gui.dir/build.make:277: recipe for target 'libs/gui/CMakeFiles/mrpt-gui.dir/src/CAboutBoxBase.cpp.o' failed
make[2]: *** [libs/gui/CMakeFiles/mrpt-gui.dir/src/CAboutBoxBase.cpp.o] Error 1
CMakeFiles/Makefile2:944: recipe for target 'libs/gui/CMakeFiles/mrpt-gui.dir/all' failed
make[1]: *** [libs/gui/CMakeFiles/mrpt-gui.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
```

Haven't seen that before. Are you guys updating master branch?

jlblancoc, 2017-10-19:
Thanks, that last msg is helpful.will try to fix it (I'm on the phone now). Looks like a problem that only occur with that version of gcc. I use gcc 7, normally, these days

alexge233, 2017-10-19:
Ok, I'm going to try with `DISABLE_OPENGL` and turn off the gui tomorrow, and see if that changes anything.

By the way, I can see that `libassimp.so.4` is downloaded and built, however one problem I constantly run into is that once I'm trying to execute an app linked to `mrpt`, the `ldloader` complains about not finding it. I'll post on that too.

In the meantime, what can I do to help for the *minification* process?

jlblancoc, 2017-10-19:
> By the way, I can see that libassimp.so.4 is downloaded and built, 

Just set BUILD_ASSIMP=OFF in CMake when configuring to solve that one.

jlblancoc, 2017-10-19:
> /home/ubuntu/mrpt/libs/gui/src/CAboutBoxBase.cpp:89:19: error: invalid operands of types ‘const char [5]’ and ‘const char [2]’ to binary ‘operator+’
>    str += "None" + "\n";

That particular error is now fixed in master. It only happened when OpenCV is not present... a situation we never tested in recent times!

alexge233, 2017-10-20:
I have a question regarding OpenCV (there was an issue on stackoverflow to which @jlblancoc replied) which keeps coming back: we use OpenCV 3.3 but MRPT links to 2.4. Quite often we get segmentation faults because of that, which is part of the reason why we're building mrpt from source. Is there a way to change this?

Regarding the previous issues:
1. I pulled from master
2. I am now using: `-DBUILD_ASSIMP=OFF -DDISABLE_OPENGL=ON -DBUILD_mrpt-gui=OFF`
3. Because this is a VM, I've just used Ubuntu's opencv-2.4
4. There is no OpenGL or Qt.

The output:

```
cmake .. -DBUILD_ASSIMP=OFF -DDISABLE_OPENGL=ON -DBUILD_mrpt-gui=OFF
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Architecture (uname -m): x86_64
-- Kernel name (uname -s): Linux
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Looking for timegm
-- Looking for timegm - found
-- Looking for _mkgmtime
-- Looking for _mkgmtime - not found
-- Looking for alloca
-- Looking for alloca - not found
-- Looking for gettid
-- Looking for gettid - not found
-- Looking for sincos
-- Looking for sincos - found
-- Looking for posix_memalign
-- Looking for posix_memalign - found
-- Looking for _aligned_malloc
-- Looking for _aligned_malloc - not found
-- Looking for strtok_r
-- Looking for strtok_r - found
-- Looking for alloca.h
-- Looking for alloca.h - found
-- Looking for linux/serial.h
-- Looking for linux/serial.h - found
-- Looking for linux/input.h
-- Looking for linux/input.h - found
-- Looking for malloc.h
-- Looking for malloc.h - found
-- Looking for malloc/malloc.h
-- Looking for malloc/malloc.h - not found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for inttypes.h
-- Looking for inttypes.h - found
-- Looking for winsock2.h
-- Looking for winsock2.h - not found
-- Looking for windows.h>
#include <winusb.h
-- Looking for windows.h>
#include <winusb.h - not found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of long double
[... 415 lines cut ...]
/usr/bin/ld: CMakeFiles/kinect-3d-view.dir/kinect-3d-view_main.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
apps/kinect-3d-view/CMakeFiles/kinect-3d-view.dir/build.make:122: recipe for target 'bin/kinect-3d-view' failed
make[2]: *** [bin/kinect-3d-view] Error 1
CMakeFiles/Makefile2:3846: recipe for target 'apps/kinect-3d-view/CMakeFiles/kinect-3d-view.dir/all' failed
make[1]: *** [apps/kinect-3d-view/CMakeFiles/kinect-3d-view.dir/all] Error 2
[ 86%] Linking CXX executable ../../bin/rawlog-grabber
/usr/bin/ld: CMakeFiles/rawlog-grabber.dir/rawloggrabber_main.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
apps/rawlog-grabber/CMakeFiles/rawlog-grabber.dir/build.make:122: recipe for target 'bin/rawlog-grabber' failed
make[2]: *** [bin/rawlog-grabber] Error 1
CMakeFiles/Makefile2:4097: recipe for target 'apps/rawlog-grabber/CMakeFiles/rawlog-grabber.dir/all' failed
make[1]: *** [apps/rawlog-grabber/CMakeFiles/rawlog-grabber.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

pthread library is installed:

```
sudo ldconfig -p | grep "pthread"
	libpthread.so.0 (libc6,x86-64, OS ABI: Linux 2.6.32) => /lib/x86_64-linux-gnu/libpthread.so.0
```

jlblancoc, 2017-10-20:
> I have a question regarding OpenCV (there was an issue on stackoverflow to which @jlblancoc replied) which keeps coming back: we use OpenCV 3.3 but MRPT links to 2.4. Quite often we get segmentation [...]

I still think the best way is to simply build MRPT against your OpenCV 3.3 version. Just delete the Ubuntu system packages for opencv.

> I am now using: -DBUILD_ASSIMP=OFF -DDISABLE_OPENGL=ON -DBUILD_mrpt-gui=OFF

Try again without `-DBUILD_mrpt-gui=OFF`. Just don't install the Qt or wxWidgets development packages and there will be no build errors (there was one issue, but it's now fixed in master).

I have even added new test cases to Travis to test the "headless" configuration and an extremely minimal
case without any extra library (without OpenCV, etc.), so any build error becomes obvious from now on.

jlblancoc, 2017-10-25:
@alexge233 was this finally solved?

alexge233, 2017-10-25:
@jlblancoc regarding the OpenCV:

that is what I am doing, but it is not a realistic scenario; for example ROS pulls its own OpenCV, and thus if another OpenCV is present this becomes an issue. For now I just build from source.

regarding the `-DBUILD_mrpt-gui=OFF` I tried again with: 
`cmake .. -DBUILD_ASSIMP=OFF -DDISABLE_OPENGL=ON` 
and this is what happens:

```
/home/ubuntu/mrpt/apps/icp-slam-live/icp-slam-live_main.cpp:107:43: note: #pragma message: TODO: Should these be global?
[ 89%] Building CXX object libs/graphslam/CMakeFiles/mrpt-graphslam.dir/src/CEdgeCounter.cpp.o
[ 89%] Linking CXX executable ../../bin/icp-slam-live
/usr/bin/ld: CMakeFiles/icp-slam-live.dir/icp-slam-live_main.cpp.o: undefined reference to symbol 'pthread_create@@GLIBC_2.2.5'
//lib/x86_64-linux-gnu/libpthread.so.0: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
apps/icp-slam-live/CMakeFiles/icp-slam-live.dir/build.make:124: recipe for target 'bin/icp-slam-live' failed
make[2]: *** [bin/icp-slam-live] Error 1
CMakeFiles/Makefile2:2782: recipe for target 'apps/icp-slam-live/CMakeFiles/icp-slam-live.dir/all' failed
make[1]: *** [apps/icp-slam-live/CMakeFiles/icp-slam-live.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 89%] Building CXX object libs/graphslam/CMakeFiles/mrpt-graphslam.dir/src/TSlidingWindow.cpp.o
[ 89%] Building CXX object libs/detectors/CMakeFiles/mrpt-detectors.dir/src/CDetectorDoorCrossing.cpp.o
[ 89%] Linking CXX executable ../../bin/kf-slam
[ 89%] Built target kf-slam
[ 89%] Building CXX object libs/detectors/CMakeFiles/mrpt-detectors.dir/src/CDetectableObject.cpp.o
[ 89%] Building CXX object libs/graphslam/CMakeFiles/mrpt-graphslam.dir/src/CWindowObserver.cpp.o
[ 89%] Linking CXX shared library ../../lib/libmrpt-detectors.so
[ 89%] Built target mrpt-detectors
[ 89%] Linking CXX shared library ../../lib/libmrpt-graphslam.so
[ 89%] Built target mrpt-graphslam
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

jolting, 2017-10-25:
What about DISABLE_OPENCV? Do you need opencv support in MRPT?

jolting, 2017-10-25:
Regarding the icp-slam-live can you add to mrpt/apps/icp-slam-live/CMakeLists.txt?
```
+find_package( Threads )
-TARGET_LINK_LIBRARIES(${PROJECT_NAME} ${MRPT_LINKER_LIBS})
+TARGET_LINK_LIBRARIES(${PROJECT_NAME} ${MRPT_LINKER_LIBS} ${CMAKE_THREAD_LIBS_INIT} )
```

alexge233, 2017-10-30:
Actually no we don't need OpenCV in MRPT so disabling it is probably the best solution!
I'm away until Wednesday so I'll get back to you then with the changes.
Many thanks for the help!

jolting, 2017-11-08:
No response in 9 days. Closing.

closed 2017-11-08 by jolting

MariaRamos89, 2017-11-10:
Hi,
I've tried to build MRPT with the following flags:
```shell
cmake .. -DBUILD_ASSIMP=OFF -DDISABLE_OPENGL=ON -DDISABLE_OPENCV=ON
```
and it's failing:

```shell
../../lib/libmrpt-base.so.1.9.9: undefined reference to `dladdr'
collect2: error: ld returned 1 exit status
apps/mrpt-perfdata2html/CMakeFiles/mrpt-perfdata2html.dir/build.make:107: recipe for target 'bin/mrpt-perfdata2html' failed
make[2]: *** [bin/mrpt-perfdata2html] Error 1
CMakeFiles/Makefile2:4541: recipe for target 'apps/mrpt-perfdata2html/CMakeFiles/mrpt-perfdata2html.dir/all' failed
make[1]: *** [apps/mrpt-perfdata2html/CMakeFiles/mrpt-perfdata2html.dir/all] Error 2
```

I am using this commit 9749168e35288a16ec86a4314df4d3629adec042 because it is the latest one that builds in travis.

jolting, 2017-11-10:
@MariaRamos89 
60be480ad90f1188b4005148765cec0525914c92 addressed that exact issue.

MariaRamos89, 2017-11-13:
Thanks @jolting 
I've tried with that commit and now a new error appears: 
```shell
/home/maria/libs/mrpt/apps/kinect-stereo-calib/kinect_calibrate_guiMain.cpp: In constructor ‘kinect_calibrate_guiDialog::kinect_calibrate_guiDialog(wxWindow*, wxWindowID)’:
/home/maria/libs/mrpt/apps/kinect-stereo-calib/kinect_calibrate_guiMain.cpp:1327:34: error: ‘class CMyGLCanvas’ has no member named ‘getOpenGLSceneRef’
  auto openGLSceneRef = m_plot3D->getOpenGLSceneRef();
                                  ^
/home/maria/libs/mrpt/apps/kinect-stereo-calib/kinect_calibrate_guiMain.cpp: In member function ‘void kinect_calibrate_guiDialog::ProcessNewGrabbedObs()’:
/home/maria/libs/mrpt/apps/kinect-stereo-calib/kinect_calibrate_guiMain.cpp:1796:14: error: ‘class CMyGLCanvas’ has no member named ‘last_timestamp’
    m_plot3D->last_timestamp = m_last_obs->timestamp;
              ^
/home/maria/libs/mrpt/apps/kinect-stereo-calib/kinect_calibrate_guiMain.cpp: In member function ‘void kinect_calibrate_guiDialog::CalibUpdate3DViewCameras()’:
/home/maria/libs/mrpt/apps/kinect-stereo-calib/kinect_calibrate_guiMain.cpp:2939:20: error: ‘class CMyGLCanvas’ has no member named ‘setOpenGLSceneRef’
  m_plot3D_cameras->setOpenGLSceneRef(scene);
                    ^
apps/kinect-stereo-calib/CMakeFiles/kinect-stereo-calib.dir/build.make:86: recipe for target 'apps/kinect-stereo-calib/CMakeFiles/kinect-stereo-calib.dir/kinect_calibrate_guiMain.cpp.o' failed
make[2]: *** [apps/kinect-stereo-calib/CMakeFiles/kinect-stereo-calib.dir/kinect_calibrate_guiMain.cpp.o] Error 1
```

I've tried also adding the flag `-DBUILD_APPLICATIONS=OFF` and undefined reference problem appears:
```
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::updateLastPos(int, int)'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::updateCameraParams(mrpt::opengl::CCamera&) const'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `vtable for mrpt::gui::CGlCanvasBase'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::setMousePos(int, int)'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::setAzimuthDegrees(float)'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `typeinfo for mrpt::gui::CGlCanvasBase'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::updateZoom(mrpt::gui::CGlCanvasBase::CamaraParams&, float) const'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::setZoomDistance(float)'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::setElevationDegrees(float)'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::clearColors()'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::setCameraFOV(float)'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::renderCanvas(int, int)'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::updateOrbitCamera(mrpt::gui::CGlCanvasBase::CamaraParams&, int, int) const'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::setCameraProjective(bool)'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::setMouseClicked(bool)'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::setCameraPointing(float, float, float)'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::setCameraParams(mrpt::gui::CGlCanvasBase::CamaraParams const&)'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::updateZoom(mrpt::gui::CGlCanvasBase::CamaraParams&, int, int) const'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::updateRotate(mrpt::gui::CGlCanvasBase::CamaraParams&, int, int) const'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::updatePan(mrpt::gui::CGlCanvasBase::CamaraParams&, int, int) const'
../lib/libmrpt-gui.so.1.9.9: undefined reference to `mrpt::gui::CGlCanvasBase::cameraParams() const'
```

MariaRamos89, 2017-11-13:
I've tried also with `-DBUILD_mrpt-gui=OFF` and the result is totally different:

```shell
struct timespec tp;
                  ^
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:125:2: warning: implicit declaration of function ‘clock_gettime’ [-Wimplicit-function-declaration]
  clock_gettime(CLOCK_REALTIME, &tp); // compile with -lrt
  ^
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:125:16: error: ‘CLOCK_REALTIME’ undeclared (first use in this function)
  clock_gettime(CLOCK_REALTIME, &tp); // compile with -lrt
                ^
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:125:16: note: each undeclared identifier is reported only once for each function it appears in
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:128:3: warning: implicit declaration of function ‘localtime_r’ [-Wimplicit-function-declaration]
   localtime_r(&tp.tv_sec,date_);
   ^
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:124:18: warning: unused variable ‘tp’ [-Wunused-variable]
  struct timespec tp;
                  ^
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c: In function ‘XsTime_msleep’:
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:272:18: error: storage size of ‘ts’ isn’t known
  struct timespec ts;
                  ^
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:274:2: error: unknown type name ‘clockid_t’
  clockid_t clockid = CLOCK_MONOTONIC;
  ^
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:274:22: error: ‘CLOCK_MONOTONIC’ undeclared (first use in this function)
  clockid_t clockid = CLOCK_MONOTONIC;
                      ^
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:281:8: warning: implicit declaration of function ‘pthread_condattr_setclock’ [-Wimplicit-function-declaration]
  ret = pthread_condattr_setclock(&ca, clockid);
        ^
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:284:3: warning: implicit declaration of function ‘pthread_condattr_getclock’ [-Wimplicit-function-declaration]
   pthread_condattr_getclock(&ca, &clockid);
   ^
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:272:18: warning: unused variable ‘ts’ [-Wunused-variable]
  struct timespec ts;
                  ^
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c: In function ‘XsTime_udelay’:
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:338:18: error: storage size of ‘ts’ isn’t known
  struct timespec ts;
                  ^
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:340:2: error: unknown type name ‘clockid_t’
  clockid_t clockid = CLOCK_MONOTONIC;
  ^
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:340:22: error: ‘CLOCK_MONOTONIC’ undeclared (first use in this function)
  clockid_t clockid = CLOCK_MONOTONIC;
                      ^
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:338:18: warning: unused variable ‘ts’ [-Wunused-variable]
  struct timespec ts;
                  ^
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c: In function ‘XsTime_getTimeOfDay’:
/home/maria/.libs/mrpt/libs/hwdrivers/src/xSens_MT4/xstypes/src/xstime.c:136:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
libs/hwdrivers/CMakeFiles/mrpt-hwdrivers.dir/build.make:1349: recipe for target 'libs/hwdrivers/CMakeFiles/mrpt-hwdrivers.dir/src/xSens_MT4/xstypes/src/xstime.c.o' failed
make[2]: *** [libs/hwdrivers/CMakeFiles/mrpt-hwdrivers.dir/src/xSens_MT4/xstypes/src/xstime.c.o] Error 1
CMakeFiles/Makefile2:681: recipe for target 'libs/hwdrivers/CMakeFiles/mrpt-hwdrivers.dir/all' failed
make[1]: *** [libs/hwdrivers/CMakeFiles/mrpt-hwdrivers.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

jlblancoc, 2017-11-13:
@MariaRamos89 
Hi Maria, 
According to the shippable build matrix, MRPT master branch now builds cleanly **without the need to add any custom CMake flag** when OpenGL, GLUT, wxWidgets, Qt, etc. etc. are missing in the system. 

Please, give it a try with `git clean` and `git checkout master`, then rebuilding from scratch in a new empty build directory.

MariaRamos89, 2017-11-23:
Thanks @jlblancoc . It's working only using the `-DDISABLE_OPENCV=ON` flag in `master` branch.
Thanks for your help!

jolting, 2017-11-23:
If you omit that flag then how does it fail?

MariaRamos89, 2017-11-24:
No, it is not failing.
I just wanted to be sure in the first try that opencv 2.4 it wasn't installed.
It hasn't been installed in any case without the flag.
Thanks for your help! :)