MRPT/mrpt #602: Consider using CMake export

jlblancoc, 2017-10-15:
Think whether it's worth changing the current `MRPTConfig.cmake` to use CMake standard "EXPORT" command. 
* Pros: should make it simpler to work with static vs dynamic libs (without all the ad-hoc CMake code currently present).
* Cons: well... the current solution "just works", despite being less elegant...

Inspirational example [here](https://github.com/RainerKuemmerle/g2o/compare/Export_cmake_Configuration#diff-dfdf08900eb0292ea317b9f75ef332ed).

jolting, 2017-10-16:
I think using the cmake convention will prevent problems in the future and avoid some potential bit rot.

jlblancoc, 2017-11-17:
Another source of inspiration for a minimal working case [here](https://github.com/strasdat/Sophus/pull/120/files).

jlblancoc, 2018-11-07:
WIP in branch `cmake-config-targets`.

jlblancoc, 2018-11-11:
Reopening, since Debian pkgs are now broken:

```
 Install configuration: "Release"
CMake Error at cmake_install.cmake:49 (file):
  file INSTALL cannot find
  "/<<PKGBUILDDIR>>/obj-x86_64-linux-gnu/unix-install/MRPTConfig.cmake".

make[2]: *** [Makefile:65: install] Error 1
```

jlblancoc, 2018-11-11:
Fixed by cbc00a7f1058a5874bead8ec42a2a22338a2a20f

PR #859 (2018-11-11): Use modern CMake exported targets
This closes #602. 
Note that the update breaks compatibility with existing code finding MRPT (pre 2.0) like:

```
find_package(MRPT REQUIRED slam vision [...])
...
target_link_libraries(mytarget  ${MRPT_LIBS})
```

which should now be updated to:

```
# Find MRPT libraries:
find_package(mrpt-poses)
find_package(mrpt-gui)

# Define your own targets:
add_executable(myapp  main.cpp)

# Link against MRPT: this will also add all required flags,
# include directories, etc.
# Note: the namespace prefix `mrpt::` is optional.
target_link_libraries(myapp
  mrpt::mrpt-po [... cut]
commit 0c0fe72c (2018-11-11): Merge pull request #859 from MRPT/cmake-config-targets

Use modern CMake exported targets. This closes #602.
Note that the update breaks compatibility with existing code finding MRPT (pre 2.0). 
Read more on #859
closed 2018-11-11 by jlblancoc