MRPT/mrpt #615: RFC: refactoring of mrpt-base

jlblancoc, 2017-11-19:
Motivated by:
* [this](https://github.com/MRPT/mrpt/commit/3aec720fe5205bfe5d974580a29360c923c523e6#commitcomment-25702556) MSCV linker limitation.
* because it's true that `mrpt-base` is really large and includes too many functionalities,  and
* because of the potential utility of having many smaller `libmrpt-XXX-dev` packages (see #588 )

we want to split mrpt-base into a more logical set of smaller modules.

WIP happens in the [dev-module-refactor](https://github.com/MRPT/mrpt/tree/dev-module-refactor) branch.

Ideas for new module distribution:
* [ ] `mrpt-macros`: Mainly preprocessor macros and bottom-line, basic functionality; e.g. `mrpt::format()`. Should have a ridiculously small footprint since will be included in all other modules.
* [ ] `mrpt-system`: contents of namespace `mrpt-system`.
* [ ] split serialization and RTTI classes into `mrpt-io` (streams, files, TTypeName, STL serialization, etc.). It would be great to split CStream stuff and RTTI/serialization into TWO different module, but I still don't see a clear way to do it (needs more thinking. Any ideas?).
* [ ] `mrpt-math` all math classes (idea needs further refining)
* [ ] The rest, remains into the thinner `mrpt-base`, which *could* be renamed `mrpt-core`, with `mrpt-base` a virtual dependency to all underlying modules for backwards compatibility.

jolting, 2017-11-19:
I wonder how many symbols belong to eigen. I think those are exported when statically linking mrpt dlls against eigen. That might be a huge source of symbols.

jlblancoc, 2017-11-20:
> I wonder how many symbols belong to eigen. 

Another good reason to experiment with implementing #496 in this new branch... ;-)

jlblancoc, 2017-11-20:
FYI: The latest graph of module deps for this branch is [here](http://mrpt.ual.es/reference/dev-module-refactor/dot_inline_dotgraph_2.png) (rebuilt after each commit):

[image]

jlblancoc, 2017-11-20:
I intend to make mrpt-random, and the new modules for "typenames" (RTTI) and serialization, as independent of MRPT as possible. Ideally, not even depending on "mrpt-core", so anyway can install them via `apt install libmrpt-xxx-dev` and use as standalone libs.
That's the intention, at least... (sigh)

jlblancoc, 2017-11-23:
BTW:  current approach in mrpt-typemeta, together with the yet-to-be mrpt-serialization, will finally solve #14

jlblancoc, 2017-11-26:
Updated the first comment with a list of checkboxes to reflect progress in this task.

jlblancoc, 2017-12-04:
Temporarily created branch [master-ref-point-before-refactoring](https://github.com/MRPT/mrpt/tree/master-ref-point-before-refactoring) based on master, with shippable all green, as a reference point to compare build times after the refactoring of this and the #496 issues, to quantify the gain of the changes.

jlblancoc, 2018-01-10:
This huge task is now 100% done, at the level of mrpt libraries and applications... Yay!
I'm opening a PR to integrate the huge number of changes in master, then will continue with the few remaining colateral issues: fixing examples & documentation, mainly.

jlblancoc, 2018-02-14:
3 months later... this big one is done after #648, Yay!

closed 2018-02-14 by jlblancoc