MRPT/mrpt #619: Particle filter resampling error (mrpt 1.5.3, 1.5.4) for RBPF-Slam

andreasrichtsfeld, 2017-11-27:
**Hi,**

The particle filter resampling delivers an error when it gets triggered. AFAIK resampling is triggered when ESS<BETA. This will not happen for the demos when a `powFactor=0.001` is defined, because ESS will be always >0.99. Changeing `powFactor=0.1` will cause lower ESS values and will then trigger the resampling, when `ESS<0.5`.

**Demo program started:**
    ./rbpf-slam ../../share/mrpt/config_files/rbpf-slam/gridmapping_intel.ini ../../share/mrpt/datasets/2006-01ENE-21-SENA_Telecom\ Faculty_one_loop_only.rawlog

**Error Message**: 

[CMetricMapBuilderRBPF|INFO |11:37:41.9310] New pose=(x,y,z,yaw,pitch,roll)=(1.7133,-1.4072,0.0000,-64.71deg,-0.00deg,0.00deg)
New ESS:0.484034
[CMetricMapBuilderRBPF|INFO |11:37:41.9311]    STDs: x=0.029 y=0.030 z=0.000 yaw=0.360deg
[CMetricMapBuilderRBPF|INFO |11:37:41.9311] New observation inserted into the map.
Map building executed in 879.084ms
Saving memory usage...Ok! (1053.6406Mb)

---------------- STEP 33 | RAWLOG ENTRY 67 ----------------
[CMetricMapBuilderRBPF|DEBUG|11:37:41.9471] processActionObservation(): Input action is CActionRobotMovement2D
[CMetricMapBuilderRBPF|DEBUG|11:37:41.9472] do_map_update=NO do_localization=NO
Map building executed in 0.064ms
Saving memory usage...Ok! (1053.6406Mb)

---------------- STEP 34 | RAWLOG ENTRY 69 ----------------
[CMetricMapBuilderRBPF|DEBUG|11:37:41.9473] processActionObservation(): Input action is CActionRobotMovement2D
[CMetricMapBuilderRBPF|DEBUG|11:37:41.9473] do_map_update=NO do_localization=NO
Map building executed in 0.040ms
Saving memory usage...Ok! (1053.6406Mb)

---------------- STEP 35 | RAWLOG ENTRY 71 ----------------
[CMetricMapBuilderRBPF|DEBUG|11:37:41.9474] processActionObservation(): Input action is CActionRobotMovement2D
[CMetricMapBuilderRBPF|DEBUG|11:37:41.9475] do_map_update=YES do_localization=YES
[CMetricMapBuilderRBPF|DEBUG|11:37:41.9475] odoIncrementSinceLastLocalization before resetting = (x,y,z,yaw,pitch,roll)=(0.7214,-0.0454,0.0000,-6.61deg,0.00deg,0.00deg)
[CMetricMapBuilderRBPF|DEBUG|11:37:42.5684] CMetricMapBuilder dtor.
dereferencing NULL generic_copier_ptr /home/ari/projects_offline/mrpt-1.5.4/libs/slam/include/mrpt/slam/PF_implementations.h:835: In `void mrpt::slam::PF_implementation<PARTICLE_TYPE, MYSELF>::PF_SLAM_implementation_pfAuxiliaryPFStandardAndOptimal(const mrpt::obs::CActionCollection*, const mrpt::obs::CSensoryFrame*, const mrpt::bayes::CParticleFilter::TParticleFilterOptions&, const mrpt::slam::TKLDParams&, bool) [with BINTYPE = mrpt::slam::detail::TPoseBin2D; PARTICLE_TYPE = mrpt::maps::CRBPFParticleData; MYSELF = mrpt::maps::CMultiMetricMapPDF]`
 /home/ari/projects_offline/mrpt-1.5.4/libs/slam/src/slam/CMultiMetricMapPDF_RBPF.cpp:148: In `virtual void mrpt::maps::CMultiMetricMapPDF::prediction_and_update_pfAuxiliaryPFOptimal(const mrpt::obs::CActionCollection*, const mrpt::obs::CSensoryFrame*, const mrpt::bayes::CParticleFilter::TParticleFilterOptions&)`
 /home/ari/projects_offline/mrpt-1.5.4/libs/base/src/bayes/CParticleFilter.cpp:99: In `void mrpt::bayes::CParticleFilter::executeOn(mrpt::bayes::CParticleFilterCapable&, const mrpt::obs::CActionCollection*, const mrpt::obs::CSensoryFrame*, mrpt::bayes::CParticleFilter::TParticleFilterStats*)`
 /home/ari/projects_offline/mrpt-1.5.4/libs/slam/src/slam/CMetricMapBuilderRBPF.cpp:248: In `virtual void mrpt::slam::CMetricMapBuilderRBPF::processActionObservation(mrpt::obs::CActionCollection&, mrpt::obs::CSensoryFrame&)`
 /home/ari/projects_offline/mrpt-1.5.4/apps/rbpf-slam/rbpf-slam.cpp:654: In `void MapBuilding_RBPF()`

Program finished for an exception!!
Press any key to continue...

**For me it looks like resampling is not working since a longer time. The last version that I used was 1.3.1 which worked well with resampling**

Thanks for any help!
Andi

jlblancoc, 2017-11-27:
Thanks for the detailed report!

I'm almost sure that the error is caused by a regression, which was already fixed in the mrpt-1.5 branch with this commit: https://github.com/MRPT/mrpt/commit/4c9eb01fc71fa42093bd549d981973c1b5fa80ec

Please, if possible, try it with the latest version in the mrpt-1.5 branch. It will help me get confident towards the imminent release of 1.5.5!

andreasrichtsfeld, 2017-11-28:
Yes, this commit fixed the problem! 

Thanks,
Andi

jlblancoc, 2017-11-28:
Good!

closed 2017-11-28 by jlblancoc