MRPT/mrpt #626: missing libassimp.so.4

OtacilioNeto, 2017-12-07:
After build and install mrpt-1.5.5 on Ubuntu 16.04 I got this error messagem when run camera-calib:

camera-calib: error while loading shared libraries: libassimp.so.4: cannot open shared object file: No such file or directory

I found libassimp.so.4 into lib in build dir.

jolting, 2017-12-07:
I don't think there is an install rule for libassimp.so. If you use the system version of libassimp you won't need to use the one in the build dir. To use the system version `apt install libassimp-dev`, rebuild and reinstall. Let me know if that works.

OtacilioNeto, 2017-12-07:
You are right. It works.
Thanks

closed 2017-12-07 by OtacilioNeto

jlblancoc, 2017-12-07:
Hi, 

Anyway, I found this to also affect Windows binary packages (sigh...), so it might have to be fixed just for that case, or precompiled binaries regenerated *without* support for assimp. The latter should not be a big deal, so I'll probably just go for it.