MRPT/mrpt #627: camera-calib GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported

OtacilioNeto, 2017-12-07:
On Ubuntu 16.04 camera-calib is showing this error message:

(camera-calib:31149): Gtk-ERROR **: GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported

Looks like a config problem?

alibenD, 2017-12-08:
I also have this problem when running ANY mrpt code that uses wxwidget (for the GUI) on DebiAN 9 (not only when using the camera-calib)

jlblancoc, 2017-12-09:
Can you please provide this info?

1) did you build MRPT from sources? which MRPT version/git branch? 

2) Cmake-found opencv version?
From the directory in which you built mrpt:
```
grep OpenCV_DIR CMakeCache.txt
``` 

3) run ldd for any of the binaries that gives you the error:

`ldd camera-calib | grep gtk`

OtacilioNeto, 2017-12-09:
Hi Professor Blanco

1. Yes I build MRPT from sources, version 1.5.5 downloaded from https://github.com/MRPT/mrpt/releases

2. Yes:
ota@rigel:~/src/mrpt/build$ cat CMakeCache.txt | grep OpenCV_DIR
OpenCV_DIR:PATH=/usr/local/share/OpenCV

This one is from the directory where I build Opencv 3
ota@rigel:~/src/opencv/build$ cat CMakeCache.txt | grep GTK
//Include GTK support
WITH_GTK:BOOL=ON
//Use GTK version 2
WITH_GTK_2_X:BOOL=OFF

3. ota@rigel:~/src/opencv/build$ ldd /usr/local/bin/camera-calib | grep gtk
	libwx_gtk2u_core-2.8.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-2.8.so.0 (0x00007f4ff7693000)
	libwx_gtk2u_gl-2.8.so.0 => /usr/lib/x86_64-linux-gnu/libwx_gtk2u_gl-2.8.so.0 (0x00007f4ff7485000)
	libgtk-x11-2.0.so.0 => /usr/lib/x86_64-linux-gnu/libgtk-x11-2.0.so.0 (0x00007f4ff3c75000)
	libgtk-3.so.0 => /usr/lib/x86_64-linux-gnu/libgtk-3.so.0 (0x00007f4fd8400000)

jlblancoc, 2017-12-09:
>This one is from the directory where I build Opencv 3
>ota@rigel:~/src/opencv/build$ cat CMakeCache.txt | grep GTK
>//Include GTK support
>WITH_GTK:BOOL=ON
>//Use GTK version 2
>WITH_GTK_2_X:BOOL=OFF

That **seems** to be causing the problem... could you try rebuilding opencv, setting it via CMake so it uses GTK2? 
I don't know what's different in the packages provided via the ROS repositories, but I have opencv3.2 and it's built **not** to use gtk3, that's why it didn't cause me any problem. 

Anyway, this might be a bug in opencv (?) CMake configuration, because I don't know why an mrpt app is linked against libgtk-3.so.0... it might make sense if you built opencv as static libs, did you?

Other option is that the libgtk-3 dependency enters via another package different than opencv. To help identifying it, try this: 

```bash
cd [MRPT_BUILD_DIR]
rm CMakeCache.txt
VERBOSE=1 cmake [MRPT_SOURCE_DIR]  > cmake.log
```

and attach the resulting log file to see where does the gtk3 come from.

OtacilioNeto, 2017-12-10:
After turn WITH_GTK_2_X:BOOL=ON on OpenCV3 configuration, and rebuild and install Opencv3 and rebuild and install mrpt the core dump goes out.
So far I know my OpenCV 3 is not linked static, but it was compiled with CUDA enabled.

OtacilioNeto, 2017-12-10:
I turned  off WITH_GTK_2_X:BOOL and rebuild and install only Opencv3 and the problem returns to camera-calib without rebuild mrpt. For me looks like a configuration problem with OpenCV3 because when WITH_GTK_2_X:BOOL is ON I got a error message on Opencv apps clamming that windows suport is not enabled.

alibenD, 2017-12-15:
Hi professor Blanco, thank your for answer.

The problem has been solved just following your instructions, building `opencv` with `WITH_GTK_2_X` again. One more question, could we build `MRPT` with `GTK3.x`? I didn't find the option in `CMakeList.txt`, I was wondering if you can tell me how to link `MRPT` with `GTK3` libraries, if you were free. Many thanks!

jlblancoc, 2017-12-15:
Hi,

It's not mrpt, but wxWidgets the one that needs to be built against gtk3...
take a look at their documentation/github to see if that's possible.

Cheers

alibenD, 2017-12-16:
Many thanks, I will try.

closed 2018-01-10 by jlblancoc

maxbader, 2018-07-11:
I just had the same issue with Ubuntu 18.04 Bionic Beaver. Installing wxWidget using the the following wxWidgets repository solved it for me.

> sudo apt-add-repository 'deb http://repos.codelite.org/wx3.1.1/ubuntu/ bionic universe'
> sudo apt-key adv --fetch-keys http://repos.codelite.org/CodeLite.asc
> sudo apt update
> sudo apt-get install libwxbase3.1-0-unofficial3 \
>                  libwxbase3.1unofficial3-dev \
>                  libwxgtk3.1-0-unofficial3 \
>                  libwxgtk3.1unofficial3-dev \
>                  wx3.1-headers \
>                  wx-common \
>                  libwxgtk-media3.1-0-unofficial3 \
>                  libwxgtk-media3.1unofficial3-dev \
>                  libwxgtk-webview3.1-0-unofficial3 \
>                  libwxgtk-webview3.1unofficial3-dev \
>                  libwxbase3.1-0-unofficial3-dbg \
>                  libwxgtk3.1-0-unofficial3-dbg \
>                  libwxgtk-webview3.1-0-unofficial3-dbg \
>                  libwxgtk-media3.1-0-unofficial3-dbg \
>                  wx3.1-i18n \
>                  wx3.1-examples

jolting, 2018-07-11:
Unfortunately, supporting both GTK2 and GTK3 is tricky. Essentially if you want to switch on your build machine between the two you have to update-alternatives wx-config. Ugh.