MRPT/mrpt #629: Decode Error for LMS151 using CLMS100Eth

alibenD, 2017-12-20:
Problem description:

Having a physical device `SICK_LMS_151`, I met a problem that when one maybe two times scan data come back, an error appears. 

```
[C2DRangeFinderAbstract|DEBUG|10:48:46.7130] read : 21
[C2DRangeFinderAbstract|DEBUG|10:48:46.7132] message : sAN SetAccessMode 1
[C2DRangeFinderAbstract|DEBUG|10:48:47.8285] read : 48
[C2DRangeFinderAbstract|DEBUG|10:48:47.8285] message : sAN mLMPsetscancfg 0 9C4 1 9C4 FFF92230 225510
[C2DRangeFinderAbstract|DEBUG|10:48:48.8317] read : 20
[C2DRangeFinderAbstract|DEBUG|10:48:48.8317] message : sWA LMDscandatacfg
[C2DRangeFinderAbstract|DEBUG|10:48:49.8457] message : sAN LMCstartmeas 0
[C2DRangeFinderAbstract|DEBUG|10:49:00.8495] message : sRA STlms 3 0 8 00:17:28 A 01.01.1970 0 0 0
[C2DRangeFinderAbstract|DEBUG|10:49:00.8496] 3
[C2DRangeFinderAbstract|DEBUG|10:49:11.8530] message : sRA STlms 3 0 8 00:17:39 A 01.01.1970 0 0 0
[C2DRangeFinderAbstract|DEBUG|10:49:11.8531] 3
[C2DRangeFinderAbstract|DEBUG|10:49:22.8560] message : sRA STlms 4 0 8 00:17:50 A 01.01.1970 0 0 0
[C2DRangeFinderAbstract|DEBUG|10:49:22.8561] 4
[C2DRangeFinderAbstract|DEBUG|10:49:33.8609] message : sRA STlms 4 0 8 00:18:01 A 01.01.1970 0 0 0
[C2DRangeFinderAbstract|DEBUG|10:49:33.8610] 4
[C2DRangeFinderAbstract|DEBUG|10:49:44.8643] message : sRA STlms 4 0 8 00:18:12 A 01.01.1970 0 0 0
[C2DRangeFinderAbstract|DEBUG|10:49:44.8644] 4
[C2DRangeFinderAbstract|DEBUG|10:49:55.8678] message : sRA STlms 7 0 8 00:18:23 A 01.01.1970 0 0 0
[C2DRangeFinderAbstract|DEBUG|10:49:55.8678] 7
**[C2DRangeFinderAbstract|ERROR|10:49:55.9111] STATUS error on LMS100: '1'**
[C2DRangeFinderAbstract|DEBUG|10:49:55.9112] Distance : OK
[C2DRangeFinderAbstract|DEBUG|10:49:55.9112] Scan Count : 541
```

```
Press any key to exit program
[C2DRangeFinderAbstract|DEBUG|10:55:52.1582] read : 21
[C2DRangeFinderAbstract|DEBUG|10:55:52.1583] message : sAN SetAccessMode 1
[C2DRangeFinderAbstract|DEBUG|10:55:53.2786] read : 48
[C2DRangeFinderAbstract|DEBUG|10:55:53.2787] message : sAN mLMPsetscancfg 0 9C4 1 9C4 FFF92230 225510
[C2DRangeFinderAbstract|DEBUG|10:55:54.2815] read : 20
[C2DRangeFinderAbstract|DEBUG|10:55:54.2816] message : sWA LMDscandatacfg
[C2DRangeFinderAbstract|DEBUG|10:55:55.2954] message : sAN LMCstartmeas 0
[C2DRangeFinderAbstract|DEBUG|10:56:06.2980] message : sRA STlms 3 0 8 00:24:34 A 01.01.1970 0 0 0
[C2DRangeFinderAbstract|DEBUG|10:56:06.2981] 3
[C2DRangeFinderAbstract|DEBUG|10:56:17.3020] message : sRA STlms 3 0 8 00:24:45 A 01.01.1970 0 0 0
[C2DRangeFinderAbstract|DEBUG|10:56:17.3021] 3
[C2DRangeFinderAbstract|DEBUG|10:56:28.3074] message : sRA STlms 4 0 8 00:24:56 A 01.01.1970 0 0 0
[C2DRangeFinderAbstract|DEBUG|10:56:28.3075] 4
[C2DRangeFinderAbstract|DEBUG|10:56:39.3111] message : sRA STlms 4 0 8 00:25:07 A 01.01.1970 0 0 0
[C2DRangeFinderAbstract|DEBUG|10:56:39.3112] 4
[C2DRangeFinderAbstract|DEBUG|10:56:50.3144] message : sRA STlms 7 0 8 00:25:18 A 01.01.1970 0 0 0
[C2DRangeFinderAbstract|DEBUG|10:56:50.3145] 7
**[C2DRangeFinderAbstract|ERROR|10:56:50.3586] STATUS error on LMS100: '1'**
[C2DRangeFinderAbstract|DEBUG|10:56:50.3587] Distance : OK
[C2DRangeFinderAbstract|DEBUG|10:56:50.3588] Scan Count : 1081
**[C2DRangeFinderAbstract|ERROR|10:56:50.4019] STATUS error on LMS100: '1'**
[C2DRangeFinderAbstract|DEBUG|10:56:50.4020] Distance : OK
[C2DRangeFinderAbstract|DEBUG|10:56:50.4020] Scan Count : 1081
**[C2DRangeFinderAbstract|ERROR|10:56:50.4449] STATUS error on LMS100: '1'**
```

alibenD, 2017-12-20:
I found that there has a little bit different for decoding between LMS100 and LMS151. So I would write a driver for sick_lms151.

jlblancoc, 2018-04-04:
Closing due to lack of activity. Also, I think there was a future PR going on related to this issue.

closed 2018-04-04 by jlblancoc