MRPT/mrpt #633: Regression in master: mem leaks in particle filters?

jlblancoc, 2018-01-17:
Building with GCC sanitizers:

```
CMAKE_CXX_FLAGS:STRING= -fsanitize=address  -fsanitize=leak -g
CMAKE_C_FLAGS:STRING=-fsanitize=address  -fsanitize=leak -g
CMAKE_EXE_LINKER_FLAGS:STRING=-fsanitize=address  -fsanitize=leak
CMAKE_MODULE_LINKER_FLAGS:STRING=-fsanitize=address  -fsanitize=leak
```

I found these memory leaks: 

```
Direct leak of 13312 byte(s) in 64 object(s) allocated from:
    #0 0x7fe9dd4ea138 in __interceptor_aligned_alloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xdf138)
    #1 0x7fe9dabce70d in mrpt::poses::CPose3DPDFParticles::operator new(unsigned long) /home/jlblanco/code/mrpt/libs/poses/include/mrpt/poses/CPose3DPDFParticles.h:39
    #2 0x7fe9dabce70d in mrpt::poses::CPose3DPDFParticles::clone() const /home/jlblanco/code/mrpt/libs/poses/src/CPose3DPDFParticles.cpp:24
    #3 0xfff9116b405  (<unknown module>)
    #4 0x61500000287f  (<unknown module>)

....

Indirect leak of 230400 byte(s) in 1600 object(s) allocated from:
    #0 0x7fe9dd4ea138 in __interceptor_aligned_alloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xdf138)
    #1 0x7fe9dabcedbb in mrpt::poses::CPose3D::operator new(unsigned long) /home/jlblanco/code/mrpt/libs/poses/include/mrpt/poses/CPose3D.h:90
    #2 0x7fe9dabcedbb in mrpt::containers::internal::CopyStatic<mrpt::poses::CPose3D>::copy(mrpt::poses::CPose3D const*) /home/jlblanco/code/mrpt/libs/containers/include/mrpt/containers/generic_copier_ptr.h:29
    #3 0x7fe9dabcedbb in mrpt::containers::internal::generic_copier_ptr<mrpt::poses::CPose3D, mrpt::containers::internal::CopyStatic<mrpt::poses::CPose3D> >::generic_copier_ptr(mrpt::containers::internal::generic_copier_ptr<mrpt::poses::CPose3D, mrpt::containers::internal::CopyStatic<mrpt::poses::CPose3D> > const&) /home/jlblanco/code/mrpt/libs/containers/include/mrpt/containers/generic_copier_ptr.h:58
    #4 0x7fe9dabcedbb in mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>::CProbabilityParticle(mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const&) /home/jlblanco/code/mrpt/libs/bayes/include/mrpt/bayes/CProbabilityParticle.h:28
    #5 0x7fe9dabcedbb in void std::_Construct<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const&>(mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>*, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const&) /usr/include/c++/7/bits/stl_construct.h:75
    #6 0x7fe9dabcedbb in std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>*> std::__uninitialized_copy<false>::__uninit_copy<std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const*>, std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>*> >(std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const*>, std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const*>, std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>*>) /usr/include/c++/7/bits/stl_uninitialized.h:83
    #7 0x7fe9dabcedbb in std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>*> std::uninitialized_copy<std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const*>, std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>*> >(std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const*>, std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const*>, std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>*>) /usr/include/c++/7/bits/stl_uninitialized.h:134
    #8 0x7fe9dabcedbb in std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>*> std::__uninitialized_copy_a<std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const*>, std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>*>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> >(std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const*>, std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> const*>, std::_Deque_iterator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>&, mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>*>, std::allocator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> >&) /usr/include/c++/7/bits/stl_uninitialized.h:289
    #9 0x7fe9dabcedbb in std::deque<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, std::allocator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> > >::deque(std::deque<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D>, std::allocator<mrpt::bayes::CProbabilityParticle<mrpt::poses::CPose3D> > > const&) /usr/include/c++/7/bits/stl_deque.h:950
    #10 0x7fe9dabcedbb in mrpt::bayes::CParticleFilterData<mrpt::poses::CPose3D>::CParticleFilterData(mrpt::bayes::CParticleFilterData<mrpt::poses::CPose3D> const&) /home/jlblanco/code/mrpt/libs/bayes/include/mrpt/bayes/CParticleFilterData.h:197
    #11 0x7fe9dabcedbb in mrpt::poses::CPose3DPDFParticles::CPose3DPDFParticles(mrpt::poses::CPose3DPDFParticles const&) /home/jlblanco/code/mrpt/libs/poses/include/mrpt/poses/CPose3DPDFParticles.h:32
    #12 0x7fe9dabcedbb in mrpt::poses::CPose3DPDFParticles::clone() const /home/jlblanco/code/mrpt/libs/poses/src/CPose3DPDFParticles.cpp:24
```

which doesn't happen in mrpt-1.5. 

This must be investigated...

jolting, 2018-04-12:
Here's some additional output from rbpf-slam. The wx stuff can probably be ignored.
```
==32765==ERROR: LeakSanitizer: detected memory leaks

Direct leak of 19968 byte(s) in 96 object(s) allocated from:
    #0 0x7f08290cc368 in __interceptor_aligned_alloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xdf368)
    #1 0x7f082198fa9e in mrpt::aligned_malloc(unsigned long, unsigned long) /home/hlaux/code/mrpt/libs/core/src/aligned_malloc.cpp:28
    #2 0x7f082495945e in mrpt::poses::CPose3DPDFParticles::operator new(unsigned long) /home/hlaux/code/mrpt/libs/poses/include/mrpt/poses/CPose3DPDFParticles.h:39
    #3 0x7f0824a3055a in mrpt::poses::CPose3DPDFParticles::clone() const /home/hlaux/code/mrpt/libs/poses/src/CPose3DPDFParticles.cpp:24
    #4 0x7f08267de9ac in operator()<std::pair<std::shared_ptr<mrpt::poses::CPose3DPDF>, std::shared_ptr<mrpt::obs::CSensoryFrame> > > /home/hlaux/code/mrpt/libs/obs/src/CSimpleMap.cpp:31
    #5 0x7f08267dea26 in for_each<std::_Deque_iterator<std::pair<std::shared_ptr<mrpt::poses::CPose3DPDF>, std::shared_ptr<mrpt::obs::CSensoryFrame> >, std::pair<std::shared_ptr<mrpt::poses::CPose3DPDF>, std::shared_ptr<mrpt::obs::CSensoryFrame> >&, std::pair<std::shared_ptr<mrpt::poses::CPose3DPDF>, std::shared_ptr<mrpt::obs::CSensoryFrame> >*>, <lambda(auto:3&)> > /usr/include/c++/7/bits/stl_algo.h:3884
    #6 0x7f08267d8f81 in mrpt::maps::CSimpleMap::operator=(mrpt::maps::CSimpleMap const&) /home/hlaux/code/mrpt/libs/obs/src/CSimpleMap.cpp:47
    #7 0x7f082875c885 in mrpt::slam::CMetricMapBuilderRBPF::getCurrentlyBuiltMap(mrpt::maps::CSimpleMap&) const /home/hlaux/code/mrpt/libs/slam/src/slam/CMetricMapBuilderRBPF.cpp:362
    #8 0x5605a5186e1e in MapBuilding_RBPF() /home/hlaux/code/mrpt/apps/rbpf-slam/rbpf-slam.cpp:686
    #9 0x5605a517f586 in main /home/hlaux/code/mrpt/apps/rbpf-slam/rbpf-slam.cpp:198
    #10 0x7f0820a5db96 in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x21b96)

Direct leak of 4352 byte(s) in 17 object(s) allocated from:
    #0 0x7f08290cbb50 in __interceptor_malloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xdeb50)
    #1 0x7f080e4648ed  (/usr/lib/x86_64-linux-gnu/libfontconfig.so.1+0x1d8ed)

Direct leak of 496 byte(s) in 1 object(s) allocated from:
    #0 0x7f08290cd458 in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xe0458)
    #1 0x7f081d73abe3 in wxInitAllImageHandlers() (/usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0+0x39dbe3)
    #2 0x7f08279e449c in CDisplayWindow_WXAPP::OnInit() /home/hlaux/code/mrpt/libs/gui/src/WxSubsystem.cpp:879
    #3 0x7f08279e6b8b in wxAppConsoleBase::CallOnInit() /usr/include/wx-3.0/wx/app.h:93
    #4 0x7f08279e4a0b in mrpt_wxEntryReal(int, char**) /home/hlaux/code/mrpt/libs/gui/src/WxSubsystem.cpp:988
    #5 0x7f08279e4d64 in mrpt::gui::WxSubsystem::wxMainThread() /home/hlaux/code/mrpt/libs/gui/src/WxSubsystem.cpp:1045
    #6 0x7f08279e98c8 in void std::__invoke_impl<void, void (*)()>(std::__invoke_other, void (*&&)()) /usr/include/c++/7/bits/invoke.h:60
    #7 0x7f08279e881f in std::__invoke_result<void (*)()>::type std::__invoke<void (*)()>(void (*&&)()) /usr/include/c++/7/bits/invoke.h:95
    #8 0x7f08279ef28b in decltype (__invoke((_S_declval<0ul>)())) std::thread::_Invoker<std::tuple<void (*)()> >::_M_invoke<0ul>(std::_Index_tuple<0ul>) /usr/include/c++/7/thread:234
    #9 0x7f08279ef210 in std::thread::_Invoker<std::tuple<void (*)()> >::operator()() /usr/include/c++/7/thread:243
    #10 0x7f08279ef171 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (*)()> > >::_M_run() /usr/include/c++/7/thread:186
    #11 0x7f08216c0472  (/usr/lib/x86_64-linux-gnu/libstdc++.so.6+0xbe472)

Direct leak of 416 byte(s) in 2 object(s) allocated from:
    #0 0x7f08290cbd38 in __interceptor_calloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xded38)
    #1 0x7f07eb8b2ab8  (/usr/lib/x86_64-linux-gnu/dri/i965_dri.so+0x418ab8)

Direct leak of 200 byte(s) in 1 object(s) allocated from:
    #0 0x7f08290cd458 in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xe0458)
    #1 0x7f081d73b331 in wxInitAllImageHandlers() (/usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so.0+0x39e331)
    #2 0x7f08279e449c in CDisplayWindow_WXAPP::OnInit() /home/hlaux/code/mrpt/libs/gui/src/WxSubsystem.cpp:879
    #3 0x7f08279e6b8b in wxAppConsoleBase::CallOnInit() /usr/include/wx-3.0/wx/app.h:93
    #4 0x7f08279e4a0b in mrpt_wxEntryReal(int, char**) /home/hlaux/code/mrpt/libs/gui/src/WxSubsystem.cpp:988
    #5 0x7f08279e4d64 in mrpt::gui::WxSubsystem::wxMainThread() /home/hlaux/code/mrpt/libs/gui/src/WxSubsystem.cpp:1045
    #6 0x7f08279e98c8 in void std::__invoke_impl<void, void (*)()>(std::__invoke_other, void (*&&)()) /usr/include/c++/7/bits/invoke.h:60
    #7 0x7f08279e881f in std::__invoke_result<void (*)()>::type std::__invoke<void (*)()>(void (*&&)()) /usr/include/c++/7/bits/invoke.h:95
    #8 0x7f08279ef28b in decltype (__invoke((_S_declval<0ul>)())) std::thread::_Invoker<std::tuple<void (*)()> >::_M_invoke<0ul>(std::_Index_tuple<0ul>) /usr/include/c++/7/thread:234
    #9 0x7f08279ef210 in std::thread::_Invoker<std::tuple<void (*)()> >::operator()() /usr/include/c++/7/thread:243
    #10 0x7f08279ef171 in std::thread::_State_impl<std::thread::_Invoker<std::tuple<void (*)()> > >::_M_run() /usr/include/c++/7/thread:186
    #11 0x7f08216c0472  (/usr/lib/x86_64-linux-gnu/libstdc++.so.6+0xbe472)

Direct leak of 200 byte(s) in 1 object(s) allocated from:
[... cut ...]
Indirect leak of 8 byte(s) in 1 object(s) allocated from:
    #0 0x7f08290cbd38 in __interceptor_calloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xded38)
    #1 0x7f07eb8b807f  (/usr/lib/x86_64-linux-gnu/dri/i965_dri.so+0x41e07f)

Indirect leak of 8 byte(s) in 1 object(s) allocated from:
    #0 0x7f08290cbd38 in __interceptor_calloc (/usr/lib/x86_64-linux-gnu/libasan.so.4+0xded38)
    #1 0x7f07eb6be030  (/usr/lib/x86_64-linux-gnu/dri/i965_dri.so+0x224030)

SUMMARY: AddressSanitizer: 231621 byte(s) leaked in 1417 allocation(s).
```

jolting, 2018-04-13:
I think this can be closed now.

closed 2018-04-13 by jolting