MRPT/mrpt #637: LibTIFF error

siddadd, 2018-01-27:
I am trying to build mrpt on Ubuntu 16.04.2 and am getting this error

/usr/lib/x86_64-linux-gnu/libwx_gtk2u_core-3.0.so: undefined reference to `TIFFReadDirectory@LIBTIFF_4.0'

I tried to create a soft link for libtiff4 from libtiff5 but to no avail. Is there any work around?

jolting, 2018-01-28:
It looks like a problem with libwxgtk3.0-0v5, but I haven't seen your issue before.

Have you tried a fresh install of your build environment? Can you describe how you set up your environment? Finally, can you post a complete log?

jlblancoc, 2018-01-31:
I neither ever found this error, despite mrpt has been tested on many different archs and Ubuntu versions. 

Seems like you are using a custom version (PPA? manually built?) of wxWidgets which is causing the problem, right? 

Will close this one unless more details are provided.

closed 2018-01-31 by jlblancoc