MRPT/mrpt #644: Sophus license is missing

richmattes, 2018-02-10:
The bundled Sophus code is licensed under the MIT license, which stipulates that the copyright notice and permissions shall be included with all copies or substantial portions of the software.

Please include a copy of the Sophus LICENSE.txt with the Sophus source.

jolting, 2018-02-10:
I added to LICENSE file to the sophus folder.
Just a reminder for myself. The debian packaging needs to be updated too.

jolting, 2018-02-10:
@richmattes Thanks for bringing this to my attention. I take licensing very seriously.

I've added the LICENSE.txt file to the 1.5 and 2.0(master) branches and updated the debian copyright files to provide proper attribution. Our next releases should be compliant with the licensing requirement.

closed 2018-02-10 by jolting

jlblancoc, 2018-02-10:
Thanks @richmattes for noticing and reporting!

richmattes, 2018-02-10:
You're welcome, glad to help!