MRPT/mrpt #665: Support AVX Eigen alignments

jolting, 2018-02-28:
I ran into an exception in test_mrpt_graphslam. H_map was misaligned in levmarq.h.
Further investigation showed that Eigen was using 32-byte alignment.

I found this.
```
32 bytes alignment if AVX is enabled.
64 bytes alignment if AVX512 is enabled.
```
- https://eigen.tuxfamily.org/dox/classEigen_1_1aligned__allocator.html

Here are some defines:
```
// If the user explicitly disable vectorization, then we also disable alignment
#if defined(EIGEN_DONT_VECTORIZE)
  #define EIGEN_IDEAL_MAX_ALIGN_BYTES 0
#elif defined(EIGEN_VECTORIZE_AVX512)
  // 64 bytes static alignmeent is preferred only if really required
  #define EIGEN_IDEAL_MAX_ALIGN_BYTES 64
#elif defined(__AVX__)
  // 32 bytes static alignmeent is preferred only if really required
  #define EIGEN_IDEAL_MAX_ALIGN_BYTES 32
#else
  #define EIGEN_IDEAL_MAX_ALIGN_BYTES 16
#endif
```

The aligned_allocator_cpp11 should match Eigen default.

jlblancoc, 2018-03-02:
Very interesting...

Did you manually add `-mavx` to gcc flags? My CPU supports AVX but the flag is not set with "-type=native", it requires adding `-mavx`.

If the crash is reproducible, I would like to add some new unit tests to detect these conditions, for example: allocating in the heap and the stack several structures with Eigen matrices, of different sizes, different struct layouts, etc. with the idea of always detecting these errors if they have a chance to happen. Any better idea?

jolting, 2018-03-02:
test_mrpt_graphslam in debug on 17.10 hits the Eigen exception every time, so I'm not sure if a new unit test is required.

It looks like gcc-7 made a lot of progress with avx512, so 64-byte alignment is coming.
https://gcc.gnu.org/gcc-7/changes.html

I did not manually add -mavx, but 17.10 probably uses a newer version of Eigen and a different compiler. I'm not sure which changed the alignment requirement.

Solution 1: draft off Eigen
I think the best solution is to draft off of the Eigen alignment macro because that it really what is forcing alignment requirements for MRPT.

Solution 2: easy
64-byte alignment shouldn't increase for the foreseeable future. That's basically the cache line size in i7. The simplest solution is just to change the default from 16 to 64. It might unnecessary waste of a few bytes of memory.

jlblancoc, 2018-03-02:
> test_mrpt_graphslam in debug on 17.10 hits the Eigen exception every time, so I'm not sure if a new unit test is required.

Hmm... I've been using gcc-7.2 for a long time, and just in case I rebuilt that test with the same Eigen version that comes with 17.10, without luck: the test runs without crashing!

Anyway, I'll apply the "solution 1", it's the one I firstly thought of too :+1:

jlblancoc, 2018-03-02:
Done. Let me know if it fixed the crash.

closed 2018-03-03 by jolting

jolting, 2018-03-04:
Yes