MRPT/mrpt #666: graphslam-engine segfault at OpenGL lines

bergercookie, 2018-02-28:


jolting, 2018-02-28:
It might be useful to rerun that with debug information. Either RelWithDebug or Debug.

bergercookie, 2018-02-28:
Ah, apologies, uploaded wrong debug file.
I updated the description. Here's the file if this is more convenient.

[debug.txt](https://github.com/MRPT/mrpt/files/1768802/debug.txt)

jolting, 2018-02-28:
This = 0x0
```
mrpt::opengl::CSetOfObjects::getByClass<mrpt::opengl::CCamera> (this=0x0, ith=@0x7fffc96fe700: 0) at /home/berger/mrpt/libs/opengl/include/mrpt/opengl/CSetOfObjects.h:189
```

Hmm reusing `o` is probably bad. https://github.com/MRPT/mrpt/blob/master/libs/opengl/include/mrpt/opengl/CSetOfObjects.h#L200

It broke here.
https://github.com/MRPT/mrpt/commit/f18b03d5f2155b0c01a27c62c09f80f8db99b8ef

jlblancoc, 2018-02-28:
seems a dup of #645 , right? thanks for the debug files!

jolting, 2018-02-28:
Probably. I didn't see the stack trace of #645. I was hoping Nikos could try the fix I suggested. The shadowed variable is causing the problem.

jlblancoc, 2018-02-28:
I'm rebuilding to test a patch right now. A similar code exists in COpenGLViewport and there is no shadowed var there... damn large refactor! 

PS: It works, well seen @jolting ! 👍

jlblancoc, 2018-02-28:
It seems -Wshadow is not included into -Wall ... will fix that too.

closed 2018-02-28 by jlblancoc

bergercookie, 2018-03-01:
Good catch Hunter!
However, I don't think we should be enabling `-Wshadow` globally.. It seems that its current implementation in gcc is broken and makes it in part unusuable.  One can also see this by inspecting the latest build log - https://travis-ci.org/MRPT/mrpt/jobs/347540101

Relevant links:

- https://stackoverflow.com/questions/2958457/gcc-wshadow-is-too-strict
- https://lkml.org/lkml/2006/11/28/253

jolting, 2018-03-01:
I think most of those warnings can be fixed with a simple change to exceptions.h

jlblancoc, 2018-03-01:
Good stuff to read, @bergercookie !

After going thru it, and without having experienced it on my own, the idea of the compiler complaining about variables and functions with the same name doesn't hurt me at all, quite the opposite...

Let's give it a try, at the risk of getting Linus telling us we're doing it wrong! XD

jolting, 2018-03-01:
Uh oh, @jlblancoc is taking on @torvalds.
C++ and C are different beasts. MRPTs and STL namespaces make most of Jesper's points irrelevant.

bergercookie, 2018-03-01:
Lol!
Fair enough, let's see how this goes! :-)

commit 1e99eef2 (2018-02-28): Fix shadowed var causing mrpt-opengl regressions

Fix #666 . Fix #645 .