MRPT/mrpt #677: Investigate IEEE 1873-2015 format

jlblancoc, 2018-03-05:
1873-2015 - IEEE Standard for Robot Map Data Representation for Navigation.

To do: Investigate if it might be legal to provide an open-source implementation of this pay-per-read standard...

jolting, 2018-03-05:
I'm not a lawyer, but I believe it's fine as long as you don't post the PDF.
At least in the US Copyright law isn't usually the issue with this stuff. Patent law is and I doubt the standard is encumbered by patents.

At least two of XML map data samples in that doc is actually GPL or LGPL licensed, so you can use that at least.

jlblancoc, 2018-03-06:
For future reference: 
http://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=8259474

jlblancoc, 2018-10-30:
I had access to the IEEE norm via our institutional Library. This task is doable, will do it someday... It's only a pity they chose XML as the underlying format instead of YAML or alike...

Funny fact: the introduction of the IEEE standard explicitly motivates the need for it because the maps of OpenSLAM, MRPT, ROS and ORCA all define maps but they are not compatible with each other. 

This is in the spirit of #846 , btw...

bergercookie, 2018-11-04:
Looks interesting! haven't read the paper but all in all sounds like a good initiative.

closed 2026-10-02 by jlblancoc