MRPT/mrpt #678: How can I put odometry data and rplidar sensor data in one rawlog file

PDFCS, 2018-03-06:
I use rawlog-grabber to get source data from rplidar sensor like following figure:
[image]
but when I use this my dataset  to execute the rbpf-slam command, there are following exception:
 =============== MRPT EXCEPTION =============
virtual void mrpt::slam::CMetricMapBuilderRBPF::processActionObservation(mrpt::obs::CActionCollection&, mrpt::obs::CSensoryFrame&), line 187:
Assert condition failed: act2D
 /build/mrpt-6QpeJH/mrpt-1.5.6~snapshot20180118-git-850da4f3-trusty/libs/slam/src/slam/CMetricMapBuilderRBPF.cpp:248: In `virtual void mrpt::slam::CMetricMapBuilderRBPF::processActionObservation(mrpt::obs::CActionCollection&, mrpt::obs::CSensoryFrame&)`
 /build/mrpt-6QpeJH/mrpt-1.5.6~snapshot20180118-git-850da4f3-trusty/apps/rbpf-slam/rbpf-slam.cpp:655: In `void MapBuilding_RBPF()`
so I want to know the reason of the exception is that my dataset don`t have odometry data? if it is,how can I fix it, and how can I put odometry data and rplidar sensor data in one rawlog file like the following figure:
[image]

jlblancoc, 2018-03-06:
One possibility would be selecting the "Convert to SF format" menu from within RawLogViewer. Actions will be empty, but you could then use the menu "Generate odometry from laser scans" in the same program. Alternatively, just create a small C++ program that loads your data and writes the modified .rawlog with fake odometry (e.g. all zeros).

PDFCS, 2018-03-06:
But I can`t find menu "Generate odometry from laser scans" in RawLogViewer.

jlblancoc, 2018-03-06:
"Sensors -> Odometry -> recalculate actions from ICP"

I'm not 100% sure (can't recall it right now) if it will work if there
are no previous odometry at all, but it's easy to give it a test.

PDFCS, 2018-03-07:
I have solved the problem through your method.Now I can use my dataset to execute rbpf-slam command.Thank you very much!

closed 2018-03-07 by jlblancoc