MRPT/mrpt #679: Add a set of constexpr math functions

jolting, 2018-03-07:
For example, `::cos` is not a constexpr.
I think this can be useful for constexpr pose composition.

jlblancoc, 2018-03-07:
Exactly, that's what remains to be done before we are able to perform complex SE(2) / SE(3) operations at compile time... :+1: 

Though, I can only think of look-up-tables as a way to evaluate sin/cos as constexpr... Let's wait some time just in case we find a better solution.

jolting, 2018-03-07:
Looks like builtins like __builtin_sinf can be constexpr in gcc. Not in clang.

Not sure about msvc.

jolting, 2018-03-12:
An alternative to a lookup table is the taylor series:
https://github.com/Morwenn/static_math/blob/3f517d6f546eeaba19669877cb57338a91a3f89c/include/static_math/detail/cmath.inl

jolting, 2018-03-12:
Here's the glibc builtin implementation
https://sourceware.org/git/?p=glibc.git;a=blob;f=sysdeps/ieee754/dbl-64/s_sin.c;hb=HEAD