MRPT/mrpt #682: Error updating MRPT from the PPA repository from Ubuntu 16.04LTS

jlblancoc, 2018-03-11:
For the records: from now on, MRPT2 (>=1.9.9) requires gcc-7. If trying to installing / updating MRPT from the PPA repository from **Ubuntu 16.04 Xenial** (**NO ACTION** is required for Ubuntu 17.10 or 18.04LTS), one will have this error:

```
sudo apt install libmrpt-dev mrpt-apps
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 mrpt-apps : Depends: libmrpt-graphslam1.9 but it is not going to be installed
             Depends: libmrpt-gui1.9 but it is not going to be installed
             Depends: libmrpt-hmtslam1.9 but it is not going to be installed
             Depends: libmrpt-hwdrivers1.9 but it is not going to be installed
             Depends: libstdc++6 (>= 6) but 5.4.0-6ubuntu1~16.04.9 is to be installed
             Recommends: mrpt-common (= 1:1.9.9~snapshot20180311-0737-git-4acd71ce-xenial-1~ppa1~xenial) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

**Solution:**

Install gcc-7 packages (see [this gist](https://gist.github.com/jlblancoc/99521194aba975286c80f93e47966dc5)):

```
sudo add-apt-repository ppa:ubuntu-toolchain-r/test
sudo apt update
sudo apt install g++-7 -y
```

It's recommended to also update the symlinks to point to the latest version of the compiler:

```
sudo update-alternatives --install /usr/bin/gcc gcc /usr/bin/gcc-7 60 \
   --slave /usr/bin/g++ g++ /usr/bin/g++-7 
sudo update-alternatives --config gcc
gcc --version
g++ --version
```

closed 2018-03-11 by jlblancoc

davinderc, 2018-05-07:
Would this cause potential issues down the line if I don't need g++-7 for other purposes? Or better put, is g++-7 backwards compatible with the latest in the official Ubuntu repos? Kind of annoying to have to "fix" these issues by breaking what's working already...

jolting, 2018-05-07:
Alternatively set CC=gcc-7 and CXX=g++-7 prior to running the cmake commands. You don't have to set those as defaults.
Ubuntu 18.04 ships with g++-7, so no issues there. Consider upgrading if you don't need 16.04.

Finally, the master branch is not stable. The 1.5 branch is, so you can still use the system compiler there.

jlblancoc, 2018-05-07:
> Would this cause potential issues down the line if I don't need g++-7 for other purposes? 

If should work with older code. The worst I remember is having many new warnings in old code... but warnings are there for some reason, and fixing them is normally a wise thing to do ;-)

Alternatively, if time pressure is high for a given product, just add `-Wno...`  flags to avoid seeing new warnings as needed, and things should go on as usual.

> Finally, the master branch is not stable. The 1.5 branch is, so you can still use the system compiler there.

Yes, worth remarking! 👍 
We work hard to make `master` as stable as possible, but there are plans to further change some parts of the API before the release of 2.0.0-beta.