MRPT/mrpt #686: Fix clang-6.0 build

jolting, 2018-03-13:
Here is the error:
```
[ 33%] Linking CXX executable math_slerp_example
In file included from /mrpt/samples/math_matrix_example/test.cpp:13:
In file included from /mrpt/libs/math/include/mrpt/math/ops_vectors.h:17:
/mrpt/libs/math/include/mrpt/math/ops_containers.h:365:11: error: no viable overloaded '*='
        out_mean *= N_inv;
        ~~~~~~~~ ^  ~~~~~
/mrpt/libs/math/include/mrpt/math/ops_containers.h:399:2: note: in instantiation of function template specialization 'mrpt::math::meanAndCovVec<std::vector<mrpt::math::dynamic_vector<double>, std::allocator<mrpt::math::dynamic_vector<double> > >, std::vector<double, std::allocator<double> >, Eigen::Matrix<double, -1, -1, 0, -1, -1> >' requested here
        meanAndCovVec(v, m, C);
        ^
/mrpt/samples/math_matrix_example/test.cpp:222:19: note: in instantiation of function template specialization 'mrpt::math::covVector<std::vector<mrpt::math::dynamic_vector<double>, std::allocator<mrpt::math::dynamic_vector<double> > >, Eigen::Matrix<double, -1, -1, 0, -1, -1> >' requested here
                 << mrpt::math::covVector<vector<CVectorDouble>, Eigen::MatrixXd>(
                                ^
3 warnings and 1 error generated.
```

jlblancoc, 2018-03-14:
thanks for filing it!

jolting, 2018-03-14:
Just a reminder for later unless you want to take a look at it.

commit 509c3704 (2018-03-14): Ops_containers: More general VECTORLIKE multiply

fixes #686
closed 2018-03-14 by jolting