MRPT/mrpt #688: ammhf error in unit tests: alignment to be fixed

jlblancoc, 2018-03-14:
To fix:

```` 
[ RUN      ] Matrices.CMatrixFixedNumeric_loadWithEigenMap
test_mrpt_math: /usr/include/eigen3/Eigen/src/Core/MapBase.h:191: void Eigen::MapBase<Derived, 0>::checkSanity(typename Eigen::internal::enable_if<(Eigen::internal::traits<OtherDerived>::Alignment > 0), void*>::type) const [with T = Eigen::Map<Eigen::Matrix<double, 3, 4, 1, 3, 4>, 16, Eigen::Stride<0, 0> >; Derived = Eigen::Map<Eigen::Matrix<double, 3, 4, 1, 3, 4>, 16, Eigen::Stride<0, 0> >; typename Eigen::internal::enable_if<(Eigen::internal::traits<OtherDerived>::Alignment > 0), void*>::type = void*]: Assertion `( ((internal::UIntPtr(m_data) % internal::traits<Derived>::Alignment) == 0) || (cols() * rows() * innerStride() * sizeof(Scalar)) < internal::traits<Derived>::Alignment ) && "data is not aligned"' failed.
Aborted (core dumped)
````

[log](https://launchpadlibrarian.net/360654533/buildlog_ubuntu-bionic-armhf.mrpt_1%3A1.9.9~snapshot20180314-0850-git-509c3704-bionic-1~ppa1~bionic_BUILDING.txt.gz)

jolting, 2018-03-15:
It is possible that it is a stack alignment compiler bug. Try allocating on the heap instead of the stack.

jlblancoc, 2018-03-15:
Let's see if this one https://github.com/MRPT/mrpt/commit/f91a54b1d4a5b879607af993b328bf75a9959a51 has some effect...

jlblancoc, 2018-03-15:
Interesting.... after the patch to align to 32 instead of 16, it now also fails for arm64: 

https://launchpad.net/~joseluisblancoc/+archive/ubuntu/mrpt/+build/14459143/+files/buildlog_ubuntu-bionic-arm64.mrpt_1%3A1.9.9~snapshot20180315-1308-git-f91a54b1-bionic-1~ppa1~bionic_BUILDING.txt.gz

jlblancoc, 2018-03-15:
Perhaps alignas() works for a struct with an array inside, but it doesn't for a bare array?

jolting, 2018-03-16:
C++ reference says

```
// the array "cacheline" will be aligned to 128-byte boundary
alignas(128) char cacheline[128];
```

There are a lot of old gcc bugs related to the stack pointer alignment on ARM. That's why I wonder if it's a problem with allocating the data on the stack. That would make it a compiler bug.

jolting, 2018-03-16:
Try making the input arrays static.

jolting, 2018-03-16:
Or add
```
-mpreferred-stack-boundary=32
```

jolting, 2018-03-16:
Never mind that is an x86 only feature.

jlblancoc, 2018-03-18:
Converting the arrays [from local to global](https://github.com/MRPT/mrpt/commit/678ba87bcc97a125deaf519ad0338520c5976abb) made the trick (!). 

So this must be a compiler bug, sigh... 

All PPA builds are green now, anyway. Thanks for the idea, @jolting !!

jolting, 2018-03-18:
We should probably submit a bug to gcc. Alignas should work for stack alignment too. At the very least it should generate a warning.

 Increasing the stack alignment requirement might have some performance hits. It certainly use more stack space.

Maybe there is some magic arm gcc flag that fixes this.

jolting, 2018-03-19:
@jlblancoc 
Worth a read:
http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2016/p0035r4.html

jolting, 2018-03-19:
We should treat these as errors in the future:
```
/<<PKGBUILDDIR>>/libs/math/src/matrix_ops5_unittest.cpp:37:31: warning: requested alignment 32 is larger than 16 [-Wattributes]
  alignas(32) double nums[3 * 4] = {1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12};
                               ^
/<<PKGBUILDDIR>>/libs/math/src/matrix_ops5_unittest.cpp: In member function ‘virtual void Matrices_EigenMatrix_loadWithEigenMap_Test::TestBody()’:
/<<PKGBUILDDIR>>/libs/math/src/matrix_ops5_unittest.cpp:50:31: warning: requested alignment 32 is larger than 16 [-Wattributes]
  alignas(32) double nums[3 * 4] = {1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12};
```

jlblancoc, 2018-03-26:
Interesting... @jolting , do you any hypothesis of the reason for [this failure](https://travis-ci.org/MRPT/mrpt/jobs/358194786) after 9013eb43d1af9605349797393c43e4525cc226ac?

jolting, 2018-03-27:
I can't reproduce in release on 17.10 with clang. Let me rebuild in debug mode before breaking out the container.

jolting, 2018-03-27:
In container I get
```
test_mrpt_math: /usr/include/eigen3/Eigen/src/Core/MapBase.h:168: void Eigen::MapBase<Eigen::Map<Eigen::Matrix<double, 3, 4, 1, 3, 4>, 32, Eigen::Stride<0, 0> >, 0>::checkSanity() const [Derived = Eigen::Map<Eigen::Matrix<double, 3, 4, 1, 3, 4>, 32, Eigen::Stride<0, 0> >, Level = 0]: Assertion `((size_t(m_data) % (((int)1 >= (int)internal::traits<Derived>::Alignment) ? (int)1 : (int)internal::traits<Derived>::Alignment)) == 0) && "data is not aligned"' failed.
```
Something different with Eigen in 16.04?

jolting, 2018-03-27:
The problem is template param`Eigen::Aligned32` in `Eigen::Map` doesn't match the alignment of the array.

jolting, 2018-03-27:
Should be fixed by 8c633ad2416666b401ae9c8cec10c998eba796f8

closed 2018-03-27 by jolting