MRPT/mrpt #692: Can I use dlls from the precompiled binaries in VS17?

pettaa123, 2018-03-16:
Hi, I want to compile the test.cpp from example1 in VS17. I included the following directories of the installation from the prebuild binaries:
mrpt-1.5.5\include\mrpt\mrpt-config
mrpt-1.5.5\libs\base\include
Linker:
mrpt-1.5.5\bin
libmrpt-base155_msvc140_x64.dll

After starting compiling I get an error saying: corrupted file libmrpt-base155_msvc140_x64.dll

Is it possibile to use the dlls from the prebuild binaries for my own VisualStudio17 projects?

jolting, 2018-03-16:
It might be possible, but I wouldn't recommend it. Also, you should link the .lib file and not the .dll file.

Are you using cmake?

pettaa123, 2018-03-17:
Hi, my bad. Compiling was not the problem, the .exe couldn't find the dll when starting it. Adding mrpt/bin/ to PATH solved my issue. I would recommend writing this little hint into installation guide for all the newbe's like me. Thanks anyway :)

closed 2018-03-17 by pettaa123