MRPT/mrpt #693: Consider replacing void_ptr_noncopy with pimpl pattern

jolting, 2018-03-19:
https://github.com/MRPT/mrpt/tree/new_pimpl_pattern
This is an implementation of the new pimpl pattern using the experimental propogate_const function.

Without propogate_const constness might suffer. A simple one for portability might work until it's finished being standardized. Either way, the unique_ptr seems much cleaner.

void_ptr_noncopy isn't quite const correct anyway.

jlblancoc, 2018-03-19:
I like the idea of the unique_ptr! However, `propogate_const` is not available under experimental in MSVC, not even for MSVC2017 (!)...

jolting, 2018-03-19:
How about a mrpt::propogate_const?

jolting, 2018-03-19:
Maybe use a slightly modified version from LLVM?
https://github.com/llvm-mirror/libcxx/blob/master/include/experimental/propagate_const

jolting, 2018-03-19:
Hopefully this compiles on windows
```c++
//===------------------------ propagate_const -----------------------------===//
//
//                     MODIFIED FROM LLVM VERSION
//
// This file is dual licensed under the MIT and the University of Illinois Open
// Source Licenses. See LICENSE.TXT for details.
//
//===----------------------------------------------------------------------===//

#ifndef _MRPT_PROPAGATE_CONST
#define _MRPT_PROPAGATE_CONST

#include <type_traits>
#include <utility>
#include <functional>

namespace mrpt
{

template <class _Tp>
class propagate_const;

template <class _Up>
inline constexpr
const _Up& get_underlying(const propagate_const<_Up>& __pu) noexcept;

template <class _Up>
inline constexpr
_Up& get_underlying(propagate_const<_Up>& __pu) noexcept;

template <class _Tp>
class propagate_const
{
public:
  typedef std::remove_reference_t<decltype(*std::declval<_Tp&>())> element_type;

  static_assert(!std::is_array<_Tp>::value,
      "Instantiation of propagate_const with an array type is ill-formed.");
  static_assert(!std::is_reference<_Tp>::value,
      "Instantiation of propagate_const with a reference type is ill-formed.");
  static_assert(!(std::is_pointer<_Tp>::value && std::is_function<typename std::remove_pointer<_Tp>::type>::value),
      "Instantiation of propagate_const with a function-pointer type is ill-formed.");
  static_assert(!(std::is_pointer<_Tp>::value && std::is_same<typename std::remove_cv<typename std::remove_pointer<_Tp>::type>::type, void>::value),
      "Instantiation of propagate_const with a pointer to (possibly cv-qualified) void is ill-formed.");

private:
  template <class _Up>
  static constexpr element_type* __get_pointer(_Up* __u)
  {
    return __u;
  }

  template <class _Up>
  static constexpr element_type* __get_pointer(_Up& __u)
  {
    return __get_pointer(__u.get());
  }

  template <class _Up>
  static constexpr const element_type* __get_pointer(const _Up* __u)
  {
    return __u;
  }

  template <class _Up>
  static constexpr const element_type* __get_pointer(const _Up& __u)
  {
    return __get_pointer(__u.get());
[... 361 lines cut ...]

  bool operator()(const mrpt::propagate_const<_Tp>& __pc1,
      const mrpt::propagate_const<_Tp>& __pc2) const
  {
    return std::less_equal<_Tp>()(mrpt::get_underlying(__pc1), mrpt::get_underlying(__pc2));
  }
};

template <class _Tp>
struct greater_equal<mrpt::propagate_const<_Tp>>
{
  typedef mrpt::propagate_const<_Tp> first_argument_type;
  typedef mrpt::propagate_const<_Tp> second_argument_type;

  bool operator()(const mrpt::propagate_const<_Tp>& __pc1,
      const mrpt::propagate_const<_Tp>& __pc2) const
  {
    return std::greater_equal<_Tp>()(mrpt::get_underlying(__pc1), mrpt::get_underlying(__pc2));
  }
};

}

#endif
```

jolting, 2018-03-20:
Hopefully, that works on Windows.

jolting, 2018-03-21:
#695

closed 2018-03-21 by jolting