MRPT/mrpt #694: Unable to use kinect-stereo-calib

kaushik-rohit, 2018-03-19:
I am using mrpt on Ubuntu 17.10 and have compiled it using the source. But with the recent update of libmrpt-dev, I am unable to use the stereo-calib.

I get the following error on executing the program:
```
kinect-stereo-calib: symbol lookup error: kinect-stereo-calib: undefined symbol: _ZTIN4mrpt3gui15CWxGLCanvasBaseE
```

jolting, 2018-03-20:
If you are using the source you shouldn't use libmrpt-dev package. I think the problem is with your environment.

kaushik-rohit, 2018-03-20:
I uninstalled libmrpt-dev package but still it segfaults.
I tried generating stack trace using gdb for slam and the output is
```
rohit@rohit-Inspiron-5577:~/projects/mrpt/build$ gdb kinect-3d-slam 
GNU gdb (Ubuntu 8.0.1-0ubuntu1) 8.0.1
Copyright (C) 2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from kinect-3d-slam...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/local/bin/kinect-3d-slam 
warning: the debug information found in "/lib64/ld-2.26.so" does not match "/lib64/ld-linux-x86-64.so.2" (CRC mismatch).

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffc46cf700 (LWP 4470)]
[New Thread 0x7fffc3ece700 (LWP 4471)]
Waiting for sensor initialization...
Warning: Calibration file [kinect_calib.cfg] not found -> Using default params. This may cause inaccurate 3D point clouds.
 Checkout the kinect-stereo-calib application:
 http://www.google.com/search?q=kinect-stereo-calib%20mrpt&btnI 
Calling CKinect::initialize()...OK
[New Thread 0x7fffc36cd700 (LWP 4472)]
[New Thread 0x7fffc2ecc700 (LWP 4473)]
[New Thread 0x7fffc26cb700 (LWP 4474)]
[New Thread 0x7fffc1eca700 (LWP 4475)]
[New Thread 0x7fffc16c9700 (LWP 4476)]
[New Thread 0x7fffc0ec8700 (LWP 4477)]
[New Thread 0x7fffa3fff700 (LWP 4478)]
terminate called without an active exception

Thread 1 "kinect-3d-slam" received signal SIGABRT, Aborted.
__GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
51	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
(gdb) bt full
#0  __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:51
        set = {__val = {0, 0, 0, 140737353726056, 140737488341984, 8454951872, 140737353725200, 140737488341968, 140737305323599, 252833149, 4294967295, 140737488342000, 140737293197184, 
            140737353730272, 140737488342064, 5729668256}}
        pid = <optimized out>
        tid = <optimized out>
#1  0x00007ffff4613f5d in __GI_abort () at abort.c:90
        save_stage = 2
        act = {__sigaction_handler = {sa_handler = 0x0, sa_sigaction = 0x0}, sa_mask = {__val = {0, 3432, 140737293706456, 45, 4412750543122677053, 7451613976710379375, 140737297212992, 45, 
              140737306443056, 45, 45, 45, 140737297217648, 0, 0, 140737297212992}}, sa_flags = -9632, sa_restorer = 0x55555582db70}
        sigs = {__val = {32, 0 <repeats 15 times>}}
#2  0x00007ffff51d3563 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
No symbol table info available.
#3  0x00007ffff51d84f6 in ?? () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
No symbol table info available.
#4  0x00007ffff51d8531 in std::terminate() () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
No symbol table info available.
#5  0x0000555555561e1f in Test_Kinect() ()
No symbol table info available.
#6  0x000055555555bf5a in main ()
No symbol table info available.
```

jolting, 2018-03-20:
You should also uninstall the rest of the libmrpt* packages.
Generally it is not necessary to do a `make install` during development. You can run it from the build directory.

jlblancoc, 2018-03-20:
To make sure you start from a fresh environment, try running `sudo
make uninstall` from the same mrpt BUILD directory where you
originally did the `sudo make install`. Hopefully, if you didn't
overwrite it already, it should clean up all your mrpt files under
`/usr/local/*`.

I would advice *not* to install using `make install` in any case;
there are alternatives to force CMake or even pkg-config to find MRPT
in its build directory without polluting /usr/....

Cheers,

jolting, 2018-03-25:
Closing due to inactivity. Please reopen if you are still having problems.

closed 2018-03-25 by jolting