MRPT/mrpt #703: Handle LGPL more cleaning

jolting, 2018-03-31:
LGPL is stricter than BSD, so it simply can't be relicensed as BSD code. It should be clear what is LGPL derived code. Ideally, LGPL code should be in its own library(.so) when linking to BSD code.

Based on explicit mentions of LGPL in MRPT code here are my suggestions.
- [ ] libs/vision/src/faster/*
Faster can be in libmrpt-vision-faster.so

libmrpt-vision-faster.so would be LGPL.

- [ ] libs/vision/src/ba_full.cpp and libs/vision/include/mrpt/vision/bundle_adjustment.h
libmrpt-vision-ba.so
libmrpt-vision-ba.so would be LGPL.

- [ ] libs/system/include/mrpt/system/memory.h
Rewrite is_aligned from scratch or copy a BSD implementation.

Still undecided
- [ ] libs/opengl/src/gl_utils.cpp
- [ ] libs/opengl/include/mrpt/opengl/CText3D.h
- [ ] libs/opengl/include/mrpt/opengl/gl_utils.h
- [ ] libs/img/src/CImage_SSE2.cpp

jlblancoc, 2018-04-24:
@jolting I think your recent branch would solve all the pending items here, right?

jolting, 2018-04-24:
libs/vision/src/ba_full.cpp and libs/vision/include/mrpt/vision/bundle_adjustment.h actually needs to be solved.