MRPT/mrpt #706: MRPT Test fails

JohnDuke259, 2018-04-02:
Hello, I complied mrpt 1.5.6 on my Ubuntu Xenial using the source file in the branch. However, when I run the test "test_mrpt_base" I get a failure message:
[  FAILED  ] Matrices.SerializeCMatrixD
[  FAILED  ] SerializeTestBase.WriteReadToMem.

Similarly, when I run "test_mrpt_slam", MRPT throws an exception and I get a failure message.
  [  FAILED  ] MonteCarlo2D.RunSampleDataset.

Please can someone tell me if there are fixes for these because I am new to mrpt.

Thank you!

jolting, 2018-04-03:
1.5.6 hasn't officially been released yet. The head of the branch should be ok.

The PPA shows the head of the branch working:
https://launchpadlibrarian.net/362114691/buildlog_ubuntu-xenial-amd64.mrpt_1%3A1.5.6~snapshot20180326-git-7d090139-xenial-1~ppa1~xenial_BUILDING.txt.gz

Please provide a full log.

jolting, 2018-04-04:
Try running that with RelWithDebInfo or Debug. That would make the Backtrace more readable.

jlblancoc, 2018-04-04:
Thanks for the report, @JohnDuke259 !
As @jolting  said, if you build with Debug symbols we'll be able to find the offending line (although, it might be that in Debug builds, the error disappears.. it happens sometimes).  
Alternatively, you can manually run it with gdb:
``` gdb tests/tests_mrpt_vision```
type `run`, then `bt` when the crash happens.

jlblancoc, 2018-04-05:
Are you building static libs by chance? (I couldn't see it in your log)

JohnDuke259, 2018-04-05:
Yes @jlblancoc. I am building static libs.

jlblancoc, 2018-04-05:
Ok, then that's a known problem of static linking: the compiler optimizes
out the automatic registration of rtti classes. I'm on the phone now, but I
recall there was a specific function to force registration of all classes
within a given mrpt module, we should take a look at it.

Static building is not tested on our CI tests, that's why this undergone
unnoticed...

jlblancoc, 2018-04-06:
The function I meant is `mrpt::utils::registerAllPendingClasses()` but the problem is that, when building static libs, the compiler optimizes out even the functions that add the "to-be-registered" list, so calling that function won't help. 

In the past I did some research and the best I could do to force each compiler to always execute a code at "init" (before "main()") is this: 

https://github.com/MRPT/mrpt/blob/213cb1c4909ec74ef3b465db08f02798aa68e59c/libs/base/include/mrpt/utils/initializer.h#L12-L43

As the comment says, static libs in MSVC was the most difficult case since I never found a way to automatically do the registration of classes. Perhaps we should make `MRPT_INITIALIZER(registerAllClasses_mrpt_opengl)` and such public functions so "users" (including the unit tests) can manually call them? Thoughts?

Obviously, for now the best I can advise is not to use static linking unless unavoidable.

jlblancoc, 2018-04-06:
See also: https://stackoverflow.com/a/2390626/1631514

JohnDuke259, 2018-04-07:
Thank you so much for your effort @jlblancoc and @jolting. Much appreciated! I will try building the shared library and let you know. The last time I attempted building a shared library on mrpt, it failed with a different error message.

jlblancoc, 2018-04-07:
On Fri, Apr 6, 2018 at 8:22 AM, Hunter Laux <notifications@github.com> wrote:
>
> Does __attribute__((used)) help with the Linux case though?
>
>
I'll try it...

jolting, 2018-04-12:
I've made similar mistakes before. I've made it a rule to never run make install. I've gone years without ever actually executing that command on my desktop.

I see you've built ffmpeg yourself and then done a make install.
On Ubuntu 16.04 you should use the packaged version.
The problem you will have is uninstalling ffmpeg. Make installs cannot easily be undone in most cases. You can try manually deleting the traces of ffmpeg out of your /usr/local folder. The only foolproof solution is the reinstall Ubuntu.

Next time use checkinstall or build in a container.

JohnDuke259, 2018-04-13:
Thanks @jolting

jlblancoc, 2018-04-14:
> I've made it a rule to never run make install.

Absolutely agree, same here!

> Does __attribute__((used)) help with the Linux case though?

Unfortunately, nope, it doesn't...

jlblancoc, 2018-04-14:
I can confirm the issue that entitles this thread (problems with static builds) still remains even after the commit referenced above. The problem is that the `MRPT_INITIALIZER()` function is never called in static builds. 

So we could keep this one open, unless: a) a workaround is found, b) we officially decide mrpt doesn't support automatic class registry in static builds.

jlblancoc, 2018-05-01:
Related WIP: https://github.com/MRPT/mrpt/commit/244cdeb8b0c0b1d2000ea12e953b3de87de85dad

jlblancoc, 2018-10-29:
What should we do with this one? 

I think we could just drop support for automatic registration of classes in static builds... Any **good** or **easy** alternative?

jolting, 2018-10-29:
So many features depend on the class registration and it will lead to breakage at runtime in unexpected places. I'd prefer just leaving this as wontfix and disabling static builds. If some enterprising engineer wants to make it work then I will certainly appreciate a PR, but I don't see this as a priority for the community.

Perhaps this is something to address if/when Modules TS becomes standardized.

PR #1222 (2022-05-21): Fix registration for static lib builds
## Changed apps/libraries

* all

TODO: 
- [x] Make recursive calls from each module to its deps in the registration method.

## PR Description

Static builds lack the .so / .dll load-triggered automatic RTTI registration of classes. This PR adds support for manual calling a replacement for that. 
   

Fixes #706
closed 2022-05-21 by jlblancoc