MRPT/mrpt #710: mrpt-performance exception in Save as JPEG tests on Windows with OpenCV 3.4.1

jolting, 2018-04-09:
```
images: Save as JPEG (640x480, quality=95%)                 OpenCV(3.4.1) Error: Assertion failed ((flags & FIXED_TYPE) != 0) in cv::_InputArray::type, file C:\build\master_winpack-build-win64-vc15\opencv\modules\core\src\matrix_wrap.cpp, line 807
Skipped due to exception:
OpenCV(3.4.1) C:\build\master_winpack-build-win64-vc15\opencv\modules\core\src\matrix_wrap.cpp:807: error: (-215) (flags & FIXED_TYPE) != 0 in function cv::_InputArray::type

c:\mrpt\libs\img\src\cimage.cpp:315: In `mrpt::img::CImage::saveToFile`

```

jolting, 2018-04-22:
https://github.com/opencv/opencv/issues/11076

jlblancoc, 2018-04-23:
Thanks for the reference.

It seems a good moment for upgrading the `IplImage*` interface and containers to `cv::Mat`. It will be a lot of dirty work since it's used all across mrpt, though... (sigh)
But it's a good thing to do.

jlblancoc, 2018-04-23:
Better, I'll merge this task into #120.

closed 2018-04-23 by jlblancoc