MRPT/mrpt #719: build error

cumtchenchang, 2018-04-22:
I installed vtk6.2, however the error is still on.
 29%] Building CXX object libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CRandomFieldGridMap3D.cpp.o
/home/cc/software/mrpt-mrpt-1.5/libs/maps/src/maps/CRandomFieldGridMap3D.cpp:20:32: fatal error: vtkStructuredGrid.h: No such file or directory
compilation terminated.
libs/maps/CMakeFiles/mrpt-maps.dir/build.make:627: recipe for target 'libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CRandomFieldGridMap3D.cpp.o' failed
make[2]: *** [libs/maps/CMakeFiles/mrpt-maps.dir/src/maps/CRandomFieldGridMap3D.cpp.o] Error 1
CMakeFiles/Makefile2:1209: recipe for target 'libs/maps/CMakeFiles/mrpt-maps.dir/all' failed
make[1]: *** [libs/maps/CMakeFiles/mrpt-maps.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2

jolting, 2018-04-22:
How did you install VTK? 6.3 on Ubuntu 18.04 works.

jolting, 2018-04-24:
No response in 2 days. Please reopen if you're still having problems. This is likely an environment issue.

closed 2018-04-24 by jolting

cumtchenchang, 2018-04-25:
I use ubuntu16.04. i install VTK using "sudo apt-get install  libvtk6.2". Besides, I also intsall vtk5, but the error is still on