MRPT/mrpt #723: Bug in CLoopCloserERD

MaxGsomGsom, 2018-04-22:
I think there should be ``if (!it->second)``

https://github.com/MRPT/mrpt/blob/50111a08efeb0744d5e97658b30f9131c5fef5f3/libs/graphslam/include/mrpt/graphslam/ERD/CLoopCloserERD_impl.h#L2515-L2527

jolting, 2018-04-22:
CC: @bergercookie

bergercookie, 2018-04-22:
Yeah, this seems to be a bug.
I'm surprised this is there because it's quite big for me not to notice it.

Anyway, feel free to make a PR for this, otherwise I can patch it in the couple of days.

MaxGsomGsom, 2018-04-23:
Also it does not compile:
https://github.com/MRPT/mrpt/blob/50111a08efeb0744d5e97658b30f9131c5fef5f3/libs/graphslam/include/mrpt/graphslam/ERD/CLoopCloserERD_impl.h#L2539-L2548

Because
https://github.com/MRPT/mrpt/blob/50111a08efeb0744d5e97658b30f9131c5fef5f3/libs/slam/include/mrpt/slam/CIncrementalMapPartitioner.h#L137-L139

jlblancoc, 2018-04-23:
Just fixed with 630ad757efc9873f64c735f36e6afe977c2a056f

MaxGsomGsom, 2018-04-23:
I have tried to compile again but it shows error:
``/home/max/Desktop/mrpt/libs/slam/include/mrpt/slam/CIncrementalMapPartitioner.h:137:11: note:   no known conversion for argument 2 from ‘std::__shared_ptr_access<mrpt::poses::CPosePDF, (__gnu_cxx::_Lock_policy)2, false, false>::element_type {aka mrpt::poses::CPosePDF}’ to ‘const mrpt::poses::CPose3DPDF&’``

jolting, 2018-04-23:
That might be an explicit conversion. Try `mrpt::poses::CPose3DPDF(*posePDF)`

jlblancoc, 2018-04-23:
Should be fixed after #726, if not please reopen...

commit 48afc736 (2018-04-24): Inverted logic for checking if laserScan invalid

Fixes #723
closed 2018-04-23 by jlblancoc