MRPT/mrpt #727: CGridMapAligner::AlignPDF

MaxGsomGsom, 2018-04-24:
``CGridMapAligner::AlignPDF()`` causes ``std::bad_alloc`` when used ``amModifiedRANSAC`` or ``amRobustMatch`` algorithms. I have tried standart params and many different combinations. Grid map of default size. Why it can happen?

Example on config:
```
featsPerSquareMeter		= 0.015
threshold_max			= 0.15 		// For considering candidate matches
threshold_delta			= 0.15
ransac_prob_good_inliers = 0.9999999999  // Prob. of a good inliers (for the number of iterations).
maxKLd_for_merge        = 0.9		// Merge of close SOG modes
min_ICP_goodness	= 0.60
max_ICP_mahadist	= 10 
ransac_minSetSizeRatio	= 0.40 // 0.20
ransac_mahalanobisDistanceThreshold	= 6		// amRobust method only
ransac_chi2_quantile	= 0.99 				// amModifiedRANSAC method only
save_feat_coors			= 0		// Dump correspondences to grid_feats
debug_save_map_pairs            = 0		// Save the pair of maps with the best correspondences
debug_show_corrs		= 0		// Debug output of graphs

# All the params of the feature detectors/descriptors
# ----------------------------------------------------------
featsType			= 1
feature_descriptor		= 8
KLTOptions.min_distance		= 6			// Pixels
KLTOptions.threshold		= 0.20 // 0.00001
harrisOptions.min_distance	= 6			// Pixels
harrisOptions.threshold 	= 0.20
SIFTOptions.implementation	= 3			// Hess
SURFOptions.rotation_invariant	= 1		// 0=64 dims, 1=128dims
SURFOptions.hessianThreshold	= 600
SURFOptions.nOctaves		= 2
SURFOptions.nLayersPerOctave    = 8
SpinImagesOptions.hist_size_distance	= 10 
SpinImagesOptions.hist_size_intensity	= 10 
SpinImagesOptions.radius			= 20
PolarImagesOptions.bins_angle			= 8
PolarImagesOptions.bins_distance		= 6
PolarImagesOptions.radius			= 20
LogPolarImagesOptions.radius			= 20
LogPolarImagesOptions.num_angles		= 8
```

jolting, 2018-04-24:
Do you have a backtrace?

jlblancoc, 2018-04-24:
I couldn't reproduce the crash using the app `grid-matching`. 
In what context are the call being done? graphslam? a custom app? 
As jolting says above, try building with `-g`  and inspect the stack trace at the throw.

jlblancoc, 2018-05-18:
Any progress here?

brunoeducsantos, 2020-09-23:
@jlblancoc  I have a similar issue with `std::bad_alloc`.
My code is the following:
```
mrpt::poses::CPose2D GridMapAligner::Align(mrpt::maps::COccupancyGridMap2D::Ptr map1, mrpt::maps::COccupancyGridMap2D::Ptr map2)
{

	mrpt::maps::CMultiMetricMap multimap1;
	{
		multimap1.maps.emplace_back(map1);
	}
	mrpt::maps::CMultiMetricMap multimap2;
	{
		multimap2.maps.emplace_back(map2);
	}
	try
	{
		mrpt::poses::CPose2D estimateMean;
      	mrpt::math::CMatrixDouble33 estimateCOV;
		mrpt::poses::CPosePDF::Ptr parts = gridAlign.Align(&multimap1, &multimap2, mrpt::poses::CPose2D(0, 0, 0), info);
		mrpt::poses::CPosePDFSOG::Ptr pdf_SOG = std::make_shared<mrpt::poses::CPosePDFSOG>();
		pdf_SOG->copyFrom(*parts);
		pdf_SOG->getMostLikelyCovarianceAndMean(estimateCOV, estimateMean);
		return estimateMean;
	}
	catch (const std::exception &e)
	{
		std::cerr << mrpt::exception_to_str(e);
	}
}
````
I used a `try catch` to get the exception but it only shows `bad_alloc`.
Any suggestions?

Thanks,
Bruno