MRPT/mrpt #728: Evaluate chaiscript for expr library

jolting, 2018-04-24:
https://github.com/ChaiScript/ChaiScript

c++14.
Lots of test coverage
Good documentation
Very complete scripting language.

https://codedocs.xyz/ChaiScript/ChaiScript/LangGettingStarted.html

bergercookie, 2018-04-24:
@jolting I've heard many nice things about ChaiScript. Is there a specific application we want to use it for or just integrate it and use it to evaluate string expressions?

jolting, 2018-04-24:
Right now you can embed scripts in ini files for certain classes. chaiscript might be a good replacement for that. Mostly I think the c++ interface is a little nicer.

jlblancoc, 2018-04-25:
@bergercookie right now it's used mainly for some reactive nav configuration files, where some entries are scripts on their own. See for example [this](https://github.com/MRPT/mrpt/blob/master/share/mrpt/config_files/navigation-ptgs/reactive2d_config.ini#L120)). 

Indirectly it's also used via the `$eval{}` keyword in CConfigFileBase, so any MRPT "config file" already allows scripting  ;-) It's barely documented here: http://mrpt.ual.es/reference/devel/config_file_format.html

jolting, 2018-04-25:
P.S. ChaiScript is Jason Turner's pet project. https://www.youtube.com/user/lefticus1
It's worth watching some of his videos if you have the time.

bergercookie, 2018-04-25:
>P.S. ChaiScript is Jason Turner's pet project. https://www.youtube.com/user/lefticus1 It's worth watching some of his videos if you have the time.

Oh, and I thought I'd heard of that guy before!
I'm listening to his [CppCast](http://cppcast.com/) podcast every day to work! :)