MRPT/mrpt #737: Travis: Figure out why PCL and lib3ds aren't being used despite being in the build container

jolting, 2018-04-28:
```
--  ______________________ OPTIONAL LIBRARIES ______________________
--  Assimp (3D models)                   : Yes (System) [Version: 4.1.0]
--  ffmpeg libs (Video streaming)        : Yes (System) [avcodec 57.107.100, avutil 55.78.100, avformat 57.83.100]
--  gtest (Google unit testing library)  : Yes (Built-in) 
--  lib3ds (3DStudio scenes)             :  No 
--  libjpeg (jpeg)                       : Yes (System) 
--  liblas (ASPRS LAS LiDAR format)      : Yes (System) 
--  mexplus                              :  No 
--  Octomap                              : Yes (System) [Version: 1.8.1]
--  OpenCV (Image manipulation)          : Yes (System) [Version: 3.2.0]
--  OpenGL                               : Yes (System) 
--  GLUT                                 : Yes (System) 
--  PCAP (Wireshark logs for Velodyne)   : Yes (System) 
--  PCL (Pointscloud library)            :  No [Version: ]
--  SuiteSparse                          : Yes (System) 
--  VTK                                  : Yes 
--  yamlcpp (YAML file format)           : Yes (System) [Version: 0.5.2]
```

jolting, 2018-04-29:
PCL was disabled.
lib3ds-config doesn't look like it's available anymore.

jlblancoc, 2018-04-29:
thanks for finding it out.

It's weird... maybe a problem packaging because the .m4 file and the man page for the lib3ds-config executable are there: https://packages.debian.org/sid/amd64/lib3ds-dev/filelist

We might have to just use cmake to find the .so file "by hand" (sigh)...

On the other hand, it might be the final drop that makes us reconsider to remove the support for lib3ds and the associated mrpt::opengl object. The assimp one is far, far superior anyway and the only reason to leave lib3ds support was to support *some* kind of 3D models for builds without assimp...

PR #739 (2018-04-29): Remove C3DScene and lib3ds support
Closes #737

It seems C3DScene wasn't used anywhere.
commit cbce8849 (2018-04-29): Remove C3DScene and lib3ds support

Closes #737

It seems C3DScene wasn't used anywhere.
closed 2018-04-29 by jlblancoc