MRPT/mrpt #743: CActionRobotMovement2D::computeFromOdometry throwing exception: dereferencing nullptr poly_ptr

RichardFieb, 2018-05-02:
```c++
mrpt::obs::CActionRobotMovement2D::TMotionModelOptions opts;
int main(int argc, char** argv)
{
    CActionRobotMovement2D act;
    CPose2D odo(1, 2, DEG2RAD(45));

    opts.modelSelection = CActionRobotMovement2D::mmThrun;
    opts.thrunModel.nParticlesCount = 300;
    opts.thrunModel.alfa1_rot_rot = 0.05f;
    opts.thrunModel.alfa2_rot_trans = 4;
    opts.thrunModel.alfa3_trans_trans = 0.01f; //0.10f // example
    opts.thrunModel.alfa4_trans_rot = 0.0001;
    opts.thrunModel.additional_std_XY = 0.001f;
    opts.thrunModel.additional_std_phi = 0.050f;

    act.computeFromOdometry(odo, opts);
    return 0;
}
```
while running the code above I get an error:
terminate called after throwing an instance of 'std::runtime_error'
  what():  dereferencing nullptr poly_ptr

Also using the RawLogViewer App brings the same exception: dereferencing nullptr poly_ptr.
Just open the App click on <Motion model> and then select "Thrun's book model" and click on <Draw samples>.

I build MRPT from source from master

jlblancoc, 2018-05-03:
can't reproduce in MSVC... will try in gcc.

jlblancoc, 2018-05-04:
I can't reproduce with current master and GCC. Even with gdb or valgrind, everything works fine. 

My system has gcc 7.3, that's the only difference, but I don't think it should be an issue.

PS: RawLogViewer -> "Thrun's model" -> "Draw particles" also works for me.

jlblancoc, 2018-05-18:
Closing due to lack of feedback. Please, reopen if the problem persists.

closed 2018-05-18 by jlblancoc