MRPT/mrpt #763: Build failing on 16.04

cressell, 2018-05-14:
Hi,

I can't seem to build the latest master version:

`/usr/local/include/eigen3/Eigen/src/plugins/IndexedViewMethods.h:41:139: error: use of deleted function ‘boost::python::extract<int>::extract(const boost::python::extract<int>&)’
   return internal::makeIndexedViewCompatible(indices, internal::variable_if_dynamic<Index,RowsAtCompileTime>(derived().rows()),Specialized);`

Is the error I'm getting. What is this related to?

Thanks-
Cressell

jlblancoc, 2018-05-14:
Hi,

`/usr/local/`

==> What version of eigen is that? In general, I discourage doing a "sudo make install" of any library. Nowadays, CMake makes it unnecessary and it's a constant source of version conflicts and problems... probably the version you have installed is too old for the new gcc-7 version; that would be my bet. 

If you don't need Eigen under /usr/local, please delete it and re-run cmake for mrpt after removing CMakeCache.txt; alternatively, mark the DOWNLOAD_EIGEN CMake variable while configuring MRPT to enforce using a modern version of Eigen.

cressell, 2018-05-15:
Thanks @jlblancoc, you're probably right there. I've been messing about with Eigen for eternity due to other packages not complying. I'll remove it and install eigen3 3.3~beta1-2 from apt-get.

Good stuff @jolting!

cressell, 2018-05-15:
Indeed Eigen related! Thanks, closing this.

closed 2018-05-15 by cressell