MRPT/mrpt #767: mrpt1 builds failing on ARM platforms

mikaelarguedas, 2018-05-15:
The version currently released in ROS fails to build on ARM for both kinetic and melodic (link to the failing jobs below).
Is mrpt1 expected to work on ARM ? if so we'll look forward the next release. Otherwise please let us know so that we can disable the arm jobs for this package.
Thanks!

http://build.ros.org/job/Mbin_dsv8_dSv8__mrpt1__debian_stretch_arm64__binary/
http://build.ros.org/view/Mbin_ubhf_uBhf/job/Mbin_ubhf_uBhf__mrpt1__ubuntu_bionic_armhf__binary/
http://build.ros.org/view/Mbin_ubv8_uBv8/job/Mbin_ubv8_uBv8__mrpt1__ubuntu_bionic_arm64__binary/

http://build.ros.org/view/Kbin_uxhf_uXhf/job/Kbin_uxhf_uXhf__mrpt1__ubuntu_xenial_armhf__binary/
http://build.ros.org/view/Kbin_uxv8_uXv8/job/Kbin_uxv8_uXv8__mrpt1__ubuntu_xenial_arm64__binary/
http://build.ros.org/view/Kbin_djv8_dJv8/job/Kbin_djv8_dJv8__mrpt1__debian_jessie_arm64__binary/

/cc @clalancette

jolting, 2018-05-15:
I believe this was recently addressed: https://github.com/MRPT/mrpt/commit/219020317d503bfac08a44ee10b2415ff7ae7150

jlblancoc, 2018-05-15:
@mikaelarguedas Exactly, as @jolting says, this was fixed just yesteday in that commit. However, we have a more serious issue (https://github.com/ros-infrastructure/ros_buildfarm/issues/543) trying to release for kinetic. I could make a new release with the patch for melodic, though. Just me know how "urgent" do you think it is, just in case we first attempt to solve the Kinetic build farm problem.

mikaelarguedas, 2018-05-15:
Thanks for the quick response.
Anytime before the melodic release ([May 23rd](https://www.google.com/search?q=world+turtle+day&oq=world+turtle+day&aqs=chrome..69i57j0l5.2896j1j7&client=ubuntu&sourceid=chrome&ie=UTF-8)) is fine

jlblancoc, 2018-05-16:
Should be done after https://github.com/ros/rosdistro/pull/17869

We'll keep an eye on the build farm anyway... 
Cheers!

jolting, 2018-05-16:
Should use the packaged octomap version instead of downloading it.

jlblancoc, 2018-05-16:
Right! 
Also, it still [fails](http://build.ros.org/job/Mbin_ubhf_uBhf__mrpt1__ubuntu_bionic_armhf__binary/7/console). Will check the CMake logic, probably I made some stupid mistake. 
After fixing those two points, will try it again.

jlblancoc, 2018-05-17:
Should be fixed after: 
* https://github.com/MRPT/mrpt/pull/768
* https://github.com/MRPT/mrpt/pull/769

Rosdistro PR: 
* https://github.com/ros/rosdistro/pull/17870

jolting, 2018-05-18:
That certainly takes a long time to build. Almost 8 hours and still no where near complete.

jlblancoc, 2018-05-18:
Building the apps [takes](http://build.ros.org/job/Mbin_ubhf_uBhf__mrpt1__ubuntu_bionic_armhf__binary/8/consoleFull) 50 minutes, so a solution would be just build (and install) the libraries. I think it can be done via a patch to the CMake script, probably as a [bloom patch](http://wiki.ros.org/bloom/Tutorials/ReleaseThirdParty#Adding_an_Install_Rule_as_a_Patch). 

Will give it a try...

jlblancoc, 2018-05-18:
Finally found [this](http://wiki.ros.org/bloom/Tutorials/ChangeBuildFlags) which seems a better idea. 
Just made [this change](https://github.com/mrpt-ros-pkg-release/mrpt1-release/commit/97f5ed6c05a9c1840f934ebf3be355dd4e4404b6). Let's repeat the release and wait for the build farm...

mikaelarguedas, 2018-05-19:
Looks like it did the job :+1:
Thanks for the quick fix!

closed 2018-05-19 by mikaelarguedas